pbatard/rufus #805: Could not get Disk Extents...Incorrect function...No volume information... I need help because I haven't figured this out after tinkering for hours and I need to move forward. TIA

binaryBronco, 2016-08-02:
<PLEASE READ THIS CAREFULLY: You *MUST* read and complete the checklist below, by placing an x into each [ ], BEFORE clicking on 'Submit new issue'. Failure to perform these steps, WHICH ARE ONLY THERE TO HELP *YOU*, will result in the issue being dismissed without warning.>

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues, using keywords relevant to my problem.
- [x] I clicked the `Log` button in Rufus and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [x] I ran a bad blocks check, by clicking the "bad blocks" check box in Rufus, and confirmed that my USB is not defective
- [x] I also tried one or more of the following:
  - [x] Using a different USB drive
  - [ ] Plugging the USB into a different port
  - [ ] Running Rufus on a different computer
- [ ] If using an ISO image, I clicked on the `#` button (at the bottom of the Rufus interface), to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Looking for help from PETE or anyone else who has experience with Rufus removing old flash drive partitions and then failing a new partition sequence. I have happily used Rufus for over a year to make my Linux boot drives, and this is the first issue I've had that I can't fix after multiple hours of debug. I am up to 3 non-responsive drives (2 of 3 can still be seen by Rufus, but 0 of 3 can be seen by Windows in Disk Management or via CL using DISKPART). 2 of the 3 are SD cards (2GB and 4GB), the third is a now non-responsive 16GB Verbatim USB drive. The two SD cards give identical failing messages below, with error [0x00000001] Incorrect function quite prevalent. I completed a Bad Block scan on one them just to show that I'm confident this is not a matter of a failing flash array. The array can still be written and read correctly, but it appears Rufus is unable to re-partition a new file system. Similarly, I can't create a new file system using Windows as the drives are unaddressable. What can I do? I would appreciate someone pointing me in the right direction because I can't grind on this unproductively forever. I appreciate the help. PETE, thank you for your development on this tool. If these drives are somehow recoverable, is there a way that the tool can be enhanced to provide this functionality? Thank you everyone in advance.

Log
---
Rufus version: 2.9.934
Windows version: Windows XP SP3 32 bit
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta3
System locale ID: 0x0409
Will use default UI locale 0x0409
0 devices found
0 devices found
0 devices found
0 devices found
0 devices found
0 devices found
0 devices found
Found card reader device 'SD Memory Card'
1 device found
Could not get Disk Extents: [0x00000001] Incorrect function.
No volume information for drive 0x81
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 1, TracksPerCylinder: 255, SectorsPerTrack: 63
Could not get layout for drive 0x81: [0x00000001] Incorrect function.
Found card reader device 'SD Memory Card'
1 device found
Could not get Disk Extents: [0x00000001] Incorrect function.
No volume information for drive 0x81
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 1, TracksPerCylinder: 255, SectorsPerTrack: 63
Could not get layout for drive 0x81: [0x00000001] Incorrect function.

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE1 for write access
No drive letter was assigned...
Will use 'G:' as volume mountpoint
Could not get Disk Extents: [0x00000001] Incorrect function.
No logical drive found (unpartitioned?)
Drive does not appear to be partitioned
Analyzing existing boot records...
Drive does not have an x86 Master Boot Record
Bad Blocks: Checking from block 0 to 7733247
Bad Blocks: Using offset 0 for fake device check
Bad Blocks: Writing test pattern 0xAA
Bad Blocks: Reading and comparing
Bad Blocks: Check completed, 0 bad blocks found. (0/0/0 errors)
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Partitioning (MBR)...
Could not reset disk: [0x00000001] Incorrect function.
Could not get Disk Extents: [0x00000001] Incorrect function.

Found card reader device 'SD Memory Card'
1 device found
Could not get Disk Extents: [0x00000001] Incorrect function.
No volume information for drive 0x81
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 1, TracksPerCylinder: 255, SectorsPerTrack: 63
Could not get layout for drive 0x81: [0x00000001] Incorrect function.

<FULL LOG>
```

pbatard, 2016-08-02:
> 2 of 3 can still be seen by Rufus, but 0 of 3 can be seen by Windows in Disk Management or via CL using DISKPART

Wow. This is really weird. That's the first time someone report not seeing a drive in Disk Management that they can see in Rufus.

The first thing I would try then, if that's possible for you, is try to repartition your drives from Linux or OS X if such a machine is available to you (a Virtual Machine running on Windows won't do - it needs to be a real Linux or OS X one. But it can be as simple as Linux running on a Raspberry Pi.). You should be able to try `fdisk /dev/sdX` then `mkfs.vfat /dev/sdX1` where `X` is the letter for your drive (caution!) and see if tha

binaryBronco, 2016-08-03:
On Linux Mint, I could not get any drives to show (0 of 3), using the command `lsblk` in terminal. I will try to see if a Macbook at home has any better luck.

pbatard, 2016-08-03:
Okay. That doesn't look to good if even Linux can't detect your drives, which is very strange.
I know there exist some USB flash drives with buggy firmware that become read-only after Rufus attempts to repartition and reformat them (PNY 16GB Retract USB and maybe some other PNY drives), but you're using SD cards along with other drives, so we're not running along these lines. This is very very puzzling...

Please let me know what comes up from your Macbook test, and also don't forget about posting a log with <kbd>Alt</kbd>-<kbd>.</kbd> in Rufus as it may provide some additional clues.

binaryBronco, 2016-08-08:
This will probably be my last entry for this issue, as I've moved on with my project and I don't think that RUFUS has much of anything to do with my problems any longer. The 16GB flash drive I mentioned previously still does not respond to any platform in any user mode or configuration. I'm pretty sure the drive is busted, but I believe it failed shortly after RUFUS completed a MAC OSX boot drive extraction from an ISO. Just end of life probably after a lot of hot writes and reads. The 4GB SD card I previously mentioned was successfully recovered using a Macbook. I'm beginning to suspect that the SD card readers in my much older Dell laptops have different controller firmware that impedes them from communicating the way that the Macbook can. Anyways, the 4GB SD card works again, which is the most important of the three that I am working with.

Finally, the 2GB SD card is still slightly operational, but very intermittently. Its only 2GB and I think the card might be damaged from age and other wear and tear, so I don't think there is much improvement that can be made to RUFUS. But I went ahead and captured some ALT-. debug mode log messages. Here is the result of a quick format request.

> 
Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE1 for write access
No drive letter was assigned...
Will use 'D:' as volume mountpoint
Could not get Disk Extents: [0x00000001] Incorrect function.
No logical drive found (unpartitioned?)
Drive does not appear to be partitioned
Analyzing existing boot records...
Drive does not have an x86 Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Deleting partitions...
Could not delete drive layout: [0x00000001] Incorrect function.
Could not reset partitions
Could not get Disk Extents: [0x00000001] Incorrect function.

Processing Hub 1:
Processing Hub 2:
Processing Hub 3:
Processing Hub 4:
Processing Hub 5:
Processing Hub 6:
  Hub[1] = '\\?\usb#root_hub#4&52a8623&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[041]: USB\VID_044E&PID_3001\5&1318C03B&0&1
Processing Hub 7:
Processing Hub 8:
Processing IDs belonging to 'USBSTOR':
  Root\LEGACY_USBSTOR\0000
Processing 'IDE' device:
  Disabled by policy
Processing 'SD' device:
  Hardware ID: 'SD\VID_27&OID_5048&PID_SD02G&REV_2.0'
Could not locate device node for 'Root\LEGACY_USBSTOR\0000'
Found card reader device 'SD Memory Card'
1 device found
Could not get Disk Extents: [0x00000001] Incorrect function.
No volume information for drive 0x81
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 1, TracksPerCylinder: 255, SectorsPerTrack: 63
Could not get layout for drive 0x81: [0x00000001] Incorrect function.
> 

And this is the result of a "check bad blocks" request.
> 

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE1 for write access
No drive letter was assigned...
Will use 'D:' as volume mountpoint
Could not get Disk Extents: [0x00000001] Incorrect function.
No logical drive found (unpartitioned?)
Drive does not appear to be partitioned
Analyzing existing boot records...
Drive does not have an x86 Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Deleting partitions...
Could not delete drive layout: [0x00000001] Incorrect function.
Could not reset partitions
Could not get Disk Extents: [0x00000001] Incorrect function.

Processing Hub 1:
Processing Hub 2:
Processing Hub 3:
Processing Hub 4:
Processing Hub 5:
Processing Hub 6:
  Hub[1] = '\\?\usb#root_hub#4&52a8623&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[041]: USB\VID_044E&PID_3001\5&1318C03B&0&1
Processing Hub 7:
Processing Hub 8:
Processing IDs belonging to 'USBSTOR':
  Root\LEGACY_USBSTOR\0000
Processing 'IDE' device:
  Disabled by policy
Processing 'SD' device:
  Hardware ID: 'SD\VID_27&OID_5048&PID_SD02G&REV_2.0'
Could not locate device node for 'Root\LEGACY_USBSTOR\0000'
Found card reader device 'SD Memory Card'
1 device found
Could not get Disk Extents: [0x00000001] Incorrect function.
No volume information for drive 0x81
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 1, TracksPerCylinder: 255, SectorsPerTrack: 63
Could not get layout for drive 0x81: [0x00000001] Incorrect function.
> 

Thanks for your prompt response, Mr. Batard.

pbatard, 2016-08-09:
Thanks for the updated logs. I'm not seeing much to explain the issue, besides possible hardware failure, especially as you're getting the `[0x00000001] Incorrect function` early, when Rufus is simply querying the drive for its partition info. I guess I'll close this issue for now, but keep my eyes open for similar reports, in case there's something I can do about it...

closed 2016-08-09 by pbatard