pbatard/rufus #807: Proposal: GUI improvements

SetTrend, 2016-08-05:
### The Rufus GUI is confusing in regard to user feedback and automatic parameter updates.

So I'd like to suggest the following GUI improvements:

- [ ] Add a read-only field displaying the currently selected ISO file path.
- [ ] Users read the GUI from top to bottom. Selecting a new ISO file alters a previously edited drive label. So I suggest to put the ISO file selection line _above_ the drive label line.
- [ ] Add a toggle button to switch from a one-line action log to a multi-line action log with vertical scroll bar. (Containing just the information from the one-line log, not the detailed log appearing after clicking the `Log` button.)

<br/>
### To sum up, the GUI should change from this:
[image]

... to this:
[image]

pbatard, 2016-08-05:
This is pretty much a duplicate of #117. __HOWEVER__, proper UI changes (that scale correctly at multiple DPIs rather that the current [atrocious hit-and-miss that the default Windows APIs provide](https://github.com/pbatard/rufus/blob/master/src/rufus.c#L1098-L1100) and which is what Rufus currently uses) are __MUCH MORE__ difficult to implement and will require __A LOT OF TIME__ to get through. So that's why the UI enhancements have been on hold for couple of years now.

They will happen. But it might very well be another year before they do.

> Add a read-only field displaying the currently selected ISO file path.

I'm already planning to do something like this (but\not\with\the\full\path\which\could\potentially\make\the\thing\way\too\long\and\which\99.99%\of\users\should\remember). Also, you can always get the currently selected ISO from the log, and it usually remains displayed for a while on the status bar. But yes, this is already something I am planning to improve.

> So I suggest to put the ISO file selection line above these input elements.

Yup. That's been planned for a long time. It's an improvement that's kind of hard to miss.

> Switch from a one-line action log to a multi-line action log with vertical scroll bar. 

No. People running on low DPI screens are already complaining that the Rufus UI is tool tall, and I can tell you that the vast majority of user are interested (or have enough attention span) for one thing only - the current last status report from Rufus. You're basically asking for a mini-log when there already exists a log. While you'll probably disagree with that assertion (to which I will apply my privilege, as the developer, of having a better "feel" for what users would appreciate to see modified in the UI, from interacting with many of them on a daily basis), I don't think this last proposed change will be useful at all for the core group of Rufus users, and, worse, it may actually confuse them. A one liner that attempts to indicate what a basic user should know about the __current__ status of the application, and doesn't bother with the past, is IMO much better.

Considering that I'm planning to integrate nr. 1 & 2. and don't see 3. as useful, I'm going to close this request.

closed 2016-08-05 by pbatard

SetTrend, 2016-08-05:
@pbatard: I have forked your solution and updated the GUI accordingly. Please have a look at my pull request.

pbatard, 2016-08-05:
Sorry, but I am not accepting UI changes, as I already have a pretty clear idea of what I want to do (which is different from your proposal, as, for instance, I plan to move the MD5/SHA checksum button next to the ISO), and I am therefore not accepting major UI changes __until__ I have started publishing my changes for Rufus 3.0.