pbatard/rufus #810: clear/partition issues on XP

pbatard, 2016-08-09:
Some XP users seem to be reporting clear/partitioning issues on XP, that apparently they did not have with 2.9. For instance, there are reports that, after formatting a drive to NTFS, trying to reformat it as FAT32 in Rufus doesn't work.

This appears to be limited to XP only (Note for XP users who think they know better: This is __exactly__ the kind of things that happens when you insist on using a __DEAD__ OS, because it's simply not possible for developers to test all the versions of Windows that have been released since XP, and therefore, bugs will accumulate for retired OSes, and may very well never be fixed &mdash; YOU HAVE BEEN WARNED!)

The logs I am seeing seem to indicate that this could be linked to Rufus clearing 0 sectors of boot records & partition tables:

```
Rufus version: 2.10.973
Windows version: Windows XP SP3 32 bit
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta3
System locale ID: 0x0409
Will use default UI locale 0x0409
Found USB device 'USB Device' (058F:6387) [GP]
1 device found
Skipping GUID volume for '\Device\Floppy0'
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 478, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00000001
Drive has an unknown Master Boot Record
Partition 1:
  Type: Small FAT16 (0x04)
  Size: 3.7 GB (3932160000 bytes)
  Start Sector: 0, Boot: No, Recognized: Yes

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE7 for write access
Will use 'L:' as volume mountpoint
Skipping GUID volume for '\Device\Floppy0'
I/O boundary checks disabled
Analyzing existing boot records...
Drive has an unknown Master Boot Record
Volume has an unknown Partition Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 0 sectors
Deleting partitions...
Partitioning (MBR)...
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (FAT32)...
Using cluster size: 4096 bytes
Quick format was selected
Creating file system...
Format completed.
Writing master boot record...
Drive has an unknown Master Boot Record
Set bootable USB partition as 0x80
Using Windows 7 MBR
Found volume GUID \\?\Volume{ad826e7f-5e24-11e6-a954-001d60c3b614}\
Skipping GUID volume for '\Device\Floppy0'
Opened drive \\?\Volume{ad826e7f-5e24-11e6-a954-001d60c3b614} for write access
Writing partition boot record...
Using Standard FAT32 partition boot record
Confirmed new volume has a primary FAT32 boot sector
Setting primary FAT32 boot sector for boot...
Confirmed new volume has a secondary FAT32 boot sector
Setting secondary FAT32 boot sector for boot...
Successfully remounted Volume{ad826e7f-5e24-11e6-a954-001d60c3b614}\ on L:\
Copying DOS files...
Patching COMMAND.COM...
Successfully wrote 'L:\COMMAND.COM' (93040 bytes)
Successfully wrote 'L:\LOCALE\DISPLAY.SYS' (17175 bytes)
Successfully wrote 'L:\LOCALE\EGA.CPI' (58870 bytes)
Successfully wrote 'L:\LOCALE\EGA2.CPI' (58870 bytes)
Successfully wrote 'L:\LOCALE\EGA3.CPI' (58753 bytes)
Patching IO.SYS...
Successfully wrote 'L:\IO.SYS' (116736 bytes)
Successfully wrote 'L:\LOCALE\KEYB.COM' (21607 bytes)
Successfully wrote 'L:\LOCALE\KEYBOARD.SYS' (34566 bytes)
Successfully wrote 'L:\LOCALE\KEYBRD2.SYS' (31942 bytes)
Successfully wrote 'L:\LOCALE\KEYBRD3.SYS' (31633 bytes)
Successfully wrote 'L:\LOCALE\KEYBRD4.SYS' (13014 bytes)
Successfully wrote 'L:\LOCALE\MODE.COM' (29239 bytes)
Successfully wrote 'L:\MSDOS.SYS' (9 bytes)
Windows KBID 0x00000809
Will use DOS keyboard 'uk' [UK-English]
Will use codepage 850 [Western-European]
Successfully wrote 'CONFIG.SYS'
Successfully wrote 'AUTOEXEC.BAT'
Finalizing, please wait...
Created: L:\autorun.inf
Created: L:\autorun.ico
Successfully remounted Volume{ad826e7f-5e24-11e6-a954-001d60c3b614}\ on L:\

Found USB device 'USB Device' (058F:6387) [GP]
Using autorun.inf label for drive L: 'BLANK'
1 device found
Skipping GUID volume for '\Device\Floppy0'
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 478, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00292776
Drive has a Windows 7 Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 3.7 GB (3931111424 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes
```

```
Rufus version: 2.10.973
Windows version: Windows XP SP3 32 bit
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta3
System locale ID: 0x0409
Will use default UI locale 0x0409
Found USB device 'USB Device' (058F:6387) [GP]
Using autorun.inf label for drive L: 'BLANK'
1 device found
Skipping GUID volume for '\Device\Floppy0'
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 478, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00292776
Drive has a Windows 7 Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 3.7 GB (3931111424 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE7 for write access
Will use 'L:' as volume mountpoint
Skipping GUID volume for '\Device\Floppy0'
I/O boundary checks disabled
Analyzing existing boot records...
Drive has a Windows 7 Master Boot Record
Drive has a FAT32 DOS Partition Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Deleting partitions...
Could not delete drive layout: [0x0000001F] A device attached to the system is not functioning.
Could not reset partitions
Skipping GUID volume for '\Device\Floppy0'
Re-mounted volume as 'L:' after error

Found USB device 'USB Device' (058F:6387) [GP]
1 device found
Skipping GUID volume for '\Device\Floppy0'
No volume information for drive 0x87
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 478, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00000001
Drive does not have an x86 Master Boot Record
Partition 1:
  Type: Small FAT16 (0x04)
  Size: 3.7 GB (3932160000 bytes)
  Start Sector: 0, Boot: No, Recognized: Yes
```

```
Analyzing existing boot records...
Drive does not have an x86 Master Boot Record
Volume does not have an x86 Partition Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 0 sectors
Deleting partitions...
Could not delete drive layout: [0x0000001F] A device attached to the system is not functioning.
Could not reset partitions
```

cmdcolin, 2016-08-22:
I think I ran into this on XP. I was aiming to upgrade to a non-dead OS using rufus to create a book disk at least :)

NSbuilder, 2016-08-22:
That's funny...

pbatard, 2016-08-29:
Replicated the issue, which is worse than I thought. It seems that as soon as we use Rufus 2.10 or later to format a BFD (Big Floppy Disk) typed UFD, we run into this error on XP, and can't even use <kbd>Alt</kbd>-</kbd>Z</kbd> (zeroing) or older versions (such as 2.9 or earlier, even if we went through zeroing first) to recover the device.

Vista (on the exact same physical machine) doesn't seem to have any issue reformatting the same UFD.

And people wonder why developers hate XP...

commit 08566fa1 (2016-08-31): [core] add half workaround for an XP issue

* This is not a complete workaround, but it will do for
  until we drop support for XP altogether.
* Closes #810
closed 2016-08-31 by pbatard