pbatard/rufus #812: Rufus 2.10 Win 10 BIOS / EUFI FAT32 doesn't boot

Sophist-UK, 2016-08-16:
If I create a BIOS / EUFI FAT32 boot stick from a freshly downloaded Win10.iso it will not boot on an old BIOS PC.

To make it work I have to write the bootsector with `bootsect nt60 e:`. And then it boots perfectly.

It looks like Rufus needs a fix to write the boot sector off the ISO.

pbatard, 2016-08-16:
What about the checklist, PLEASE?

I need a FULL LOG before I can start to help you out.

pbatard, 2016-08-16:
Also, since you ignored the checklist, and therefore didn't look at the FAQ as was advised, there's no such thing as a dual BIOS/UEFI drive for Windows ISOs (unless it's UEFI-CSM, which is really BIOS).

Since you chose to ignore the checklist, which is only there to try to help __you__ out in case you are running into a common reauest, I'm going to assume that you didn't read [this](https://github.com/pbatard/rufus/wiki/FAQ#Why_doesnt_Rufus_create_a_Windows_installation_USB_that_can_be_booted_in_both_BIOS_and_UEFI_mode) and close this issue.

closed 2016-08-16 by pbatard

Sophist-UK, 2016-08-17:
The reason I didn't follow the checklist was that I had closed Rufus and no longer had the log. So I will have to recreate this once I have finished using the USB drive for building the PC..

And there definitely is a dual BIOS / UEFI capability - if you write the ISO to DVD, then it will boot on either an old BIOS PV or a new PC under full UEFI - and the USB stick will do the same if you run the `bootsect /NT60` command on it after creating it with Rufus.

Essentially under UEFI, the PC gets the boot code from the EFI directory rather than from sector 0, but under BIOS it goes to sector 0. And sector 0 is different depending upon the O/S on the media. DOS has one type of boot sector, XP had NTLDR, Vista and later uses BCD.

So as far as I can tell, you need to read the boot sector off the ISO and write it to the USB stick. Rufus may already be trying to do this.

Anyway, if you reopen this I will post the log later today - otherwise I will open a new issue for it.

pbatard, 2016-08-17:
Seriously, did you read the entry I pointed to in the FAQ???

The entry explains WHY, even as it exists on the ISO, dual BIOS/UEFI boot capability is voluntarily disabled by Rufus for Windows images.

Both the checklist and my earlier comments pointed you to read the FAQ. I have no time for people who can't follow directions to get their questions answered.

Sophist-UK, 2016-08-17:
Apologies - I had not read that FAQ entry. Although I would personally feel that support for dual boot should be included as it is not IMO confusing, and some people (like me) are creating a USB stick to install WIndows on several machines (both old BIOS and new UEFI), however this is a judgement call and I can see where you are coming from and respect your decision. 

However, I created this USB stick with *MBR partition scheme for BIOS or UEFI computer* but on FAT32 and so according to the FAQ it should have been BIOS bootable, but it wasn't.

I will rebuild the USB and retest and post the log - and then I will try it with the Alt-E cheat and see if that helps.

Sophist-UK, 2016-08-17:
See #813 for correctly submitted issue.