pbatard/rufus #813: Rufus 2.10 BIOS/UEFI-CSM FAT32 will not boot on old BIOS PC

Sophist-UK, 2016-08-17:
<PLEASE READ THIS CAREFULLY: You *MUST* read and complete the checklist below, by placing an x into each [ ], BEFORE clicking on 'Submit new issue'. Failure to perform these steps, WHICH ARE ONLY THERE TO HELP *YOU*, will result in the issue being dismissed without warning.>

**Checklist**
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues, using keywords relevant to my problem.
- [x] I clicked the `Log` button in Rufus and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking the "bad blocks" check box in Rufus, and confirmed that my USB is not defective
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive
  - [ ] Plugging the USB into a different port
  - [ ] Running Rufus on a different computer
- [ ] If using an ISO image, I clicked on the `#` button (at the bottom of the Rufus interface), to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

**Issue description**
-----------------
I downloaded the Win10 1608 ISO (x32/x64) from Microsoft and used Rufus to create a BIOS bootable USB image using the following settings:

[image]

When I tried to boot this on a BIOS PC and got the error message `Invalid System Disk`.

I then ran `bootsect /nt60 e:` and the image was bootable and I was able to install Windows 10 from it.

Note: I was unable to find MD5/SHA checksums for the UK English x32/x64 image to check valid download, but since I was able to install fro it in the end I do not believe that it is an issue of a bad download.

**Log**
---
```
Rufus version: 2.10.973
Windows version: Windows 10 64-bit (Build 10586)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta3
System locale ID: 0x0809
Will use default UI locale 0x0809
Found USB 2.0 device 'Generic- Multi-Card USB Device' (0BDA:0138)
Could not get device number for device \\.\Y:: [0x00000001] Incorrect function.
Could not get device number for device \\.\Z:: [0x00000001] Incorrect function.
Could not get device number for device \\.\Y:: [0x00000001] Incorrect function.
Could not get device number for device \\.\Z:: [0x00000001] Incorrect function.
Found USB 3.0 device 'SanDisk Ultra Fit USB Device' (0781:5583)
Could not get device number for device \\.\Y:: [0x00000001] Incorrect function.
Could not get device number for device \\.\Z:: [0x00000001] Incorrect function.
Could not get device number for device \\.\Y:: [0x00000001] Incorrect function.
Could not get device number for device \\.\Z:: [0x00000001] Incorrect function.
2 devices found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 3880, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x001BB037
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: exFAT (0x07)
  Size: 29.7 GB (31910789120 bytes)
  Start Sector: 8192, Boot: No, Recognized: Yes
2 devices found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 7559, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x0FC3860E
Drive has a Windows 7 Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 57.9 GB (62175313920 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes
Scanning image...
ISO analysis:
[... 2420 lines cut ...]
  Size: 29.7 GB (31910789120 bytes)
  Start Sector: 8192, Boot: No, Recognized: Yes
Found USB 2.0 device 'Generic- Multi-Card USB Device' (0BDA:0138)
Could not get device number for device \\.\Y:: [0x00000001] Incorrect function.
Could not get device number for device \\.\Z:: [0x00000001] Incorrect function.
Could not get device number for device \\.\Y:: [0x00000001] Incorrect function.
Could not get device number for device \\.\Z:: [0x00000001] Incorrect function.
Found USB 3.0 device 'SanDisk Ultra Fit USB Device' (0781:5583)
Could not get device number for device \\.\Y:: [0x00000001] Incorrect function.
Could not get device number for device \\.\Z:: [0x00000001] Incorrect function.
Could not get device number for device \\.\Y:: [0x00000001] Incorrect function.
Could not get device number for device \\.\Z:: [0x00000001] Incorrect function.
2 devices found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 3880, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x001BB037
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: exFAT (0x07)
  Size: 29.7 GB (31910789120 bytes)
  Start Sector: 8192, Boot: No, Recognized: Yes
```

I will try the Alt-E checat code and report back.

Sophist-UK, 2016-08-17:
Using Alt-E for same ISO on FAT32 - doesn't boot on BIOS PC.

**Log**
```
2 devices found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 7559, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x13D950C5
Drive has a Windows 7 Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 57.9 GB (62175313920 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE2 for write access
Could not get device number for device \\.\Y:: [0x00000001] Incorrect function.
Could not get device number for device \\.\Z:: [0x00000001] Incorrect function.
Will use 'E:' as volume mountpoint
I/O boundary checks disabled
Analyzing existing boot records...
Drive has a Windows 7 Master Boot Record
Volume has an unknown FAT16 or FAT32 Partition Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 2048 sectors
Deleting partitions...
Partitioning (MBR)...
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (Large FAT32)...
Opened drive \\?\Volume{4f881f81-cf9a-11e4-93d2-4c72b96f3c36} for write access
I/O boundary checks disabled
Size : 57.9 GB 121436160 sectors
Cluster size 32768 bytes, 512 Bytes Per Sector
Volume ID is 1d0e:2d70
32 Reserved Sectors, 14821 Sectors per FAT, 2 FATs
1896976 Total clusters
1896975 Free Clusters
Clearing out 29738 sectors for reserved sectors, FATs and root cluster...
Initializing reserved sectors and FATs...
FAT #0 sector at address: 32
FAT #1 sector at address: 14853
Writing partition boot record...
Using Standard FAT32 partition boot record
Confirmed new volume has a primary FAT32 boot sector
Setting primary FAT32 boot sector for boot...
Confirmed new volume has a secondary FAT32 boot sector
Setting secondary FAT32 boot sector for boot...
Setting Label (This may take while)...
Format completed.
Writing master boot record...
Drive has a Zeroed Master Boot Record
Partition is already FAT32 LBA...
Set bootable USB partition as 0x80
Using Windows 7 MBR
Found volume GUID \\?\Volume{4f881f81-cf9a-11e4-93d2-4c72b96f3c36}\
Opened drive \\?\Volume{4f881f81-cf9a-11e4-93d2-4c72b96f3c36} for write access
Writing partition boot record...
Using Standard FAT32 partition boot record
Confirmed new volume has a primary FAT32 boot sector
Setting primary FAT32 boot sector for boot...
Confirmed new volume has a secondary FAT32 boot sector
Setting secondary FAT32 boot sector for boot...
Successfully remounted Volume{4f881f81-cf9a-11e4-93d2-4c72b96f3c36}\ on E:\
Copying ISO files...
Extracting files...
Image is an UDF image
Extracting: E:\boot\en-gb\bootsect.exe.mui (16 KB)
[... 2201 lines cut ...]
Partition 1:
  Type: exFAT (0x07)
  Size: 29.7 GB (31910789120 bytes)
  Start Sector: 8192, Boot: No, Recognized: Yes
Found USB 2.0 device 'Generic- Multi-Card USB Device' (0BDA:0138)
Could not get device number for device \\.\Y:: [0x00000001] Incorrect function.
Could not get device number for device \\.\Z:: [0x00000001] Incorrect function.
Could not get device number for device \\.\Y:: [0x00000001] Incorrect function.
Could not get device number for device \\.\Z:: [0x00000001] Incorrect function.
Found USB 3.0 device 'SanDisk Ultra Fit USB Device' (0781:5583)
Could not get device number for device \\.\Y:: [0x00000001] Incorrect function.
Could not get device number for device \\.\Z:: [0x00000001] Incorrect function.
Could not get device number for device \\.\Y:: [0x00000001] Incorrect function.
Could not get device number for device \\.\Z:: [0x00000001] Incorrect function.
2 devices found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 3880, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x001BB037
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: exFAT (0x07)
  Size: 29.7 GB (31910789120 bytes)
  Start Sector: 8192, Boot: No, Recognized: Yes
```

pbatard, 2016-08-17:
> I downloaded the Win10 1608 ISO (x32/x64) from Microsoft

I hate to call bullshit on this, but I'm not aware of Microsoft ever publishing a retail 32+64 bit ISO...
The [official Windows 10 ISO download website](https://www.microsoft.com/en-us/software-download/windows10ISO/) doesn't offer any, and, unless I'm mistaken, the MSDN (which __does__ list every single ISO that ever got released by Microsoft) [only offers 32 __OR__ 34-bit ISOs, but not both](https://msdn.microsoft.com/en-us/subscriptions/downloads/#searchTerm=&ProductFamilyId=657&Languages=en&FileExtensions=.iso&PageSize=100&PageIndex=0&FileId=0).

Also, Microsoft calls the latest Windows refresh 1607, not 1608. So, unless you renamed it, that `Windows10-32-64-1608.iso` listed in your log doesn't seem like something you can actually download from Microsoft.

So, while I would very much like to be proved wrong, there's already something fishy about the ISO you use, which looks like something custom...

Of course, you do realize that the only Windows ISOs I can officially support are the ones that are officially released by Microsoft.

At this stage, I am therefore going to assume that you are using a __custom__ ISO, which, as I said, is unsupported. My guess is that it might have be based on WinPE, and that you are therefore running into a bug which I [recently fixed](https://github.com/pbatard/rufus/commit/6d3b1c856fd6783b3b59f94cde6d67304083c91f) that could allow people to use FAT32, where they should only have been able to use NTFS, but that only affected custom ISOs.

In that case, I have a __very early__ [test version of Rufus 2.11](https://rufus.akeo.ie/testing/rufus-2.11_TEST1.exe) that addresses this issue and that I's like you to try. Do you see the same issue there?

Oh, and of course, if you have a link for where you downloaded your ISO, I'd like to have it so that I can test it myself.

pbatard, 2016-08-17:
PS: An SHA-1 of your ISO (which is something I also ask for in the checklist, but that you chose to skip) would also provide invaluable information about the nature of your ISO. There's a reason why I ask for all these things you know &mdash; it helps validate what people report...

Sophist-UK, 2016-08-17:
To answer your nit-pick, it is the 1607 version, but it was downloaded in August hence 1608 (though I agree I should have named it 1607). It is not a custom ISO - it is absolutely as downloaded from Microsoft. 

To make the dual ISO, you use the [Media Creation Tool](https://www.microsoft.com/en-gb/software-download/windows10). This allows you to download an ISO - which can be Pro/Home or just Home, and x32, x64 or both and to select a default locale too. So there are 2 x 3 x 38 locales i.e. 228 different possible ISOs. AFAIK, MS has not published the checksums for these.

The checksums for my ISO are:

* MD5: 0a39b609fb38ac17a76207ae158b5a33
* SHA1: 1f54554a88bab50765469868f659f408cd9fff3c
* SHA256: 57f36106e9617c6f3ea52eda45cdb289bdbdc6756c8ff94259ec55a669a21b42

Please feel free to download the English (UK) version of the Pro x32/x64 ISO (only a 6GB download) and calculate your own checksums to confirm that they are the same as mine.

I ran Rufus 2.10 again using NTFS and it boots fine. I will test the alpha 2.11 version tomorrow as soon as I have finished work.

Sophist-UK, 2016-08-17:
Actually I have tested it now:

[image]

I get the following message box:

[image]

The log is empty.

Is this what you intended?

Sophist-UK, 2016-08-17:
P.S. I noticed that the User Rufus MBR checkbox is not set by default in the test version. But checking it makes no difference.

Sophist-UK, 2016-08-17:
P.P.S. Happy to try more test versions if it will help.

pbatard, 2016-08-17:
Ah OK. All signs seemed to point out to a custom All-In-One ISO (I do get a lot of requests using these), but I stand corrected. I wasn't aware the Media Creation Tool offered the 32+64 bit option, and I tend to avoid using it for creating ISOs, as Microsoft managed to create `install.wim` when using the Media Creation Tool that can't be opened with their own WIM extraction APIs...

I guess I'll download one these 32+64 bit ISOs and play with it a little bit.

> Is this what you intended?

Yup. Rufus is meant to force NTFS for Windows ISOs whenever you create a BIOS bootable drive (only pure UEFI should allow the selection of FAT32). So FAT32 should not have been selected, and won't be in the next version of Rufus.

Now, if you pick NTFS, it should boot fine from BIOS (and hopefully from UEFI if you use <kbd>Alt</kbd>-<kbd>E</kbd>, though that's one thing I'm planning to test with the MCT ISO).

Sophist-UK, 2016-08-17:
It occurs to me that you may believe that this is a custom WinPE image because it doesn't have `\sources` but instead has `\x32\sources` and `\x64\sources` instead. But I promise it is as downloaded from MS. (I have 1 x 32-bit PC and 5 x 64-bit PCs to re-image, hence wanting a dual architecture USB stick.

Also, happy to upload my image to a web server you can download it from if you want a copy of the exact ISO image I am using.

pbatard, 2016-08-17:
Nah, I do believe you and I happily admit that I was wrong to assume you were using something that didn't originate from Microsoft. I tested the MCT and I confirm that it offers a 32+64 bit option.

I'm downloading my own MCT ISO as we speak, which I fully expect to behave exactly as yours, so, while I appreciate the offer, you don't have to upload anything, thanks!

Sophist-UK, 2016-08-17:
Don't worry. In your place I would probably be highly sceptical and unsympathetic myself. I can quite imagine just how many people you get telling you that they are using a MS ISO when in fact they are using a custom one. 

Looking at a few bits of your code I can see that you do a lot of stuff which is dependent on the type of ISO - its obviously not as simple (as I naively thought) as creating the partition, copying the boot sector(s) from the ISO and then copying the files over in the same structure (starting with the files in the root directory before recursing - so as to put the boot files in low sectors). If it were that simple, then you might be able to support any ISO - but once you get into ISO-format-specific activities, then you can obviously only support a limited set of ISO formats that you code for and can test.

Anyway, back to you.

Sophist-UK, 2016-08-17:
> Rufus is meant to force NTFS for Windows ISOs whenever you create a BIOS bootable drive (only pure UEFI should allow the selection of FAT32). So FAT32 should not have been selectable, and won't be in the next version of Rufus.
> 
> Now, if you pick NTFS, it should boot fine from BIOS (and hopefully from UEFI if you use `Alt-E`, though that's one thing I'm planning to test with the MCT 

I am confused. I was under the impression that you cannot boot from an NTFS partition on a pure UEFI machine because UEFI only understands FAT. So if you want an image which is bootable on both BIOS and UEFI machines - doesn't it need to be FAT32 (or have a FAT32 UEFI-loader partition). If you then want to be able to read (and write) the USB partitions from Windows as well as boot from it, doesn't that raise further issues because Windows cannot handle multiple partitions on a USB stick? (I tried it once and it wouldn't allow it or couldn't see the subsequent partitions).

Is there a technical reason that you can't have a FAT32 file system for joint BIOS / UEFI images? (I haven't tried the working BIOS/UEFI FAT32 image I created (using `bootsect /NT60`) on a UEFI machine (I only have one of these - and it is is reasonably constant use). but I can't see why it wouldn't work.

pbatard, 2016-08-17:
> I was under the impression that you cannot boot from an NTFS partition on a pure UEFI machine because UEFI only understands FAT

There are [some ways to make UEFI boot NTFS](https://github.com/pbatard/uefi-ntfs)... :smile:
In short, create an UEFI bootable FAT32 partition at the end of your drive that loads an UEFI NTFS driver, and then hands over boot to the UEFI bootloader on the NTFS partition, and transparently booting an NTFS partition from UEFI is not a problem. 

> Is there a technical reason that you can't have a FAT32 file system for joint BIOS / UEFI images?

No. If I wanted to, it would be fairly easy to add it in Rufus. But I have to cater to people who have no clue whether their computer booted in BIOS or UEFI mode, and will get annoyed if they spent a couple hours installing Windows on their UEFI computer, only to discover they actually installed it in BIOS/CSM mode.

If I were to allow FAT32, and keep this feature, I'd have to start __removing__ files from the original ISO to prevent UEFI boot when only BIOS/CSM should be enabled, and one of the major design goals of Rufus is to keep the USB content as close as possible to the ISO. It's much easier to force BIOS boot for Windows to only use NTFS as, if you don't install the UEFI:NTFS partition, it will (usually) not be bootable at all from UEFI.

Sophist-UK, 2016-08-17:
Ok - I have read your FAQ again plus the readme from your UEFI:NTFS repo, and I think I understand. 

And my hat off to you for creating UEFI:NTFS - which seems a very elegant solution.

So, whilst there may be no theoretical reason that you can't have a FAT32 dual BIOS / UEFI solution, there is no need for it if you have UEFI:NTFS - and not allowing FAT32 in this situation means that you don't have to support it (which is a very good reason not to allow it). In which case, perhaps better to allow only NTFS for Windows NT onwards. 

In which case my only comment is that rather than giving the above error box when someone tries to create a FAT32 image, it would be better to grey out the option in the File System drop-down and provide a tooltip explaining why it is disabled.

pbatard, 2016-08-17:
Oh, and another very important thing that I need to mention here, is the 4GB file limit for FAT32 system. Microsoft is know to actually have released an official ISO that contained a >4GB `install.wim`.
I guess I could always disable the ability to select FAT32 then (which is what I do for pure UEFI when there's a >4GB file on a Windows ISO), but then I'd have to tell users why they suddenly can no longer select FAT32. Much easier to only allow NTFS and not have users worry about this detail.

pbatard, 2016-08-17:
> it would be better to grey out the option in the File System drop-down and provide a tooltip explaining why it is disabled.

Yeah, I actually thought about doing that, but greyed out options can be confusing, and more difficult to implement/maintain than a messagebox. Heck, I did manage to screw up the conditions that prompt the "use a different FS with this type of ISO" to appear, and didn't notice it for some time, so I'm not that optimistic about my ability to ensure that greyed out options stay greyed out. Also, since I'm hoping that in the long run I can add some testing to check for breakage, message boxes are much easier to convert to a PASS/FAIL test compared to checking if a line is read-only in a dropdown. I'm still a very long way to add automated testing though...

Sophist-UK, 2016-08-17:
Ok - busy creating an NTFS compressed image to try (Alt-E/S/N) - and yes I know compression is unsupported. Writing this is also MUCH (an order of magnitude?) slower. And it seems to have saved only c. 3% of disk space, so hardly worth it.

(In passing I note that the NTFS volume is created with the allow file contents to be indexable flag set - I have no idea whether this is right or not, but you might want to think about whether it should be off.)

Sophist-UK, 2016-08-17:
For the greyed out options, you might want to consider implementing a lookup table which defines what is greyed out and provides the text (or I18N key) for a tooltip explanation.

This would make the code generic and easier to test, and make it much simpler to make changes to what is greyed out when. You could extend this to include the `Use Rufus MBR` checkbox as well.

You have 3 choices for partition scheme, and four choices for file system and 8 choices for bootable disk type so the table would have 96 rows.

Finally on a UI note, can I lobby to have the Alt-E cheat moved to a check box in Advanced options.

pbatard, 2016-08-17:
> And it seems to have saved only c. 3% of disk space, so hardly worth it.

I guess you know why compression is only available as a cheat mode. Somebody really wanted it, and it was relatively easy to add, so I did, but Microsoft really should take some lessons on compression (their VHD compression algorithms also suck)...

> In passing I note that the NTFS volume is created with the allow file contents to be indexable flag set

As far as I know, that's what the default Windows format tool does (with no option to disable it), and I try to follow what the Microsoft format tool does where possible. If it's easy to do, I'll see if I can add that feature as a cheat mode (I have now made a note for that in #799).

pbatard, 2016-08-17:
As to adding a new item to the Advanced options, I'm not planning to do UI changes before I can get through some UI redesign (#117), that I've been trying to get started for more than a few years now, [but that I may very well be the only one to understand the need for](https://github.com/pbatard/rufus/pull/808#issuecomment-238010792))...

Sophist-UK, 2016-08-18:
Actually, reflecting on the Alt-E UI comment above whilst in the shower this morning, I think it makes more sense to have two items in the Partition Scheme dropdown - one for `BIOS and UEFI (BIOS compatibility mode)` (which is IMO better than the not-well-known CSM acronym) and one for `BIOS and native UEFI`.

And I can see why you are hesitant about starting on a UI rework.

pbatard, 2016-08-27:
Okay, I played with the MCT dual 32/64 bit ISO, and I think it behaves as expected when converted to bootable USB with the upcoming Rufus 2.11, so I will close this issue.

As to the suggestion of using "BIOS Compatibility", that's something I played with when I added the options, but the problem here is width. This dropdown is usually the one's that the largest in all translations (NB: it may look like there's tons of space to do it on modern Windows, but you need to consider XP), and using "BIOS Compatibility" means that I would have to expand the width of the application.

Doesn't sound like a big deal, huh? Well it is when your translation framework relies on manually resizing all the elements according to the application width &rarr; doing so would be a major undertaking as I'd need to update the size 15 controls for almost every single one of the 36 translations.
Having adaptive width according to the largest text element, and therefore doing away with the need to add resize statements in the translation is also one of my goals with the new UI framework (and then I might review the text placed in the dropdown).

However, and much to my chagrin, many people seem to be oblivious to it, there are very helpful tooltips for every single of the options from that dropdown, which will tell you precisely which one you should select. Also, IMO, people need to become familiar with the CSM acronym, just as they have become familiar with UEFI or Secure Boot, especially as many UEFI firmwares will use CSM and not "BIOS Compatibility" in their options. So trying to shy away from knowing the proper terminology of the stuff you use is not a good bet, and will confuse users further.

Finally, you'll be happy to hear that I've added a new cheat mode (<kbd>Alt</kbd>-<kbd>Q</kbd>) to disable file indexing on format, for the files systems that support it (NTFS, UDF and ReFS). This will be part of Rufus 2.11

closed 2016-08-27 by pbatard