pbatard/rufus #816: [INTERNAL] List of 2.12 small enhancements and fixes

pbatard, 2016-08-26:


NSbuilder, 2016-10-05:
Can I enable BIOS+UEFI mode in Rufus for normal windows installation or it is only working with windows to go?

And how can I choose the boot method for each installation ( choose BIOS/UEFI for installation )?

NSbuilder, 2016-10-06:
>  Allow selection of FAT32 for Windows ISOs when dual BIOS+UEFI is active

I'm trying to make a dual BIOS + UEFI windows 10 installation media.
I want an option to install windows in bios mode and in native uefi mode - how can I do that?
I need to toggle dual BIOS + UEFI but after that - Do I need FAT32 or it will work on NTFS?

pbatard, 2016-10-06:
Please don't comment on INTERNAL issues. I will delete your post in 1 hour.

NSbuilder, 2016-10-06:
Sorry... I will create an issue.

ljporljp, 2016-10-17:
sounds good ！

closed 2017-01-27 by pbatard