pbatard/rufus #817: Error: iso image extraction failure

celtek, 2016-08-26:
<PLEASE READ THIS CAREFULLY: You *MUST* read and complete the checklist below, by placing an x into each [ ], BEFORE clicking on 'Submit new issue'. Failure to perform these steps, WHICH ARE ONLY THERE TO HELP *YOU*, will result in the issue being dismissed without warning.>

Checklist
---------
- [ ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ ] I performed a search in the issue tracker for similar issues, using keywords relevant to my problem.
- [ ] I clicked the `Log` button in Rufus and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking the "bad blocks" check box in Rufus, and confirmed that my USB is not defective
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive
  - [ ] Plugging the USB into a different port
  - [ ] Running Rufus on a different computer
- [ ] If using an ISO image, I clicked on the `#` button (at the bottom of the Rufus interface), to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Trying to install Ubuntu Server onto the USB, tried two different USB sticks in USB 3.0 and USB 2.0 ports, on two different machines. Continue to get the same error in the log file:

Extracting: E:\.disk\base_installable (0 bytes)
  Unable to create file: [0x00000057] The parameter is incorrect.
Unable to create E:\syslinux.cfg - booting from USB will not work

Log
---

Rufus version: 2.10.973
Windows version: Windows 10 64-bit (Build 10586)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta3
System locale ID: 0x0409
Will use default UI locale 0x0409
Found USB 3.0 device 'Kingston DataTraveler 3.0 USB Device' (0951:1666)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 3774, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x988FB237
Drive has a Windows 2000/XP/2003 Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 28.9 GB (31039488000 bytes)
  Start Sector: 8064, Boot: Yes, Recognized: Yes
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/isolinux/isolinux.cfg' for Syslinux
  Detected Syslinux version: 6.03/20151222 (from '/isolinux/isolinux.bin')
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'Ubuntu-Server 16.04.1 LTS amd64'
  Size: 699400192 bytes
  Has a >64 chars filename
  Uses: Syslinux/Isolinux v6.03
  Uses: EFI
  Note: This ISO uses symbolic links, which will not be replicated due to file system limitations.
  Because of this, some features from this image may not work...
Using image: ubuntu-16.04.1-server-amd64.iso
Will reuse 'ldlinux.sys' and 'ldlinux.bss' from './rufus_files/rufus_files/syslinux-6.03/20151222/' for Syslinux installation

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE1 for write access
Will use 'E:' as volume mountpoint
I/O boundary checks disabled
Analyzing existing boot records...
Drive has a Windows 2000/XP/2003 Master Boot Record
[... 132 lines cut ...]
Using existing './rufus_files/syslinux-6.03/20151222/ldlinux.sys' ✓
Using existing './rufus_files/syslinux-6.03/20151222/ldlinux.bss' ✗
Successfully wrote 'ldlinux.sys'
Successfully wrote Syslinux boot record
Successfully remounted Volume{facc1612-46a7-11e6-a96e-64006a8b5c60}\ on E:\
Copying ISO files...
Extracting files...
Image is an ISO9660 image
This image will be extracted using Rock Ridge extensions (if present)
Extracting: E:\.disk\base_installable (0 bytes)
  Unable to create file: [0x00000057] The parameter is incorrect.
Unable to create E:\syslinux.cfg - booting from USB will not work

Found USB 3.0 device 'Kingston DataTraveler 3.0 USB Device' (0951:1666)
1 device found
No volume information for drive 0x81
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 3774, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x338D1E95
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 28.9 GB (31042568192 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes

pbatard, 2016-08-26:
Thanks for the detailed report!

Let me download the same ISO and see if I can replicate the issue. My first guess would be that there could be a Windows API limitation when trying to create a `.something\` folder (I vaguely remember having some issues doing that in the past), but I'll need to test.

celtek, 2016-08-26:
_Thanks_ for the quick response. Eagerly awaiting your results.

pbatard, 2016-08-26:
Hmmm, I'm not seeing this issue when I test with the same ISO (confirmed that the SHA match with yours) on Windows 10 x64, and the `.disk` directory is created without problem...

Are you positive the bad blocks check didn't report an issue and that you have been trying with more than one flash drive?

From your log, you've been getting different errors when trying to extract the same file (`Unable to create file: [0x000003ED] The volume does not contain a recognized file system.` and `Unable to create file: [0x00000057] The parameter is incorrect.`), which would seem to point to a flash drive issue. Especially, for Windows to report that "The volume does not contain a recognized file system", is like telltale sign of connectivity or flash drive problems...

So I really would like to make sure that:
- Your bad blocks check reports success
- You aren't using an extension cable to connect your flash drive. Especially with USB 3.0 drives, poor quality extension cables can wreck havoc on your ability to write to the device.
- You tried another device than the 'Kingston DataTraveler 3.0 USB Device' (if you have one)

celtek, 2016-08-26:
I'm going to retest for bad blocks, and I have numerous of these drives unfortunately I don't want to format all their contents.

Will be back with results.

celtek, 2016-08-26:
Finished block checking -- negative. 

Will reuse 'ldlinux.sys' and 'ldlinux.bss' from './rufus_files/rufus_files/syslinux-6.03/20151222/' for Syslinux installation

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE1 for write access
Will use 'E:' as volume mountpoint
I/O boundary checks disabled
Analyzing existing boot records...
Drive has a Syslinux Master Boot Record
Volume has an unknown Partition Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Deleting partitions...
Partitioning (MBR)...
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (FAT32)...
Using cluster size: 16384 bytes
Quick format was selected
Creating file system...
Format completed.
Writing master boot record...
Drive has a Zeroed Master Boot Record
Partition is already FAT32 LBA...
Set bootable USB partition as 0x80
Using Syslinux MBR
Found volume GUID \\?\Volume{facc1612-46a7-11e6-a96e-64006a8b5c60}\
Installing Syslinux 6.03...
Opened drive \\?\Volume{facc1612-46a7-11e6-a96e-64006a8b5c60} for write access
Using existing './rufus_files/syslinux-6.03/20151222/ldlinux.sys' ✓
Using existing './rufus_files/syslinux-6.03/20151222/ldlinux.bss' ✗
Successfully wrote 'ldlinux.sys'
Successfully wrote Syslinux boot record
Successfully remounted Volume{facc1612-46a7-11e6-a96e-64006a8b5c60}\ on E:\
Copying ISO files...
Extracting files...
Image is an ISO9660 image
This image will be extracted using Rock Ridge extensions (if present)
Extracting: E:\.disk\base_installable (0 bytes)
  Unable to create file: [0x00000057] The parameter is incorrect.
Unable to create E:\syslinux.cfg - booting from USB will not work

Found USB 3.0 device 'Kingston DataTraveler 3.0 USB Device' (0951:1666)
1 device found
No volume information for drive 0x81
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 3774, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x3397955B
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 28.9 GB (31042568192 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes
Will reuse 'ldlinux.sys' and 'ldlinux.bss' from './rufus_files/rufus_files/syslinux-6.03/20151222/' for Syslinux installation

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE1 for write access
Will use 'E:' as volume mountpoint
Analyzing existing boot records...
Drive has a Syslinux Master Boot Record
Volume has an unknown Partition Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Deleting partitions...
Bad Blocks: Checking from block 0 to 60632063
Bad Blocks: Using offset 0 for fake device check
Bad Blocks: Writing test pattern 0xAA
Bad Blocks: Reading and comparing
Bad Blocks: Check completed, 0 bad blocks found. (0/0/0 errors)
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Partitioning (MBR)...
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (FAT32)...
Using cluster size: 16384 bytes
Quick format was selected
Creating file system...
Format completed.
Writing master boot record...
Drive has a Zeroed Master Boot Record
Partition is already FAT32 LBA...
Set bootable USB partition as 0x80
Using Syslinux MBR
Found volume GUID \\?\Volume{facc1612-46a7-11e6-a96e-64006a8b5c60}\
Installing Syslinux 6.03...
Opened drive \\?\Volume{facc1612-46a7-11e6-a96e-64006a8b5c60} for write access
Using existing './rufus_files/syslinux-6.03/20151222/ldlinux.sys' ✓
Using existing './rufus_files/syslinux-6.03/20151222/ldlinux.bss' ✗
Successfully wrote 'ldlinux.sys'
Successfully wrote Syslinux boot record
Successfully remounted Volume{facc1612-46a7-11e6-a96e-64006a8b5c60}\ on E:\
Copying ISO files...
Extracting files...
Image is an ISO9660 image
This image will be extracted using Rock Ridge extensions (if present)
Extracting: E:\.disk\base_installable (0 bytes)
  Unable to create file: [0x00000057] The parameter is incorrect.
Unable to create E:\syslinux.cfg - booting from USB will not work

Found USB 3.0 device 'Kingston DataTraveler 3.0 USB Device' (0951:1666)
1 device found
No volume information for drive 0x81
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 3774, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x33DDE2FA
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 28.9 GB (31042568192 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes

Gonna try another USB of the same type.

celtek, 2016-08-26:
Rufus version: 2.10.973
Windows version: Windows 10 64-bit (Build 10586)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta3
System locale ID: 0x0409
Will use default UI locale 0x0409
Found USB 3.0 device 'ADATA USB Flash Drive USB Device' (125F:DC1A)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 1886, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x3388B1F4
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 14.5 GB (15517876224 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/isolinux/isolinux.cfg' for Syslinux
  Detected Syslinux version: 6.03/20151222 (from '/isolinux/isolinux.bin')
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'Ubuntu-Server 16.04.1 LTS amd64'
  Size: 699400192 bytes
  Has a >64 chars filename
  Uses: Syslinux/Isolinux v6.03
  Uses: EFI
  Note: This ISO uses symbolic links, which will not be replicated due to file system limitations.
  Because of this, some features from this image may not work...
Using image: ubuntu-16.04.1-server-amd64.iso
Will reuse 'ldlinux.sys' and 'ldlinux.bss' from './rufus_files/rufus_files/syslinux-6.03/20151222/' for Syslinux installation

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE1 for write access
Will use 'F:' as volume mountpoint
I/O boundary checks disabled
Analyzing existing boot records...
Drive has a Syslinux Master Boot Record
Volume has an unknown Partition Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Deleting partitions...
Partitioning (MBR)...
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (FAT32)...
Using cluster size: 8192 bytes
Quick format was selected
Creating file system...
Format completed.
Writing master boot record...
Drive has a Zeroed Master Boot Record
Partition is already FAT32 LBA...
Set bootable USB partition as 0x80
Using Syslinux MBR
Found volume GUID \\?\Volume{a3cf3605-3b6f-11e6-a96d-64006a8b5c60}\
Installing Syslinux 6.03...
Opened drive \\?\Volume{a3cf3605-3b6f-11e6-a96d-64006a8b5c60} for write access
Using existing './rufus_files/syslinux-6.03/20151222/ldlinux.sys' ✓
Using existing './rufus_files/syslinux-6.03/20151222/ldlinux.bss' ✗
Successfully wrote 'ldlinux.sys'
Successfully wrote Syslinux boot record
Successfully remounted Volume{a3cf3605-3b6f-11e6-a96d-64006a8b5c60}\ on F:\
Copying ISO files...
Extracting files...
Image is an ISO9660 image
This image will be extracted using Rock Ridge extensions (if present)
Extracting: F:\.disk\base_installable (0 bytes)
  Unable to create file: [0x00000057] The parameter is incorrect.
Unable to create F:\syslinux.cfg - booting from USB will not work

Found USB 3.0 device 'ADATA USB Flash Drive USB Device' (125F:DC1A)
1 device found
No volume information for drive 0x81
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 1886, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x33E30CB1
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 14.5 GB (15517876224 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes

pbatard, 2016-08-26:
If both the ADATA and Kingston fail, then I'd say there might be something wrong about your system.

Have you tried plugging these drives on an USB 2.0 port?
And, as asked from the checklist, have you tried from a different computer?
Are you using USB 3.0 accelerator applications (I know that ASUS and probably other motherboard manufacturers have their own version of such a thing, which of course can create their own issues &mdash; If you do, please try to disable them).

Considering that I can't replicate your issue at all, and that many people (~2 million downloads/month) are using Rufus without running into this problem, the best explanation is that there is an environmental component that is linked to your specific system, and I'm afraid that this is not something I can troubleshoot remotely. Also as I mentioned, the fact that the error isn't always the same also points to something that would be specific to a flaky system, rather than an application limitation or bug...

A few more things you want to try:
1. Rufus will prompt you whether you want to write in ISO image or DD image mode after you press Start. have you tried DD image mode?
2. Try <kbd>Alt</kbd>-<kbd>.</kbd> in Rufus, which enables USB enumeration debug mode, and post a log with your drive plugged (no need to start a format operation). Maybe it will provide details about your system that could explain things.
3. Try to create a FreeDOS drive and see if that works.

pbatard, 2016-08-26:
One more thing I'm noticing. Your `ldlinux.bss` seems to be corrupted.
```
Using existing './rufus_files/syslinux-6.03/20151222/ldlinux.bss' ✗
```
It should have a check mark (✓) like `ldlinux.sys` but doesn't.

Please delete the `rufus_files\` folder on your computer (which you can also do with <kbd>Alt</kbd>-<kbd>X</kbd>) and try again until you see this in the log:
```
Using existing './rufus_files/syslinux-6.03/20151222/ldlinux.sys' ✓
Using existing './rufus_files/syslinux-6.03/20151222/ldlinux.bss' ✓
```
If your system cannot properly download files, or corrupts them, I can obviously make no guarantees about Rufus' ability to run on it. How stable is your system, really?

celtek, 2016-08-26:
FreeDOS works:

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE1 for write access
Will use 'F:' as volume mountpoint
I/O boundary checks disabled
Analyzing existing boot records...
Drive has a Windows 7 Master Boot Record
Drive has a FAT32 FreeDOS Partition Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Deleting partitions...
Partitioning (MBR)...
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (FAT32)...
Using cluster size: 8192 bytes
Quick format was selected
Creating file system...
Format completed.
Writing master boot record...
Drive has a Zeroed Master Boot Record
Partition is already FAT32 LBA...
Set bootable USB partition as 0x80
Using Windows 7 MBR
Found volume GUID \\?\Volume{a3cf3605-3b6f-11e6-a96d-64006a8b5c60}\
Opened drive \\?\Volume{a3cf3605-3b6f-11e6-a96d-64006a8b5c60} for write access
Writing partition boot record...
Using FreeDOS FAT32 partition boot record
Confirmed new volume has a primary FAT32 boot sector
Setting primary FAT32 boot sector for boot...
Confirmed new volume has a secondary FAT32 boot sector
Setting secondary FAT32 boot sector for boot...
Successfully remounted Volume{a3cf3605-3b6f-11e6-a96d-64006a8b5c60}\ on F:\
Copying DOS files...
Successfully wrote 'F:\COMMAND.COM' (65814 bytes)
Successfully wrote 'F:\KERNEL.SYS' (45494 bytes)
Successfully wrote 'F:\LOCALE\DISPLAY.EXE' (3648 bytes)
Successfully wrote 'F:\LOCALE\KEYB.EXE' (11446 bytes)
Successfully wrote 'F:\LOCALE\MODE.COM' (16254 bytes)
Successfully wrote 'F:\LOCALE\KEYBOARD.SYS' (40360 bytes)
Successfully wrote 'F:\LOCALE\KEYBRD2.SYS' (29750 bytes)
Successfully wrote 'F:\LOCALE\KEYBRD3.SYS' (32178 bytes)
Successfully wrote 'F:\LOCALE\KEYBRD4.SYS' (13105 bytes)
Successfully wrote 'F:\LOCALE\ega.cpx' (6464 bytes)
Successfully wrote 'F:\LOCALE\ega2.cpx' (7059 bytes)
Successfully wrote 'F:\LOCALE\ega3.cpx' (5469 bytes)
Successfully wrote 'F:\LOCALE\ega4.cpx' (4431 bytes)
Successfully wrote 'F:\LOCALE\ega5.cpx' (7217 bytes)
Successfully wrote 'F:\LOCALE\ega6.cpx' (7409 bytes)
Successfully wrote 'F:\LOCALE\ega7.cpx' (5387 bytes)
Successfully wrote 'F:\LOCALE\ega8.cpx' (6973 bytes)
Successfully wrote 'F:\LOCALE\ega9.cpx' (5785 bytes)
Successfully wrote 'F:\LOCALE\ega10.cpx' (5543 bytes)
Successfully wrote 'F:\LOCALE\ega11.cpx' (7228 bytes)
Successfully wrote 'F:\LOCALE\ega12.cpx' (8119 bytes)
Successfully wrote 'F:\LOCALE\ega13.cpx' (6281 bytes)
Successfully wrote 'F:\LOCALE\ega14.cpx' (7758 bytes)
Successfully wrote 'F:\LOCALE\ega15.cpx' (6458 bytes)
Successfully wrote 'F:\LOCALE\ega16.cpx' (7793 bytes)
Successfully wrote 'F:\LOCALE\ega17.cpx' (8929 bytes)
Successfully wrote 'F:\LOCALE\ega18.cpx' (5158 bytes)
Windows KBID 0x00000409
Will use DOS keyboard 'us' [US-English]
Will use codepage 437 [US-English]
Successfully wrote 'AUTOEXEC.BAT'
Finalizing, please wait...
Created: F:\autorun.inf
Created: F:\autorun.ico
Successfully remounted Volume{a3cf3605-3b6f-11e6-a96d-64006a8b5c60}\ on F:\

Found USB 3.0 device 'ADATA USB Flash Drive USB Device' (125F:DC1A)
Using autorun.inf label for drive F: '16GB'
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 1886, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x33F4CCE2
Drive has a Windows 7 Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 14.5 GB (15517876224 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes

This system serves as a NFS share server and is stable.

I've tried the operation on multiple systems.

I attempted to write in DD mode and it worked.

pbatard, 2016-08-26:
> This system serves as a NFS share server and is stable.

Well, I can tell you that foe some reason, your system did corrupt the `ldlinux.bss` download, which may explain your issue. I know that, in some case, some poorly designed security solution may replace content in files that aren't downloaded through a browser (they would insert something like "You are not authorized to download this content" in lieu of the payload), but then I would expect `ldlinux.sys` to also be flagged as invalid.

If you're still seeing this issue on this specific machine, I would really encourage you to see what happens when you try on a different PC. And also, please go through the cleaning of the `rufus_files\` directory.

pbatard, 2016-08-29:
Hi,

Any update on this? Did you try deleting your `rufus_files` directory and can you tell us what happens if you have the double checkmark for the `ldlinux.###` files?

celtek, 2016-08-29:
Hi,

Deleting the rufus_files directory didn't result in success. Ultimately, I had to create the drive using the DD Mode in order to properly create the boot disk. 

Thanks for your assistance.

pbatard, 2016-08-29:
Can you provide a log after you deleted the `rufus_files` directory? I'd like to see what it says, because it really looks to be like the corrupted `ldlinux.bss` is the kind of thing that would cause the error you observed. Also, if you still see an `✗` for `ldlinux.bss`, could you please e-mail me your file to: pete@akeo.ie?

pbatard, 2016-09-06:
No update &rarr; Closed.

closed 2016-09-06 by pbatard