pbatard/rufus #821: Windows 10 boot disk not work

speller, 2016-09-02:
Hi,

I have astrange issue. Laptop Fujitsu CH55 can't start Windows 10 setup from UEFI:NTFS usb stick. The screen shows Rufus's 64 bit loader starts successfully, but next it can't find the 64 bit efi loader at the windows setup partition (/efi/boot/bootx64.efi). I tried few Windows 10 distributions and all of them (original and customized) contain only 32 bit loader (bootia32.efi). I tried to copy bootia32.efi to bootx64.efi but it just failed to start. 

If I format the usb stick to MBR the setup starts successfully but not in EFI mode. I need to start the installer in EFI mode.

I tried to delete bootx64.efi from the UEFI:NTFS partition of the usb stick to allow boot 32bit loader but laptop just can't load from the usb stick at all, and shows "no bootable device" error.

Plese advice me how to fix this issue?

Thank you.

pbatard, 2016-09-08:
> Rufus's 64 bit loader starts successfully,

Then your UEFI firmware __IS__ 64-bit, and you are __wasting your time__ trying to use a 32-bit Windows 10 ISO. So the first thing you need to do is get a 64-bit version of your Windows ISO. You will __NEVER__ be able to boot in UEFI mode unless you use a 64-bit Windows.

> I tried few Windows 10 distributions and all of them (original and customized) contain only 32 bit loader

You must not have looked very hard. Official 64-bit ISOs of Windows 10 are [very easy to find](https://www.microsoft.com/en-us/software-download/windows10ISO/) so please do yourself a favour and download one of those.

> I can't use FAT32 because windows setup contain 5GB wim file.

If you have a 5GB `wim` file, then you aren't using an official Microsoft ISO. Again, please do yourself a favour, and get an official 64-bit ISO.

As I should have answered your query, I will now close this issue.

closed 2016-09-08 by pbatard

speller, 2016-09-08:
Why not to show an alert that selected ISO can not boot *before* creating USB?

pbatard, 2016-09-08:
Because it's impossible to tell if your target system's UEFI is 32 or 64 bit. You do understand that Rufus can not guess the type of machine you will be trying to boot your USB on. If you were trying to boot an ARM computer, would you be complaining that the USB you created from an x86 ISO does not boot? Of course not, because __YOU__ are the only one who can know the type of CPU being used in the computer you are trying to boot, and therefore select the appropriate ISO to use.

Well, this is exactly the same here. You selected a 32-bit ISO, which makes the implicit promise that the UEFI system you are trying to boot is 32-bit. And if your UEFI system was really 32-bit, it would have booted alright. but of course, if __YOU__ get your target system wrong, and try to use an ISO that doesn't match its type, there's absolutely nothing Rufus can do about that.

speller, 2016-09-09:
My opinion as of usual user is that if your utility makes bootable USB and it accepts an ISO it either should make working USB or should show a clear information about possible problems. I didn't saw any warnings about your UEFI bootloader that it is 64-bit.
Is it too hard to read ISO label and if it contain "x86" show a warning? Or show warning dialog box when user selects uefi:ntfs for a first time with a message describing bit capacities of the bootloader and the image.
Anyway, I don't mean your program is bad or not working. It is great. But I just wan it to be more clear for users who do not have any experiense of using it. Windows 10 is (and will be, I suppose) most common usage of your program and if you explain somehow about possible failures then it will save a lot of time of your users.
Thank you.

pbatard, 2016-09-09:
> Is it too hard to read ISO label and if it contain "x86" show a warning?

Yes, because then it needs to be translated into many languages, and this is a new path of code that needs to be tested for breakage. And if I were to add a warning about 32 vs 64 bit, then I would also need to add a warning about BIOS vs UEFI and try to explain what the differences are (along with CSM), as this is a much bigger issue with non tech-savvy users than 32 vs 64 bit. So now I've got to add tons of popups that try to hold users by the hand with every single operation they undertake, because the part where you would like to see some guidance is just one of __MANY__ people have been requested to see guidance for. Well, that's not very realistic.

Rufus does expect users to have a minimum level of knowledge to accomplish their task. So you are supposed to know what UEFI, BIOS and CSM are. You are also supposed to know about Secure Boot and that you may have to disable it to be able to boot your USB. You are supposed to know that there might be a Legacy option in your BIOS that is tied to CSM. And you are supposed to know that a 64 bit UEFI is not gonna boot a 32 bit ISO in UEFI mode. It is NOT Rufus' job to explain all of the above to you, because if you aren't aware of these things, then you are most likely going to run into other issues that you won't be able to solve while booting, and you shouldn't expect to be held by the hand every step of the way.

Then again, you are most than welcome to write a usage guide for Rufus, that explains all of this (and get it translated into 36 languages), because quite frankly, since Rufus is only one of many things I'm involved with, I'm not planning to do that, and it would be a lot better, IMO, to have such a guide written from the standpoint of a regular user than from a specialist. If you are actually serious about saving time for Rufus users (after all, why should I be the only one?), you could help many users by doing just so, especially as you can closely related to the pitfall they are likely to have.