pbatard/rufus #823: Windows To Go features

NSbuilder, 2016-09-04:
Hi.
I downloaded the image "Windows.iso" from Media Creation Tool.
It is a x86 + x64 esd Windows 10 installation file.

I downloaded WinToUSB and choose the Windows.iso image.
Then I opened rufus and choose the Windows.iso image.

Image: http://i.imgur.com/QBor7jW.png

The Windows.iso image contain several windows editions ( pro, home... ) and the edition that will be installed will be determined by the product key you enter, if you skip and don't enter the product key - you will be able to choose edition ).

Anyway, WinToUSB let me see the editions in the ISO and let me choose the edition I want to make Windows to Go with.
But Rufus doesn't even show the Windows to go option... ( see image ).

1. Why can't rufus create Windows to go from this image?
2. Can rufus create a vhd/vhdx based Windows to go ( like WinToUSB does )?

NSbuilder, 2016-09-04:
WinToUSB is a software used to make windows to go drives.

pbatard, 2016-09-04:
> 1. Why can't rufus create Windows to go from this image ( and let me choose the edition )?

Because Rufus follows the __official process__ for Windows To Go creation from Microsoft, which requires an `install.wim` and a FIXED drive. Considering that anybody can easily download the official retail Windows 10 ISOs (from [here](https://www.microsoft.com/en-us/software-download/windows10ISO/)), which are fully compatible with the Windows To Go official installation method (and therefore the one used by Rufus), I don't consider that not being able to use the Media Creation Tool ISOs for Windows To Go as an issue that needs addressing. You are always better off using the retail ISOs anyway.

2. 2. Can rufus create a vhd/vhdx based Windows to go ( like WinToUSB does )?

No, and I'm not planning to add that feature, because since it deviates from the official Windows To Go requirements from Microsoft, there's no telling if/when Microsoft will break this method with future versions of Windows. The use of REMOVABLE drives (or VHDs) for Windows To Go is NOT supported by Microsoft, and therefore there is no guarantee that you aren't going to run into issues if you try to use VHD + removable. I therefore suggest that you keep using WinToUSB if you want to use this method of creating Windows To Go drives, as I am not interested in supporting unofficial methods.

To understand my position, you have to understand the following: If any Windows To Go user faces a Windows issue and tries to contact Microsoft to get it solved, they __will__ get turned down as soon as Microsoft finds out that the Windows To Go drive was not created by following their official recommendations. This means that the person who created the tool who produced the _unofficial_ Windows To Go drive would have to provide support instead. Maybe the WinToUSB people are happy to do that, but I personally don't think it helps users in the long run, so I am not. This is even more true as it's very easy to get an ISO that will be compatible with official Windows To Go installation if you want a Windows To Go drive.

As there is nothing to address in this issue, I will be closing it.

closed 2016-09-04 by pbatard

NSbuilder, 2016-09-04:
I understand.
But if so - why do we need a windows to go feature on Rufus? this is also unofficial... the official method is using windows to go component from control panel.
And Rufus allow installation of windows to go on removable drives and on drives that are smaller than 32G and the official windows to go doesnt...

pbatard, 2016-09-04:
> why do we need a windows to go feature on Rufus?

Until Windows 10 the control panel component was only available for the Enterprise version. If you had Windows 8.1 Pro, you did not have the ability to easily create To Go drives. Besides, I'm not even sure the Home version of Windows 10 has it. Plus the purpose of Rufus is to create bootable USB drives, and Windows To Go is one type of such drives, so that's a relevant feature to have.

> And Rufus allow installation of windows to go on removable drives and on drives that are smaller than 32G and the official windows to go doesnt...

I don't remember the official guide (which is what I followed) having such a requirement. Now, if you can point to official MS documentation that limits the drive size, I'll make sure to apply that restriction in Rufus.