pbatard/rufus #825: C2001,newline in constant

ljporljp, 2016-09-05:
My page code is 936(Chinese Simplified),i really cloned from git, and this problem confuses me long time!
Is the file-encoding problem?

ljporljp, 2016-09-05:
First thanks!

pbatard, 2016-09-05:
I already replied to you earlier. Please don't double post.

closed 2016-09-05 by pbatard

ljporljp, 2016-09-06:
i checked git global config ,--> autocrlf = false , safecrlf = true, and i checkout the code again ,recompile and the warning still exist . I searched many things about this. Change the Adanced Save Mode to utf-8 with bom or Unicode is ok ,but then the dos_locale.c will show the warning C4819  and some others show the warning C4566, Is that Vs 's bug ?

ljporljp, 2016-09-06:
I guess it is the vc compiler 's problem, because when i choose the rufus (Clang).sln that uses the clang compiler just don't tell me the error and warnings.