pbatard/rufus #826: Windows 10 Anniversary update

devastatorius, 2016-09-05:
Hello.
I tried to write manjaro linux image with the rufus with no luck. After Requesting Disk Access, Rufus just pop error message "Could not open media".
I also tried to use different usb flash drives and connected to different ports.
Rufus was working properly before Anniversary patch. After i updated, then stopped working.
 
Log
---
```
Rufus version: 2.10.973
Windows version: Windows 10 64-bit (Build 14393)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta3
System locale ID: 0x0409
Will use default UI locale 0x0409
Found USB 3.0 device 'UFD 3.0 Silicon-Power32G USB Device' (090C:1000)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 3916, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x0000E5AC
Drive has an unknown Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 30.0 GB (32211206144 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/isolinux/isolinux.cfg' for Syslinux
  Detected Syslinux version: 6.03 (from '/isolinux/isolinux.bin')
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'MJRO1608'
  Size: 1561657344 bytes
  Uses: Syslinux/Isolinux v6.03
  Uses: EFI
Using image: manjaro-xfce-16.08-x86_64.iso
Will reuse 'ldlinux.sys' and 'ldlinux.bss' from './rufus_files/rufus_files/syslinux-6.03/' for Syslinux installation

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE1 for write access
Could not get exclusive access to device \\.\PHYSICALDRIVE1: [0x00000005] Access is denied.

Found USB 3.0 device 'UFD 3.0 Silicon-Power32G USB Device' (090C:1000)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 3916, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x0000E5AC
Drive has an unknown Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 30.0 GB (32211206144 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes

pbatard, 2016-09-05:
> Rufus was working properly before Anniversary patch. After i updated, then stopped working.

Many people (including myself) are using the anniversary update, so I can tell you, this is __NOT__ the issue, and there is no incompatibility that I know of, between Rufus and any version of Windows 10.

Usually `[0x00000005] Access is denied.` means that there is __another__ application trying to access your drive, and I can guarantee that the problem does not originate from Rufus. Did you, for instance, install one of the security solutions [listed at the end of the FAQ](https://github.com/pbatard/rufus/wiki/FAQ#Software), that are known to try to keep accessing USB media even when they don't need to?

Or are you __ABSOLUTELY POSITIVE__ that you don't have any application open that is using something from your USB? For instance, have you tried rebooting your computer to see if you still had the issue.

Note that if you are positive that now application (that you know of, coz I can __guarantee__ there does exist an external application trying to access your USB if you see that error), you can try the <kbd>Alt</kbd>-<kbd>,</kbd> cheat mode in Rufus, to try to disable exclusive locking, but you may still face issues after that.

devastatorius, 2016-09-05:
Thanks for fast reply. Alt-, trick was solved problem.

closed 2016-09-05 by pbatard