pbatard/rufus #830: How to compile on *.deb

overtube, 2016-09-14:
<PLEASE READ THIS CAREFULLY: You *MUST* read and complete the checklist below, by placing an x into each [ ], BEFORE clicking on 'Submit new issue'. Failure to perform these steps, WHICH ARE ONLY THERE TO HELP *YOU*, will result in the issue being dismissed without warning.>

Checklist
---------
- [ ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ ] I performed a search in the issue tracker for similar issues, using keywords relevant to my problem.
- [ ] I clicked the `Log` button in Rufus and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking the "bad blocks" check box in Rufus, and confirmed that my USB is not defective
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive
  - [ ] Plugging the USB into a different port
  - [ ] Running Rufus on a different computer
- [ ] If using an ISO image, I clicked on the `#` button (at the bottom of the Rufus interface), to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<Please describe your issue here>

Log
---
```
<FULL LOG>
```

overtube, 2016-09-14:
Ho to compile this source on debian

pbatard, 2016-09-14:
1. You are __NOT__ going to be able to produce a Linux application from the Rufus sources if that's what you're trying to accomplish. Rufus is __WINDOWS ONLY__, and its source is not portable.
2. Therefore, the only way you are going to be able to use the sources on Debian is if you __cross-compile__ a Windows application. You can do that by installing [MinGW](http://www.mingw.org/), but you have to realize that it will still produce a Windows executable, not a Linux one.
3. [I have no plans to port Rufus to Linux](https://github.com/pbatard/rufus/wiki/FAQ#Do_you_plan_to_port_Rufus_to_LinuxOS_XSome_other_OS).

closed 2016-09-14 by pbatard

overtube, 2016-09-18:
thank's for your answer