pbatard/rufus #831: Error: Could not copy files to target drive.

GreccoTM, 2016-09-15:
Issue description
-----------------

I'm trying to create MSDOS pendrive bootable, but get error on copy phase.
Follow the full log

[rufus.log.txt](https://github.com/pbatard/rufus/files/475242/rufus.log.txt)

```

pbatard, 2016-09-17:
This is probably an issue with your security solution. `Access is denied` means an EXTERNAL application is preventing access. You'll have to figure out what it is on your own. Also, next time, please follow the guidelines, as it makes people wore willing to help you.

closed 2016-09-17 by pbatard

danieljethro, 2017-04-02:
hello im also having the smae problem 
it won't let me reformat. pls help. its 125gb flashdrive

\\?\Volume{23f7863e-e66e-11e6-82dc-c80aa950ad1f}\ was already mounted as G:\
Re-mounted volume as 'G:' after error

Found USB 2.0 device 'Generic- SD/MMC USB Device' (0EC2:2015)
1 device found
No volume information for drive 0x81
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 16317, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: SFD (Super Floppy Disk) or Unpartitioned
Partition 0:
  Type: FAT32 LBA (0x00)
  Size: 0 bytes (0 bytes)
  Start Sector: 0, Boot: Yes

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE1 for write access
Will use 'G:' as volume mountpoint
Analyzing existing boot records...
Drive has a Windows 7 Master Boot Record
read_sectors: Read error [0x00000057] The parameter is incorrect.
  Read: 0, Expected: 512
  StartSector: 0x00000000, nSectors: 0x1, SectorSize: 0x200
Volume does not have an x86 Partition Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 2176 sectors
Deleting partitions...
Partitioning (MBR)...
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (Large FAT32)...
Opened drive \\?\Volume{23f7863e-e66e-11e6-82dc-c80aa950ad1f} for write access
Size : 125.0 GB 262141952 sectors
Cluster size 32768 bytes, 512 Bytes Per Sector
Volume ID is 1515:3010
32 Reserved Sectors, 31992 Sectors per FAT, 2 FATs
4094967 Total clusters
4094966 Free Clusters
Clearing out 64080 sectors for reserved sectors, FATs and root cluster...
Initializing reserved sectors and FATs...
FAT #0 sector at address: 32
FAT #1 sector at address: 32024
Writing partition boot record...
Using FreeDOS FAT32 partition boot record
Confirmed new volume has a primary FAT32 boot sector
Setting primary FAT32 boot sector for boot...
Confirmed new volume has a secondary FAT32 boot sector
Setting secondary FAT32 boot sector for boot...
Setting Label (This may take while)...
Could not set label: [0x00000057] The parameter is incorrect.
Format completed.
Writing master boot record...
Drive has an unknown Master Boot Record
Warning: converting a non FAT32 partition to FAT32 LBA: FS type=0x00
Set bootable USB partition as 0x80
Using Windows 7 MBR
Found volume GUID \\?\Volume{23f7863e-e66e-11e6-82dc-c80aa950ad1f}\
Warning: Failed to refresh label: [0x00000057] The parameter is incorrect.
Opened drive \\?\Volume{23f7863e-e66e-11e6-82dc-c80aa950ad1f} for write access
Writing partition boot record...
Using FreeDOS FAT32 partition boot record
read_sectors: Read error [0x00000057] The parameter is incorrect.
  Read: 0, Expected: 512
  StartSector: 0x00000000, nSectors: 0x1, SectorSize: 0x200
New volume does not have a primary FAT32 boot sector - aborting
Successfully remounted Volume{23f7863e-e66e-11e6-82dc-c80aa950ad1f}\ on G:\
Copying DOS files...
Unable to create file 'G:\COMMAND.COM': [0x00000057] The parameter is incorrect.
.
\\?\Volume{23f7863e-e66e-11e6-82dc-c80aa950ad1f}\ was already mounted as G:\
Re-mounted volume as 'G:' after error

Found USB 2.0 device 'Generic- SD/MMC USB Device' (0EC2:2015)
1 device found
No volume information for drive 0x81
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 16317, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: SFD (Super Floppy Disk) or Unpartitioned
Partition 0:
  Type: FAT32 LBA (0x00)
  Size: 0 bytes (0 bytes)
  Start Sector: 0, Boot: Yes
Saved C:\Users\Jeth\Desktop\rufus.log