pbatard/rufus #833: Windows 7 Ultimate UEFI boot problem

itzzmeshashi, 2016-09-17:
<PLEASE READ THIS CAREFULLY: You *MUST* read and complete the checklist below, by placing an x into each [ ], BEFORE clicking on 'Submit new issue'. Failure to perform these steps, WHICH ARE ONLY THERE TO HELP *YOU*, will result in the issue being dismissed without warning.>

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues, using keywords relevant to my problem.
- [x] I clicked the `Log` button in Rufus and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking the "bad blocks" check box in Rufus, and confirmed that my USB is not defective
- [x] I also tried one or more of the following:
  - [x] Using a different USB drive
  - [x] Plugging the USB into a different port
  - [ ] Running Rufus on a different computer
- [ ] If using an ISO image, I clicked on the `#` button (at the bottom of the Rufus interface), to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
I have been using this windows image for quite a long time in MBR partition mode, now I thought to switch out to UEFI+GPT. I used rufus 2.11 version to boot the iso into a sandisk cruzer 8GB drive.
I disabled the secure boot and fast boot options in my MSI Z170A PRO motherboad, but i keep getting the following error:
**[FAIL] Could not locate '\efi\boot\bootx64.efi': [14] Not Found**
**Press any key to exit**

[image]

Thanks

Log
---
```
PFA
[rufus_18_9_16.txt](https://github.com/pbatard/rufus/files/478382/rufus_18_9_16.txt)

```

pbatard, 2016-09-17:
Your ISO is for 32-bit only, as indicated by the log:
```
Extracting: E:\efi\boot\bootia32.efi (1.3 MB)
```
(You'll noticed that there is no `Extracting: E:\efi\boot\bootx64.efi` line), which means that it will NEVER be bootable on an UEFI x64 system, which is what you have (as indicated by the `*** UEFI:NTFS (x64) ***` from your screenshot).

If you want to install Windows 7 in UEFI mode on this specific machine, you will need to find a Windows 7 64-bit ISO, as the one you have will not work.

closed 2016-09-17 by pbatard

itzzmeshashi, 2016-09-17:
It is a AIO. When i used it in non-UEFI mode it showed up all x86 and x64 versions and i used to select ultimate x64 which was a 64 bit OS.

pbatard, 2016-09-17:
Non-UEFI mode supports 32-bit, since it doesn't use any of the `bootia32.efi`/`bootx64.efi` bootloaders (it uses a special "BIOS emulation mode"). UEFI mode doesn't. There's nothing you can do, if you want to boot in UEFI mode, but find an ISO that includes an `efi\boot\boox64.efi` file, a.k.a. a proper 64-bit ISO.

pbatard, 2016-09-17:
Also, please understand that AIO ISOs are non official, therefore the people who build them tend to screw up, which is exactly why you're running into this issue, and why I could just as well tell you "Please use an official retail Microsoft ISO".