pbatard/rufus #834: Rufus 2.11 does not open drop-down boxes, nor finds devices

nodevice, 2016-09-23:
<PLEASE READ THIS CAREFULLY: You *MUST* read and complete the checklist below, by placing an x into each [ ], BEFORE clicking on 'Submit new issue'. Failure to perform these steps, WHICH ARE ONLY THERE TO HELP *YOU*, will result in the issue being dismissed without warning.>

Checklist
---------
- [x ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ x] I performed a search in the issue tracker for similar issues, using keywords relevant to my problem.
- [x ] I clicked the `Log` button in Rufus and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking the "bad blocks" check box in Rufus, and confirmed that my USB is not defective
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive
  - [ ] Plugging the USB into a different port
  - [ ] Running Rufus on a different computer
- [ ] If using an ISO image, I clicked on the `#` button (at the bottom of the Rufus interface), to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<Please describe your issue here>
Rufus 2.11 worked when I first downloaded it on my 64bit Samsung laptop. But i wasn't ready to burn iso image to a flash DVD disc. Now that i want to, the Rufus software won't find devices, nor give any selections in the other drop-down boxes. The hash button doesnt work. The "list USB devices" ticks the box, says enabled, but still says no devices.
Log
---Rufus version: 2.11.995
Windows version: Windows 7 64-bit
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta3
System locale ID: 0x0409
Will use default UI locale 0x0409
Found USB 2.0 device 'Generic STORAGE DEVICE USB Device' (05E3:0745)
Device eliminated because it appears to contain no media
Found USB 2.0 device 'Generic- Multi-Card USB Device' (0BDA:0158)
Device eliminated because it appears to contain no media
0 devices found
Found USB 2.0 device 'Generic STORAGE DEVICE USB Device' (05E3:0745)
Device eliminated because it appears to contain no media
Found USB 2.0 device 'Generic- Multi-Card USB Device' (0BDA:0158)
Device eliminated because it appears to contain no media
0 devices found
```
<FULL LOG>
```Rufus version: 2.11.995
Windows version: Windows 7 64-bit
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta3
System locale ID: 0x0409
Will use default UI locale 0x0409
Found USB 2.0 device 'Generic STORAGE DEVICE USB Device' (05E3:0745)
Device eliminated because it appears to contain no media
Found USB 2.0 device 'Generic- Multi-Card USB Device' (0BDA:0158)
Device eliminated because it appears to contain no media
0 devices found
Found USB 2.0 device 'Generic STORAGE DEVICE USB Device' (05E3:0745)
Device eliminated because it appears to contain no media
Found USB 2.0 device 'Generic- Multi-Card USB Device' (0BDA:0158)
Device eliminated because it appears to contain no media
0 devices found

pbatard, 2016-09-23:
`Device eliminated because it appears to contain no media` is usually indicative of a dead flash drive.

Unfortunately, flash memory doesn't last forever (it has a limited number of R/W cycles before it fails), and usually when that happens, the USB controller on the flash drive will be unable to access the flash and report that there is no media.

It looks to me like you will have to purchase a new flash drive...

nodevice, 2016-09-23:
Actually I am using a DVD writer, not the USB port. And the DVD writer still does burn discs! So why wont Rufus 2.11 show my DVD writer? And also, why did it show my devices yesterday, (both DVD and USB), but today it shows nothing, not even in any of the other boxes like the picture shows, like file system, or cluster size etc.

nodevice, 2016-09-23:
Could my antivirus have disabled something. I have AVG free edition, but I see nothing in the virus vault, nor in the history events

pbatard, 2016-09-23:
> So why wont Rufus 2.11 show my DVD writer? 

Because Rufus is __NOT__ a CD/DVD burning application. It only created USB flash drives. If you want to burn an ISO to DVD, you need to use a different application.

nodevice, 2016-09-23:
oh that sure explains it!  So I need to use the USB

pbatard, 2016-09-23:
Yup. But I'm a bit concerned about the following device:

```
Found USB 2.0 device 'Generic STORAGE DEVICE USB Device' (05E3:0745)
Device eliminated because it appears to contain no media
```

If this is a flash drive that you plugged into your computer, then it looks like it has an issue.

On the other hand, if you don't have anything plugged, then you can probably ignore that message.

nodevice, 2016-09-23:
Its a Transcend USB 2 stick, which seems to respond to the computer when I plug it in the pilot light (blue) comes on... I thought I might check "device manager" because I dont see the USB port listed under "computer"

pbatard, 2016-09-23:
The light probably only indicates if there are USB transfers (which there can be even if the flash is dead). But once again `Device eliminated because it appears to contain no media` almost certainly indicates a dead flash drive...

nodevice, 2016-09-23:
No im not transfering anything. I only pushed it into the socket. Unfortunately this silly Dell USB ports are both located one above the other and right next to the power cable socket, and the USB stick is rather a wide one, so I cant try it in the upper USB socket!! I cant unplug the Dell and use the battery so that the cable isnt in the way of the USB socket, because my battery is old/dead!  I am trying now to see if the USB needs a driver update.

pbatard, 2016-09-23:
> No im not transfering anything.

Devices can fail even if you're not transferring anything.
And there are no driver updates (that I know of) for mass storage media such as USB flash drives. The driver comes from Windows.

nodevice, 2016-09-23:
Sorry I first told you it was a Samsung, but Im using the old Dell to try to reboot and repair in safe mode my Samsung laptop which has a BSOD Unmountable Boot Volume. I dont want to lose my data on it!! I have downloaded the Windows 8.1 iso file but now i want to create a flash bootable disc. You are very kind to answer my questions. :)

nodevice, 2016-09-23:
Would you please be able to answer this one question : if i am able to get a flash disc made of Windows 8.1, will I be able to repair a ble screen Unmountable boot Volume? Do you know?

pbatard, 2016-09-23:
Possibly. But you'll have to ask on Windows users forums, as this issue tracker is only for Rufus.

nodevice, 2016-09-23:
I checked Windows to update the USB drivers and it reported : Driver software (WPD FileSystem Volume Driver) for your device is up to date> I checked both the :E and the :F drives, so now I am fully confused why the USB doesn't work

pbatard, 2016-09-23:
I'm afraid USB flash drives do fail for no specific reasons, precisely because, as I indicated, flash memory has a limited number of read/write cycles before failure. If you are unlucky, one of the flash memory cells used by the USB controller on your flash drive will fail early, leaving you with a dead drive.

There's no specific reason for this to happen (it can happen whether you are copying files in Windows or using Rufus or doing absolutely nothing), but when it does, there's not much you can do bu replace the drive.

nodevice, 2016-09-23:
The reason I found Rufus in the first place was looking for a solution to fix the Blue screen issue! so, frankly< I would think that Rufus should be able to back their claim that this software to create a bootable disk will open my computer in the Windows image which it will use, right??

pbatard, 2016-09-23:
Not if you USB is dead. Rufus cannot magically revive a dead USB.
Also, there are plenty of boot methods used by various ISOs, and not all of them are compatible with Rufus. Most Windows __official__ ISOs should work, but there's no guarantee of anything.

nodevice, 2016-09-23:
Sorry, you misunderstood me. I was referring to your reply to whether specifically the RUFUS rebootable disc will open and repair Windows > You said " Possibly. But you'll have to ask on Windows users forums, as this issue tracker is only for Rufus."

nodevice, 2016-09-23:
I mean, if it won't then there's no point in my expense to repair the USB

pbatard, 2016-09-23:
I stand by what I say. Rufus will allow you to create a bootable Windows installation USB. After that, whether you can use that installation USB to repair your computer is not something that is related to Rufus, and you will have to ask somewhere else.

NSbuilder, 2016-09-23:
I think you can do it: http://www.intowindows.com/wp-content/uploads/2010/03/repairwindows7fromusbflashdrive.jpg

nodevice, 2016-09-23:
Thank you for that link> I really appreciate it.  I guess that"s the extent of my needing answers, and I thank you very much for your replies. Good day and go well, and safely. Bye :)

NSbuilder, 2016-09-23:
You can use Rufus to create a bootable windows usb flash drive and use it to recover your pc.
But if your usb flash drive is dead, you can't use it.

nodevice, 2016-09-23:
I figured that out! :) I will need to try and use some software which will create it with my DVD writer>

nodevice, 2016-09-23:
Thank you, NSbuilder, for that link. I will try and find one for windows 8.1

NSbuilder, 2016-09-23:
This was a picture...

NSbuilder, 2016-09-23:
It was a link to a picture... but if it helped you, good luck!

nodevice, 2016-09-23:
NSbuilder, i copied part of the link and changed 7 to 8.1 and found a whole lot of useful information regarding backing-up etc. Great help, and thanks!! Cheers!

closed 2016-10-03 by pbatard