pbatard/rufus #835: vesamenu upgrade not compatible with HP Service Pack iso, otherwise fine.

nidarian, 2016-09-28:
<PLEASE READ THIS CAREFULLY: You *MUST* read and complete the checklist below, by placing an x into each [ ], BEFORE clicking on 'Submit new issue'. Failure to perform these steps, WHICH ARE ONLY THERE TO HELP *YOU*, will result in the issue being dismissed without warning.>

Checklist
---------
- [x ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x ] I performed a search in the issue tracker for similar issues, using keywords relevant to my problem.
- [0 ] I clicked the `Log` button in Rufus and copy/pasted the log into the line that says `<FULL LOG>` below.
- [0 ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [x ] I ran a bad blocks check, by clicking the "bad blocks" check box in Rufus, and confirmed that my USB is not defective
- [x ] I also tried one or more of the following:
  - [x] Using a different USB drive
  - [x ] Plugging the USB into a different port
  - [x ] Running Rufus on a different computer
- [x ] If using an ISO image, I clicked on the `#` button (at the bottom of the Rufus interface), to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
vesamenu upgrade not compatible with HP Service Pack iso, otherwise fine.

No log file produced, serveral USB sticks not bootable if vesamenu file is upgraded from internet.

pbatard, 2016-09-28:
Please __ALWAYS__ include the log!!!

There's a __VERY GOOD REASON__ I'm asking for it, __ALWAYS__, and that's because, if you had copied/paste it:
1. I would know which version of Windows/Rufus you are using.
2. I would know if the download of `vesamenu.c32` was successful or not.
3. I would know the exact name of the 'HP Service Pack iso' you are using, giving me a chance to locate it... If you don't tell me where I can find that ISO, how do you expect me to fix this potential issue..

Because you didn't follow the very clear rules, I am very close to just ignore your report and only answer requests from people who are actually capable of following the rules. There is nothing more infuriating than not being able to __help__ someone, because they decided that a set of rules they don't understand, but which are CRUCIAL to provide help, should not apply to them. Don't do that!!

nidarian, 2016-09-28:
I sincerely apologize for my lazy behavior. I will reproduce my efforts and submits logs.

nidarian, 2016-09-28:
Here is a log of a working instance of the ISO recorded to USB. No changes made.
[rufus211_not_upgrading_Vesamenu.log.txt](https://github.com/pbatard/rufus/files/498158/rufus211_not_upgrading_Vesamenu.log.txt)

pbatard, 2016-09-28:
Thanks. According to the log, the `vesamenu.c32` was not replaced. You should have gotten something like:
```
...
Extracting: E:\system\squashfs_sys.img (218.1 MB)
Extracting: E:\system\vesamenu.c32 (138.2 KB)
  Replaced with local version ✓
Extracting: E:\system\vmlinuz (4.7 MB)
...
```
But you only got:
```
...
Extracting: E:\system\squashfs_sys.img (218.1 MB)
Extracting: E:\system\vesamenu.c32 (138.2 KB)
Extracting: E:\system\vmlinuz (4.7 MB)
...
```

I'm currently downloading the ISO (bit of a [pain in the ass to get](https://www.reddit.com/r/sysadmin/comments/4da6k9/new_hpe_service_pack_for_proliant_spp_version/), even though it's just a service pack) and will test it to see if there's anything I need to do in Rufus.

pbatard, 2016-09-28:
I think your download failed. Here's what I get on my machine:
```
Rufus version: 2.11.995
Windows version: Windows 10 64-bit (Build 14393)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta3
System locale ID: 0x0809
Will use default UI locale 0x0809
Found USB 3.0 device 'SanDisk Extreme USB Device' (0781:5580)
Using autorun.inf label for drive F: 'ESXI-5.1.0-799733-STANDARD'
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 1946, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x012E25DA
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 14.9 GB (16012894208 bytes)
  Start Sector: 2048, Boot: Yes
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/usb/syslinux.cfg' for Syslinux
  Detected Syslinux version: 3.75 (from '/system/isolinux.bin')
Disk image analysis:
  Image does not have an x86 Master Boot Record
ISO label: 'SPP2016040'
  Size: 6167623680 bytes
  Has a >64 chars filename
  Uses: Syslinux/Isolinux v3.75
    With an old vesamenu.c32
  Uses: EFI
Using image: 864794_001_spp-2016.04.0-SPP2016040.2016_0317.20.iso
Downloading 'vesamenu.c32' from http://rufus.akeo.ie/files/syslinux-4.07/vesamenu.c32
File length: 154128 bytes
Successfully downloaded 'vesamenu.c32'

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE5 for write access
Will use 'F:' as volume mountpoint
I/O boundary checks disabled
Analyzing existing boot records...
Drive has a Syslinux Master Boot Record
Volume has an unknown Partition Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 2048 sectors
Deleting partitions...
Partitioning (MBR)...
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (Large FAT32)...
Opened drive \\?\Volume{624c96e4-6ddd-11e6-9109-fcaa14e7bd8e} for write access
I/O boundary checks disabled
Size : 14.9 GB 31275184 sectors
Cluster size 8192 bytes, 512 Bytes Per Sector
Volume ID is 19fd:3c48
32 Reserved Sectors, 15257 Sectors per FAT, 2 FATs
1952789 Total clusters
1952788 Free Clusters
Clearing out 30562 sectors for reserved sectors, FATs and root cluster...
Initializing reserved sectors and FATs...
FAT #0 sector at address: 32
FAT #1 sector at address: 15289
Writing partition boot record...
Using Standard FAT32 partition boot record
Confirmed new volume has a primary FAT32 boot sector
Setting primary FAT32 boot sector for boot...
Confirmed new volume has a secondary FAT32 boot sector
[... 103 lines cut ...]
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 1946, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x01B25D15
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 14.9 GB (16012894208 bytes)
  Start Sector: 2048, Boot: Yes
Saved C:\Users\nil\Desktop\rufus.log
```

Are you trying to run Rufus from a read-only file system by any chance (e.g. CD-ROM or network drive)? Rufus needs to have the ability to create a `rufus_files` folder in the directory it is running from. If you don't have the rights to do that, then you can't expect all functionality to work as expected.

If not, can you please compress your `rufus_files` directory and send it to pete@akeo.ie? I'd like to see what your directory contains.

Finally, can you try deleting the directory if it exists and try creating your USB again.
You should see something like this in the log before the format process:
```
Downloading 'vesamenu.c32' from http://rufus.akeo.ie/files/syslinux-4.07/vesamenu.c32
File length: 154128 bytes
Successfully downloaded 'vesamenu.c32'
```
This only happens for the first download, or if the `rufus_files` directory does not exist. You may have gotten an error then, which we'll probably see if you try downloading the file again.

nidarian, 2016-09-28:
Sorry about getting the iso file problems. I could have sent the file to you. 

Write/Read filesystem: Yes
Rufus rights to write: yes
compressed rufus_files sent to email.
rufus211 log with downloaded new vesamenu file and still not working.
[rufus211_downloaded_new_vesamenu_file.log.txt](https://github.com/pbatard/rufus/files/499286/rufus211_downloaded_new_vesamenu_file.log.txt)

I have been using rufus2.9 until recently. But now 2.9 did not work either
[rufus2.9.934_not_upgrading_vesamenu.log.txt](https://github.com/pbatard/rufus/files/499324/rufus2.9.934_not_upgrading_vesamenu.log.txt)

pbatard, 2016-09-28:
Strange, your new log does show:
```
Extracting: E:\system\vesamenu.c32 (138.2 KB)
  Replaced with local version ✓
Extracting: E:\system\vmlinuz (4.7 MB)
Extracting: E:\usb\hpusbkey\common.cfg (420 bytes)
Extracting: E:\usb\hpusbkey\eula.en (9.6 KB)
Extracting: E:\usb\hpusbkey\eula.ja (9.4 KB)
Extracting: E:\usb\hpusbkey\hpmboot.cfg (72 bytes)
Extracting: E:\usb\hpusbkey\hpuninst.xml (528 bytes)
Extracting: E:\usb\hpusbkey\hpusb.jpg (6.2 KB)
Extracting: E:\usb\hpusbkey\hpusbkey.exe (3.2 MB)
Extracting: E:\usb\hpusbkey\hpusbkey.inf (6 KB)
Extracting: E:\usb\hpusbkey\install.xml (204 bytes)
Extracting: E:\usb\hpusbkey\mainmenu.cfg (120 bytes)
Extracting: E:\usb\hpusbkey\syslinux.cfg (111 bytes)
Extracting: E:\usb\hpusbkey\syslinux.exe (26.5 KB)
Extracting: E:\usb\hpusbkey\usbhlpen.htm (28 KB)
Extracting: E:\usb\hpusbkey\usbhlpja.htm (31.4 KB)
Extracting: E:\usb\hpusbkey\vesamenu.c32 (144.3 KB)
  Replaced with local version ✓
Extracting: E:\usb\sdgconfig (10 bytes)
Extracting: E:\usb\syslinux (24.9 KB)
```

So at least we know that the `vesamenu.c32` files should have been replaced.
I really don't get how your earlier log would not have shown these lines if the file was there (which according to the zip file you sent it was). This is very puzzling. Did you edit your earlier log at all??

Now, because your earlier log didn't how these files being replaced, I didn't test actually booting that USB, so it's possible the problem is still downstream. I'll try to do that when I get a chance.

nidarian, 2016-09-29:
Hello Sir, I have not edited the log in any ways. My experience with this ISO is that it will boot on any system, it will not run the "HP" software after it detects the system is non-HP. 
I am building more USB sticks today, in an effort to make more logs. Will post soon.

nidarian, 2016-09-29:
I found something new, it may help with testing.
We already download the latest SPP, but there is a feature I was not aware of.
Open the downloaded SPP and extract the usb/hpusbkey folder.
Run the HP USB key utility.
This may show differences in the final USB product. 
I did not use this before.

pbatard, 2016-09-29:
Here's what you need to the USB drive created by Rufus to work:
1. Edit `\syslinux.cfg` on the root of the USB
2. Replace its content with:
```
DEFAULT loadconfig

LABEL loadconfig
  CONFIG /system/isolinux.cfg
  APPEND /system/
```

If you do that, your USB will work.

The problem is there are multiple isolinux/syslinux on the ISO (one in `usb/` the other n `system/`) and of course Rufus has to try to guess which one is the right one. Unfortunately, it's the one in `system/`, whereas Rufus picks the one in `usb/` by default.

pbatard, 2016-09-29:
Now, I have to add that, had HP decided not to use their own custom USB generation tool on the ISO (which tries to duplicate what Rufus can do), we wouldn't be having this issue...

nidarian, 2016-09-29:
[log.txt](https://github.com/pbatard/rufus/files/500870/log.txt)
Here is a log of all the files and directories from the HP USB tool. This USB stick works.

nidarian, 2016-09-29:
my log file is late. Sorry. I see you already discovered the issue. Congrats

pbatard, 2016-10-03:
I don't think I will add a special case for a very specific HP ISO. Whenever there exists two config files for Syslinux/Isolinux on the same disc, Rufus tries to use the one with the shortest path, as this is the logic that makes most sense. It so happens that, in the case of this specific media, the logic doesn't work, but it's very possible that someone may create an ISO with a config file in `/usb` which is the one to pick, so trying to fix things for this HP ISO may have negative consequences.

Therefore I will not fix this issue.

closed 2016-10-03 by pbatard

worldsdream, 2017-09-11:
This is the issue that I just opened up.

https://github.com/pbatard/rufus/issues/1021#issuecomment-328594480

pbatard, 2017-09-11:
And I will give the same answer:

> I don't think I will add a special case for a very specific HP ISO. Whenever there exists two config files for Syslinux/Isolinux on the same disc, Rufus tries to use the one with the shortest path, [...]

worldsdream, 2017-09-11:
Ok then I will be using unetbootin just for the HP iso. 

@nidarian format your usb with fat32 in windows. Then use unetbootin with the iso. Works perfect.

DonKiShoot, 2018-01-31:
Hi,

I tried the same with Proliant Service Pack 2017.10.1 ISO

After the "Could not find kernel image : vesamenu.c32 ... vmlinuz" bug like here,
i tried to replace "usb" by "system" in /syslinux.cfg like @pbatard suggest but now i have : 

Could not find kernel image : /system/syslinux.cfg

It's isolinux.cfg in the folder system

Regards,
[rufus.log](https://github.com/pbatard/rufus/files/1682017/rufus.log)

contoso-eu, 2019-03-19:
Event tough this article is closed, but continuing beeing one of the first entries in search engines, i post this for other people that searching for this solution.

**Problem:**
First thing, why you are trying to make a bootable usb stick, may because you have a slow internet connection for update inside intelligent provisioning or the downloads in intelligent provisioning fail. So now you picked up a mess when:

You try to write a usb flash drive with the iso image of HPE ProLiant Service Pack - SPP AND you decide to write the image with Rufus (my version is 3.4 with standard settings) because, as like me, you know Rufus as one of the fastest an simplest usb tools. Thereforce it was your first choice. However afterwards, your ProLiant Server won't boot from the stick or he does not recognize it as bootable, or you getting mount errors.

**So now, some facts:**
- It does not matter if you select ISO or DD mode
- It does not matter if you change `vesamenu.c32` with that one Rufus suggest or not
- The syslinux does not match the needed version for the SPP ISO (4.07 vs 3.75 at this time)
- Rufus select the wrong isolinux.cfg in addition
- UnetBootin also didn't get it working
Some people say: 
Rufus finds two config files, one at `/usb/isolinux.cfg` and another at `/system/isolinux.cfg`. It picks the one at `/usb` but this is the wrong one, and points to it at `\syslinux.cfg`. So you have to edit `\syslinux.cfg` and change it to:
```
DEFAULT loadconfig

LABEL loadconfig
  CONFIG /system/isolinux.cfg
  APPEND /system/ 
```

In my tests, this was true for usb sticks greater 32GB. Without the syslinux.cfg corrections the usb stick was not recognized at boot time (F11). But it had no impact on sticks with or below 32GB. There it makes no difference which of the two isolinux.cfg the syslinux.cfg file points to. They always where recognized and booting. But:

Finally, after all tests, after correcting the `syslinux.cfg` file, replacing `vesamenu.c32` i was most time able to boot, but i got his error message after selecting automatic or interactive update.
_Warning!!! Unable to mount the file system [cdrom]
Warning!!! Unable to moun the file system
Preboot maintenance mode
/bin/ash:  can't access tty:  job control turned off_

I did my best, but at this point i have to say, there's no "rufus-way" to get the "very special" HP SPP ISO working. At least i did'nt find it.

**Working alternative to Rufus - the HPE usbkey tool:**
-Your USB Stick should NOT be greater than 32GB, instead it's not viewed by the tool.
-Within the SPP iso file, there is `/usb/usbkey/usbkey.exe`. With this HPE Tool you can create you stick and it works anytime. However it's much slower than Rufus, but it works and thats the point. Remember you can simply browse the content of the iso file by open with windows explorer.

**Conclusions:**
To get a working HP ProLiant SPP usb flash disk working, you have to choose one of this solutions:
1) The simplest solutions is using the usb drive key tool at `/usb/usbkey/usbkey.exe` inside the iso file, with a usb stick not greater than 32GB. Then it works, but you just have to have patience and wait. It works.
2) If you have access to ILO, then use the iso file booting with ILO - this works.

There's only one thing left to say... You have nice servers, but shame on you HPE for the most messy iso arround the globe!