pbatard/rufus #836: ASMT 2135 USB 3.1 Chipset Support

unclerunkle, 2016-09-28:
<PLEASE READ THIS CAREFULLY: You *MUST* read and complete the checklist below, by placing an x into each [ ], BEFORE clicking on 'Submit new issue'. Failure to perform these steps, WHICH ARE ONLY THERE TO HELP *YOU*, will result in the issue being dismissed without warning.>

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues, using keywords relevant to my problem.
- [x] I clicked the `Log` button in Rufus and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking the "bad blocks" check box in Rufus, and confirmed that my USB is not defective
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive
  - [ ] Plugging the USB into a different port
  - [ ] Running Rufus on a different computer
- [ ] If using an ISO image, I clicked on the `#` button (at the bottom of the Rufus interface), to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Recently purchased one of the fancy new USB 3.1 m.2 enclosures and it came with the ASMT 2135 chipset.  Rufus appears to detect it as a hard drive.  Any way I can force detection?

Log
---
```
Rufus version: 2.11.995
Windows version: Windows 10 64-bit (Build 10240)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta3
System locale ID: 0x0409
Will use default UI locale 0x0409
Found UAS (USB 3.0) device 'ASMT 2135 SCSI Disk Device' (174C:1351)
Device eliminated because it was detected as a Hard Drive (score 3 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives in 'Advanced Options' (after clicking the white triangle)
Found non-USB removable device 'ST31500541AS' => Eliminated
0 devices found
```

pbatard, 2016-09-28:
>  Any way I can force detection?

1. Please read the FAQ more carefully [next time](https://github.com/pbatard/rufus/wiki/FAQ#Help_my_USB_drive_is_not_detected). Don't just treat those checkboxes as an annoyance...
2. Or, you can also read the content of the log you just pasted, which tells you __precisely__ what you can do:
```
NOTE: You can enable the listing of Hard Drives in 'Advanced Options' (after clicking the white triangle)
```
Now, can you tell more about that 'ASMT 2135 SCSI Disk Device' peripheral. Is it really a flash drive? If so, do you have the manufacturer/model from when you purchased it (as I strongly doubt it was sold as 'ASMT 2135 SCSI Disk Device')?

unclerunkle, 2016-09-28:
Hello, apologies on the FAQ.  Usually I'm better at reading comprehension.  Advanced options does list and formats as expected.  Thank you.

To get the features I wanted (USB 3.1 and m.2), I was forced to purchase from eBay from a listing such as this:  http://www.ebay.com/itm/80mm-USB-C-USB-3-1-Type-C-to-M-2-NGFF-PCI-E-2-Lane-SSD-Enclosure-for-E431-E531-/252091148221?

pbatard, 2016-09-28:
The device was properly identified as a HDD/SSD adapter, and therefore not listed by default, as per Rufus' design. There's nothing to fix, so I'll close this issue.

closed 2016-09-28 by pbatard