pbatard/rufus #838: Create a Dual BIOS + UEFI windows 10 installation media

NSbuilder, 2016-10-06:
I'm trying to make a dual BIOS + UEFI windows 10 installation media.

> This is done to avoid confusion, as it can be difficult for non expert users to know whether they

> actually booted in UEFI or BIOS mode, when an USB Flash Drive can be booted in both modes, and

NSbuilder, 2016-10-06:
I created an issue.

pbatard, 2016-10-06:
> Do I need to convert it to FAT32?

You shouldn't have to.

> ( I think UEFI can only boot with FAT32 ).

No, that's [wrong](https://github.com/pbatard/uefi-ntfs). Rufus will automatically install an UEFI NTFS driver if needed.

> How can I know if I actually booted in UEFI mode or BIOS mode when installing on UEFI computer?

1. Don't use dual BIOS + UEFI 😺 That's why this is disabled by default so that then you can be 100% certain that, if you used "GPT partition scheme for UEFI" in Rufus, it will install in UEFI mode, or, if you used "MBR partition scheme for BIOS or UEFI-CSM" (the "CSM" part is __super__ important), it will install in BIOS mode. You should only use dual BIOS+UEFI if you know what you need to do in your firmware to select BIOS or UEFI boot.
2. After install, run `msinfo32`. You will see a "BIOS mode" entry that tells whether Windows is running in BIOS or UEFI mode. Of course, you'll only find out after install, in which case, refer to 1. to make sure you install Windows in the mode you want.

NSbuilder, 2016-10-06:
Ok thanks.

NSbuilder, 2016-10-07:
Do I need to uncheck "Use Rufus MBR with BIOS ID"?

pbatard, 2016-10-07:
You shouldn't have to touch it.

pbatard, 2016-10-07:
The Rufus MBR is convenient to keep so that you can do unattended install (in BIOS mode), and prevent the computer from booting straight to USB again, on first reboot (remember that you'll get multiple reboots during install). But if you'll be in front of your computer, you can uncheck it, though I'd recommend leaving it selected.

closed 2016-10-08 by NSbuilder