pbatard/rufus #840: BIOS/GPT Configuration

github2016github, 2016-10-19:
<PLEASE READ THIS CAREFULLY: You *MUST* read and complete the checklist below, by placing an x into each [ ], BEFORE clicking on 'Submit new issue'. Failure to perform these steps, WHICH ARE ONLY THERE TO HELP *YOU*, will result in the issue being dismissed without warning.>

Checklist
---------
- [x ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x ] I performed a search in the issue tracker for similar issues, using keywords relevant to my problem.
- [x ] I clicked the `Log` button in Rufus and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking the "bad blocks" check box in Rufus, and confirmed that my USB is not defective
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive
  - [ ] Plugging the USB into a different port
  - [ ] Running Rufus on a different computer
- [ ] If using an ISO image, I clicked on the `#` button (at the bottom of the Rufus interface), to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Does Rufus support a BIOS/GPT configuration, using Grub2 and a "BIOS boot partition (type ef02)" for Arch Linux, as described in the GRUB - ArchWiki? If so how do I set up Rufus?

thanks,

Log
---
```
<FULL LOG>
```

pbatard, 2016-10-19:
> Does Rufus support a BIOS/GPT configuration

No, since BIOS is __incompatible__ with GPT, and anything that mixes GPT for BIOS uses is either for protective goals (which is useless for USB installers), or hacks such as ISOHybrid (that is also irrelevant for Rufus, since Rufus does support ISOHybrid images and doesn't need to do anything).

What is the __issue__ you are trying to solve? What exactly makes you think you need support for this, knowing that I tend to only add a feature if it solves an __actual issue__ that I expect many people to run into, rather than because something is mentioned on a wiki? Have you tried installing Arch and encountered an issue?

github2016github, 2016-10-19:
Hi Pete,

Thanks for replying.

I was considering installing Arch Linux on a system with a BIOS motherboard, but I wanted to use a GPT partitioned disk. I understand that this would not be possible on a Windows system but according to the Arch Wiki it should be possible when installing Arch. At this point I could just burn the Arch installer ISO to a CD and try it out, but I would rather use Rufus to create a USB flash drive. I don't see why you could not support this configuration; maybe in a cheat code.

Thanks,

Ron

pbatard, 2016-10-19:
> I was considering installing Arch Linux on a system with a BIOS motherboard, but I wanted to use a GPT partitioned disk.

Why? If your PC is BIOS based, using GPT will not bring you ANYTHING, as it's incompatible with BIOS. This has nothing to do with Windows or Linux, it's just a BIOS limitation.

So that BIOS/GPT thing is really only useful if you plan to use the same disk on __BOTH__ a BIOS and GPT system, and, for reasons that escape me (coz UEFI is fully compatible with MBR disks in the first place, so you don't really need GPT) somehow want to have your disk seen as GPT partitioned on an UEFI system.

Well, unless you are creating an ISOHybrid image (which you aren't), and are pedantic to the point that you want UEFI systems to see a GPT partition scheme just because, there's absolutely ZERO point in wanting to have anything remotely GPT on a BIOS system. It won't make your system go faster, it won't make it more modern (coz your BIOS will still limit what you can do, and BIOS can't do anything with GPT), and what's more, it will make your life __more__ complicated for __no logical reason__.

Now, as I already indicated, my understanding is that the BIOS/GPT thing is just something that Arch want to use to produce their own ISOHybrid images, which is what you download for them. And the thing is, if they set BIOS/GPT on their side, then, if you select DD mode when prompted by Rufus (which you will when it detects that the ISO you selected is an ISOHybrid), then your USB will be created in this mode by Rufus &mdash; You don't have to do anything special for that.

But again, I really don't see the point since you're NEVER going to be able to boot a __pure__ GPT disk from a BIOS system, and the best you can accomplish, by jumping through a lot of __brittle hoops__ (that I can almost guarantee you will regret using the day you have a boot loading issue and need to fix your system), is some kind of hybrid MBR + GPT disk, which serves very little purpose when all your target machine will ever understand is BIOS.

At any rate, __PLEASE__ try selecting your Arch ISO in Rufus, and you should find that, as long as you keep "MBR partition scheme for BIOS or UEFI" selected, you will have no issue installing Arch on your BIOS system. You can even select "DD mode" instead of "ISO mode" when prompted, if you really believe you need that weird Arch mode, but the result of the install will be exactly the same, and, if your __BELIEF__ is that because Rufus does not allow the explicit selection of a weird custom mode that's only documented on the Arch Wiki, it will not work, I will have to ask you to PLEASE try it, before considering that it's going to fail.

As I explained previously, I try to solve __actual__ issues, not issues that people __think__ they will be having because they have misinterpreted something they have read somewhere. I really don't understand how you come from the conclusion that you will need to install Arch from an optical media, when I don't believe there's anything on the Arch Wiki or what I posted earlier that is actually telling you that, unless you use GPT/BIOS, Arch install cannot work.

As to your _"I don't see why you could not support this configuration; maybe in a cheat code."_, I already tried to explain that I only add feature for problems that are demonstrated to be likely to impact a lot of people (For the record, I regularly test the latest Arch ISOs with Rufus, and have yet to find an issue with the bootable USB created then. And neither have the 3 million people who download Rufus each month), and, again, there is no issue that I know of (or feature that you will somehow miss during the Arch install) with not using GPT/BIOS. Also, see [this](https://github.com/pbatard/rufus/wiki/FAQ#Why_cant_you_just_add_the_feature_I_want_I_mean_youre_a_developer__it_shouldnt_be_that_hard). If I were to add the feature to request, I'd have to __support__ it, explain why it is needed (which you haven't made any case for, so that will be hard to do) and make sure it doesn't get regressed from version to version, by regularly testing... all of this for something that is likely to be used by a single person (__you__) because they wrongly assume that this is a requirement for Arch install. That's just not gonna happen, sorry.

closed 2016-10-19 by pbatard