pbatard/rufus #841: [QUESTION] OEM windows license

NSbuilder, 2016-10-19:
I don't think it is really related to Rufus but I want to know.

I created a dual BIOS + UEFI windows 10 installation with Rufus and installed it on my blank BIOS laptop and everything worked fine.

Then I booted it on a UEFI Windows 8/10 HP laptop that already had a windows home license ( OEM ) and after I deleted all the partitions and clean installed from the USB - It didn't asked me for a key and installed windows 10 home with the key from the previous installation and with the HP logo on startup.

1. How it was activated with the key from the previous installation if I deleted all the partitions? What is this OEM thing?

2. Can I do a true Clean Install without the auto key activation?

pbatard, 2016-10-19:
> How it was activated with the key from the previous installation

Windows 10 contacts Microsoft servers for licensing, and sends a digital hardware footprint, that allows them to relicense the same machine, if it was already licensed. As far as I know, it only cares about the key for the very first activation of a specific machine. After that, it just looks for the hardware signature in Microsoft's licensing database, and validates the licensing rights there.

> Can I do a true Clean Install without the automatic key activation

I don't think you can avoid automatic relicensing of the same machine, unless you contacted Microsoft to tell them that the license you were using is no longer valid and that they need to remove it from their licensing DB. Note however that, if you are hoping to be able to use this to transfer an existing Windows 10 license from one computer to another, you may find out from Microsoft that you aren't entitled to do that, as it is my current understanding that Windows 10 licenses are non transferable (though I wouldn't mind seeing this officially confirmed).

pbatard, 2016-10-19:
Of course, I would __STRONGLY__ advise you to contact Microsoft support with this question, as I make no guarantee whatsoever about the above, and __ONLY MICROSOFT__ can answer your question.

closed 2016-10-19 by pbatard