pbatard/rufus #842: Rufus crashes everytime i try to load an ISO.

fortafyhawk, 2016-10-23:
Windows 10 Pro x64
i wanna install Windows 7 on my PC (Which is UEFI) and everytime i try to open any iso with Rufus it only freezes and i can only end the program via Task Manager lol.

 i tried running it as admin, compatibility mode and a lot of things even kicking off my laptop lol.
Checklist
---------
- [X] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [X] I performed a search in the issue tracker for similar issues, using keywords relevant to my problem.
- [ ] I clicked the `Log` button in Rufus and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking the "bad blocks" check box in Rufus, and confirmed that my USB is not defective
- [X ] I also tried one or more of the following:
  - [ X] Using a different USB drive
  - [ X] Plugging the USB into a different port
  - [ ] Running Rufus on a different computer
- [ ] If using an ISO image, I clicked on the `#` button (at the bottom of the Rufus interface), to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<Please describe your issue here>

Log
---
```
<FULL LOG>
```

pbatard, 2016-10-23:
I'm afraid you're the only person to report this issue, despite the fact that Rufus is downloaded by millions of people each month, so if this happens with every ISO you attempt to select, then it looks like there is an issue with your environment.

Can you confirm that this really happens with all the ISOs you tried? Also can you provide the SHA-1 of some of the ISOs you tried as well as where you obtained them? And are you sure you have full administrative privileges on your machine? What happens if you try on a different computer?

Finally, you can get the Rufus log by installing ans running [DebugView](https://technet.microsoft.com/en-us/sysinternals/debugview.aspx). Can you please do that and copy paste what you get (you should see some date being generated when running Rufus) as it may provide some explanation on the behaviour you observed.

fortafyhawk, 2016-10-23:
Going to upload a video, wait a min.

pbatard, 2016-10-25:
Just wondering if you had a chance to run DebugView and can copy/paste the log you get there.

pbatard, 2016-10-28:
No update &rarr; Closed

closed 2016-10-28 by pbatard