pbatard/rufus #844: Formatting Error.

Alienbyte, 2016-10-24:
<PLEASE READ THIS CAREFULLY: You *MUST* read and complete the checklist below, by placing an x into each [ ], BEFORE clicking on 'Submit new issue'. Failure to perform these steps, WHICH ARE ONLY THERE TO HELP *YOU*, will result in the issue being dismissed without warning.>

Checklist
---------
- [X] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [X] I performed a search in the issue tracker for similar issues, using keywords relevant to my problem.
- [X] I clicked the `Log` button in Rufus and copy/pasted the log into the line that says `<FULL LOG>` below.
- [X] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [X] I ran a bad blocks check, by clicking the "bad blocks" check box in Rufus, and confirmed that my USB is not defective
- [X] I also tried one or more of the following:
  - [X] Using a different USB drive
  - [X] Plugging the USB into a different port
  - [X] Running Rufus on a different computer
- [X] If using an ISO image, I clicked on the `#` button (at the bottom of the Rufus interface), to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
I've been trying to install tails, and essentially at this point after trying to use this program last night I ran into an issue where my USB drive just wasn't really changing. I can't format it, I can't delete anything, and if I do it just comes right back. Nothing saves AT ALL. I've tried clean, but it displays errors as well. I understand this could be my fault, but I've searched for hours and hours and I just want to get this to work, so if ANYONE can help I would more than appreciate it! I will say I think what may have happened is I made the mistake of accidentally trying to flash the usb drive with a .iso file that was on the USB drive already, not on the desktop.

Log
---
```
Rufus version: 2.11.995
Windows version: Windows 10 64-bit (Build 14393)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta3
System locale ID: 0x0409
Will use default UI locale 0x0409
Found USB 2.0 device 'PNY USB 2.0 FD USB Device' (154B:009F)
Found USB 3.0 device 'Corsair Survivor 3.0 USB Device' (1B1C:1A0A)
2 devices found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 985, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0xC3072E18
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 7.5 GB (8103366656 bytes)
  Start Sector: 56, Boot: Yes
2 devices found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 3891, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0xC3072E18
Drive has a Windows 2000/XP/2003 Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 29.8 GB (32008798208 bytes)
  Start Sector: 64, Boot: No
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/isolinux/isolinux.cfg' for Syslinux
  Detected Syslinux version: 6.03/20150819 (from '/EFI/BOOT/isolinux.bin')
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'TAILS 2.6 - 20160920'
  Size: 1167990784 bytes
  Uses: Syslinux/Isolinux v6.03
  Uses: EFI
Using image: tails-i386-2.6.iso
Will reuse 'ldlinux.sys' and 'ldlinux.bss' from './rufus_files/rufus_files/syslinux-6.03/20150819/' for Syslinux installation

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE2 for write access
Will use 'X:' as volume mountpoint
I/O boundary checks disabled
Analyzing existing boot records...
Drive has a Windows 2000/XP/2003 Master Boot Record
Volume has an unknown Partition Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 64 sectors
Deleting partitions...
Partitioning (MBR)...
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (FAT32)...
Using cluster size: 16384 bytes
Quick format was selected
Creating file system...
Error while formatting
Format error: Undetermined error while formatting.
Re-mounted volume as 'X:' after error

Found USB 2.0 device 'PNY USB 2.0 FD USB Device' (154B:009F)
Found USB 3.0 device 'Corsair Survivor 3.0 USB Device' (1B1C:1A0A)
2 devices found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 3891, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0xC3072E18
Drive has a Windows 2000/XP/2003 Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 29.8 GB (32008798208 bytes)
  Start Sector: 64, Boot: No

Computing checksum for 'C:\Users\Alienbyte\Desktop\tails-i386-2.6.iso'...
  MD5:	 7cb169c190388da90a1c5729af0bb1f4
  SHA1:	 a8a49b14fde031fe143ed3b8bf2d526766ed5d62
  SHA256: d9332b64412b3d4f5ade0e6cbf3adff233f3dd38e3bc26987a10b887b10e0ef2
```

pbatard, 2016-10-24:
> I've tried clean, but it displays errors as well.

I assume that was using Windows' `diskpart`. Or was it with Rufus?

Usually, `Format error: Undetermined error while formatting` comes from the Windows formatting service (which Rufus merely invokes), so I assume you are also getting an issue when you try to format the drive using Windows rather than Rufus. Can you confirm?

The one thing you may want to try is force Rufus to use Large FAT32 for formatting, as, in this case, it won't be relying on the Windows formatting service. To do that, just press <kbd>Alt</kbd>-<kbd>L</kbd> in Rufus. It should display `Force Large FAT32 formatting enabled` at the bottom to confirm. I'm not hopeful that this will actually solve your issue, but it's worth a try.

Alienbyte, 2016-10-24:
```
Rufus version: 2.11.995
Windows version: Windows 10 64-bit (Build 14393)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta3
System locale ID: 0x0409
Will use default UI locale 0x0409
Found USB 2.0 device 'PNY USB 2.0 FD USB Device' (154B:009F)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 985, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0xC3072E18
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 7.5 GB (8103366656 bytes)
  Start Sector: 56, Boot: Yes
Found USB 2.0 device 'PNY USB 2.0 FD USB Device' (154B:009F)
Found USB 3.0 device 'Corsair Survivor 3.0 USB Device' (1B1C:1A0A)
2 devices found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 985, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0xC3072E18
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 7.5 GB (8103366656 bytes)
  Start Sector: 56, Boot: Yes
Found USB 2.0 device 'PNY USB 2.0 FD USB Device' (154B:009F)
Found USB 3.0 device 'Corsair Survivor 3.0 USB Device' (1B1C:1A0A)
2 devices found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 985, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0xC3072E18
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 7.5 GB (8103366656 bytes)
  Start Sector: 56, Boot: Yes
2 devices found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 3891, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0xC3072E18
Drive has a Windows 2000/XP/2003 Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 29.8 GB (32008798208 bytes)
  Start Sector: 64, Boot: No
Found USB 2.0 device 'PNY USB 2.0 FD USB Device' (154B:009F)
Found USB 3.0 device 'Corsair Survivor 3.0 USB Device' (1B1C:1A0A)
2 devices found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 985, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0xC3072E18
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 7.5 GB (8103366656 bytes)
  Start Sector: 56, Boot: Yes
2 devices found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 3891, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0xC3072E18
Drive has a Windows 2000/XP/2003 Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 29.8 GB (32008798208 bytes)
  Start Sector: 64, Boot: No
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/isolinux/isolinux.cfg' for Syslinux
  Detected Syslinux version: 6.03/20150819 (from '/EFI/BOOT/isolinux.bin')
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'TAILS 2.6 - 20160920'
  Size: 1167990784 bytes
  Uses: Syslinux/Isolinux v6.03
  Uses: EFI
Using image: tails-i386-2.6.iso
Will reuse 'ldlinux.sys' and 'ldlinux.bss' from './rufus_files/rufus_files/syslinux-6.03/20150819/' for Syslinux installation

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE2 for write access
Will use 'X:' as volume mountpoint
I/O boundary checks disabled
Analyzing existing boot records...
Drive has a Windows 2000/XP/2003 Master Boot Record
Volume has an unknown Partition Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 64 sectors
Deleting partitions...
Partitioning (MBR)...
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (Large FAT32)...
Opened drive \\?\Volume{ee55d13a-63e8-11e6-9c01-dc53606ffe18} for write access
I/O boundary checks disabled
Size : 29.8 GB 62517184 sectors
Cluster size 16384 bytes, 512 Bytes Per Sector
Volume ID is 13f7:2247
32 Reserved Sectors, 15256 Sectors per FAT, 2 FATs
1952707 Total clusters
1952706 Free Clusters
Clearing out 30576 sectors for reserved sectors, FATs and root cluster...
write_sectors: Write error [0x00000079] The semaphore timeout period has expired.
  StartSector: 0x00000000, nSectors: 0x80, SectorSize: 0x200
Error clearing reserved sectors
Format error: Write error.
Re-mounted volume as 'X:' after error

Found USB 2.0 device 'PNY USB 2.0 FD USB Device' (154B:009F)
Found USB 3.0 device 'Corsair Survivor 3.0 USB Device' (1B1C:1A0A)
2 devices found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 3891, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0xC3072E18
Drive has a Windows 2000/XP/2003 Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 29.8 GB (32008798208 bytes)
  Start Sector: 64, Boot: No
```

pbatard, 2016-10-24:
From what I can see, the problem is either with your USB drive or your environment, so there's not much that can be done to help you out.

Have you __really__ tried using a different USB flash drive and from a different PC???

pbatard, 2016-10-28:
As it seems either your USB is defective or your PC has an issue, and you haven't answered the question I asked you, I will close this issue.

closed 2016-10-28 by pbatard