pbatard/rufus #845: Rufus never appears on screen but running in task manager?

ruinit, 2016-10-26:
<PLEASE READ THIS CAREFULLY: You *MUST* read and complete the checklist below, by placing an x into each [ ], BEFORE clicking on 'Submit new issue'. Failure to perform these steps, WHICH ARE ONLY THERE TO HELP *YOU*, will result in the issue being dismissed without warning.>

Checklist
---------
- [ x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x ] I performed a search in the issue tracker for similar issues, using keywords relevant to my problem.
- [ ] I clicked the `Log` button in Rufus and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking the "bad blocks" check box in Rufus, and confirmed that my USB is not defective
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive
  - [ ] Plugging the USB into a different port
  - [ ] Running Rufus on a different computer
- [ ] If using an ISO image, I clicked on the `#` button (at the bottom of the Rufus interface), to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Cant get Rufus to show up on screen? It shows in task manager I have waited 10 mins plus many times and different version.

I looked around and could not find my issue. I have tried 2.10, 2.11 both regular and portable. I am running Windows 10 x64. I Rebooted system and did all the normal technical crap. I even tried compatibility mode. What am I missing?

Log
---
```
<FULL LOG>
```

pbatard, 2016-10-26:
Can you please install and run [DebugView](https://technet.microsoft.com/en-us/sysinternals/debugview.aspx). Rufus also sends its log data there, so even if you don't see the UI, we may get some information on where the application might be stuck.

If you can copy/paste the content you get from DebugView after launching Rufus, I would appreciate it.

Also, are you sure you have full administrative privilege on the computer you are trying to run Rufus on? If you are missing some privileges, it may not be possible for Rufus to run properly.

pbatard, 2016-10-28:
Any update on what I asked above?

pbatard, 2016-11-11:
No update &rarr; Closed.

closed 2016-11-11 by pbatard

xdHampus, 2019-03-16:
I experienced the same issue as @crutchcorn and it was also resolved by unplugging my USBs.