pbatard/rufus #846: Rufus Error

KohaviZ, 2016-10-26:
I have another computer which had it's hard disk switched to an empty one. Dell gave me a windos ISO file that is made for that computer. I am using another computer to make a bootable usb with the ISO on it but I get an error from Rufus.

Heres the log:

Rufus version: 2.11.995
Windows version: Windows 10 64-bit (Build 14393)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta3
System locale ID: 0x040D
Detected installed Windows Language Pack for 0x0409 (English (English))
0 devices found
Scanning image...
ISO analysis:
  Image is an UDF image
Could not locate file /sources/install.swm in ISO image
Disk image analysis:
  Image does not have an x86 Master Boot Record
ISO label: 'W10_x64'
  Size: 4294963200 bytes
  Uses: EFI
  Uses: Bootmgr
  Uses: Install.wim (version 255.255.255)
  Note: This WIM version is NOT compatible with Windows To Go
Using image: 9663HA00_W10x64ROW_home(DL)_v3.iso
Found USB 2.0 device 'SanDisk Cruzer Blade USB Device' (0781:5567)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 3800, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: GPT, NB Partitions: 1
Disk GUID: {53873819-D506-4588-A65B-0B843D394A37}
Max parts: 128, Start Offset: 17408, Usable = 31260670464 bytes
Partition 1:
  Type: {EBD0A0A2-B9E5-4433-87C0-68B6B72699C7}
  Name: 'Microsoft Basic Data'
  ID: {0E939770-033C-4816-A87E-05AC8557EDC4}
  Size: 29.1 GB (31259639296 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Found USB 2.0 device 'SanDisk Cruzer Blade USB Device' (0781:5567)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 3800, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: GPT, NB Partitions: 1
Disk GUID: {53873819-D506-4588-A65B-0B843D394A37}
Max parts: 128, Start Offset: 17408, Usable = 31260670464 bytes
Partition 1:
  Type: {EBD0A0A2-B9E5-4433-87C0-68B6B72699C7}
  Name: 'Microsoft Basic Data'
  ID: {0E939770-033C-4816-A87E-05AC8557EDC4}
  Size: 29.1 GB (31259639296 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE1 for write access
Will use 'E:' as volume mountpoint
I/O boundary checks disabled
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Volume has an unknown Partition Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Deleting partitions...
Partitioning (GPT)...
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (FAT32)...
Using cluster size: 16384 bytes
Quick format was selected
Creating file system...
Format completed.
Found volume GUID \\?\Volume{03d6d563-9b84-11e6-89cc-bc8556de6b66}\
Successfully remounted Volume{03d6d563-9b84-11e6-89cc-bc8556de6b66}\ on E:\
Copying ISO files...
Extracting files...
Image is an UDF image
Extracting: E:\Sources\BOOT.wim (577.3 MB)
Extracting: E:\Sources\OEM\AfterImage.CMD (2.1 KB)
  Error reading UDF file /Sources/OEM/AfterImage.CMD

Found USB 2.0 device 'SanDisk Cruzer Blade USB Device' (0781:5567)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 3800, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: GPT, NB Partitions: 1
Disk GUID: {3B2EA530-4080-4380-BFFE-F2782EEE8395}
Max parts: 128, Start Offset: 17408, Usable = 31260670464 bytes
Partition 1:
  Type: {EBD0A0A2-B9E5-4433-87C0-68B6B72699C7}
  Name: 'Microsoft Basic Data'
  ID: {DCFA9DE9-D89D-452F-B4FB-20AD4D02EF33}
  Size: 29.1 GB (31259639296 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000

pbatard, 2016-10-26:
Looks like your ISO is corrupted. Can you extract `Sources/OEM/AfterImage.CMD` with [7-zip](http://www.7-zip.org/)?

KohaviZ, 2016-10-26:
yes i can (winrar but yes)

pbatard, 2016-10-26:
Then I'll need to have access to your ISO. Can you upload it somewhere?

KohaviZ, 2016-10-26:
where?

pbatard, 2016-10-26:
It's up to you. I think [mega.nz](https://mega.nz/) offers upload services. I'm afraid I don't have a dedicated server for uploads.

KohaviZ, 2016-10-26:
i am currently uploading to google drive. fine?

pbatard, 2016-10-26:
Yup, whatever suits you best. 
Also, if you don't want to share the download link with everybody, you can e-mail it to: pete@akeo.ie.

KohaviZ, 2016-10-26:
its taking for ever

pbatard, 2016-10-28:
Did you manage to upload your ISO anywhere? If so can you provide me with the link? Thanks.

KohaviZ, 2016-10-28:
Not yet... had issues with that...

pbatard, 2016-10-28:
No worries. Feel free to take your time and let me know when you've had a chance to do it.

pbatard, 2016-10-30:
Got your ISO now, and I'm afraid that, regardless of what happens with WinRAR, the problem __is__ that your ISO is either corrupted or was not mastered properly. Coz here's what happens when I try to extract the `Sources/OEM/AfterImage.CMD` with 7-zip:

[image]

If both Rufus and 7-zip indicate that there is an issue with your ISO, and you confirmed that the ISO wasn't altered, then you will need to contact that company who created that ISO and tell them that they didn't master it properly (I would suggest that they test data extraction with 7-zip if they need proof).

Since there is nothing I can do if the ISO is problematic (I suspect that the WinRAR application is simply choosing to ignore errors rather than reporting them, which is not something I want to emulate in Rufus), I will now close this issue.

closed 2016-10-30 by pbatard