pbatard/rufus #847: Windows to go: Unable to select index of install.wim

Igorjan666, 2016-10-27:
При создании флешки индекс просто не запрашивается. По умолчанию используется "1". Что, соответственно не позволяет выбрать редакцию ОС (В частности, Win Server 2016 всегда устанавливается в варианте Standart, Server core)
-----------------
<Please describe your issue here>

Log
---
```
<FULL LOG>
```

pbatard, 2016-10-27:
Please point to an official Microsoft ISO that I can test from [here](https://msdn.microsoft.com/en-us/subscriptions/downloads/#searchTerm=&ProductFamilyId=665&Languages=ru,en&FileExtensions=.iso&PageSize=10&PageIndex=0&FileId=0). If you are using a custom ISO, please be aware that they are not supported.

Also, I still want to see your FULL LOG. Unless you provide your FULL LOG, as requested by the instructions, I will just close your issue.

Igorjan666, 2016-10-27:
I've used VLSC image 
File: SW_DVD9_Win_Svr_STD_Core_and_DataCtr_Core_2016_64Bit_Russian_-2_MLF_X21-22838.ISO
CRC-32: 65875c75
   MD4: d408113ab635562ae1b707af8e458a0e
   MD5: 68c7e21451188e133e52ac2e124d6e5a
 SHA-1: 169bd4bf65e83a9193ed508e3f08717d32d3a6bb

I can try MSDN image tomorrow

pbatard, 2016-10-27:
`SW_DVD9_Win_Svr_STD_Core_and_DataCtr_Core_2016_64Bit_Russian_-2_MLF_X21-22838.ISO` doesn't look like an official Microsoft ISO. Rufus only supports official ISOs, so unless you experience the same issue with the MSDN ISOs, I'm not going to do anything about it.

Igorjan666, 2016-10-28:
-update-
another PC, another USB drive, MSDN image

`Opening: \\.\CDROM0\sources\install.wim:[1]`
How to tell RUFUS open install.wim[2]?

[rufus.log.txt](https://github.com/pbatard/rufus/files/557782/rufus.log.txt)

pbatard, 2016-10-28:
Okay, I looked at the official English and Russian ISOs for Windows Server 2016 and it looks like the `install.wim` contains separate entries for:
* Windows Server 2016 Standard
* Windows Server 2016 Standard (Desktop Experience)
* Windows Server 2016 Datacenter
* Windows Server 2016 Datacenter (Desktop Experience)

I'll try to modify Rufus to enable selecting the desired entry, but it will probably take a few months before I do so.

commit 7d302d34 (2016-12-13): [togo] enable selection of Windows version

* Closes #847
* Also set rufus-next to 2.12
closed 2016-12-13 by pbatard