pbatard/rufus #852: Possible to support smaller screen sizes/resolutions?

SumatraPeter, 2016-11-11:
The other day I had to use Rufus 2.11.995 on a bunch of netbooks running Win10 at their max. 1024x600 resolution. On these I noticed that toggling Advanced Options meant that the bottom portion of both the app's windows became inaccessible behind the taskbar. This included the About/Log/Start/Close buttons and the status bar on the **main window**, as well as the Clear Log/Save Log/Close Log buttons on the **log window**.

I know that devices with such resolutions might not be popular, and one can always close the Advanced Options to reduce the size of the windows, but wondered if there might be some way you can tackle this if it doesn't involve too much of an effort?

At the very least, I don't see why the log window (which currently isn't resizeable) needs to be stretched in height to match the main window such that its buttons become inaccessible. The text portion of that window is scrollable anyway, so its default height is perfect even on a 600px height screen with the taskbar visible.

The main window might be a bit tough to fix unless you add a scrollbar when extending its height or perhaps juggle the placement of the additional UI elements, so if that's not possible to alter I completely understand and would be happy with just the log window being fixed. Thank you.

[Note: This is a UI issue so I didn't bother to include the log which is completely irrelevant. Please don't close the issue outright because of this deliberate omission.]

pbatard, 2016-11-11:
This has been requested before, but I'm afraid that, seeing how few people are expected to use Rufus on low resolution devices) I don't see enough incentive to work on that. Also, this is an issue that was created by Microsoft, when they decided to make a U-turn on their guidelines for the minimum resolution for Windows 8 (in which Rufus fitted the screen). Unfortunately, I just don't have enough time to __support__ ensuring that Rufus works at lower resolutions in the long run - the cost/benefit ratio is just too high.

So I'm afraid, I'll have to reject this request - sorry.

closed 2016-11-11 by pbatard

SumatraPeter, 2016-11-11:
Wow, quick response! :) As I wrote above, I understand if you don't want to tackle the main window since it might involve extra effort, but how about simply disabling the stretching of the log window on low-res devices? That ought to be a simple fix and will solve at least half the issue, since I really can't see any advantage in having the log window's buttons also being pushed off screen.

pbatard, 2016-11-11:
I'm doing size computations for the placement of the log Window. I'd rather not throw them out of the Window and risk making the placement look bad for everybody, just so that the couple of people who use low-res devices are happy. The less changes I introduce, that will only benefit a handful of people, the less potential issues there are.

SumatraPeter, 2016-11-11:
Ok, got it. Thanks for reading and responding.