pbatard/rufus #853: Add tooltip with file info. to the status bar

SumatraPeter, 2016-11-11:
The status bar displays the name of the currently loaded ISO (or other) image, except of course that more often than not the name's truncated because of the app window's width. Would it be possible to add a tooltip that displays the full name (or better still the full path and name) of the loaded file? I know this info. is available in the log, but also having it available via a simple tooltip would be useful too. Thanks.

[Note: This is a UI issue so I didn't bother to include the log which is completely irrelevant. Please don't dismiss the issue outright because of this deliberate omission.]

pbatard, 2016-11-11:
I'm planning to redesign the UI for Rufus 3.0, so I'll see what I can do.

SumatraPeter, 2016-11-11:
Thanks for such a great utility and looking forward to 3.0 then. (Just curious though, what does "ER" expand to?)

pbatard, 2016-11-11:
ER: Enhancement Request.

I use that to quickly differentiate enhancements from bugs when looking at the list.

pbatard, 2017-01-02:
Since this is a small enhancement, I'm going to move it in the list of small enhancements for 3.0 and close this one.

closed 2017-01-02 by pbatard