pbatard/rufus #855: Windows ISO NTFS vs FAT32

nunojpg, 2016-11-12:
Having checked and understood the following:

https://github.com/pbatard/rufus/wiki/FAQ#Why_doesnt_Rufus_create_a_Windows_installation_USB_that_can_be_booted_in_both_BIOS_and_UEFI_mode

Could you please clarify why is NTFS the default for BIOS mode and FAT32 the default for UEFI?

Is this made as an attempt to avoid the firmware reading the EFI files when it is created in BIOS mode?

Thanks!

pbatard, 2016-11-12:
NTFS can't (usually) be read by UEFI, except if you add a driver (which Rufus can do through UEFI:NTFS).

So by using NTFS, and not adding the driver, Rufus can enforce BIOS or UEFi-CSM booting, on systems that would have been able to boot the USB in pure UEFI mode otherwise.

Also, the __BIG__ advantage of using NTFS is you're not limited to 4GB when it comes to the `install.wim`.

I'd rather not have to go through a long speech, with people who might not be that computer literate, about limitations of FAT32.

I believe the current defaults are both NTFS for BIOS and UEFI, though I allow FAT32 for UEFI in case the user wants it, and there doesn't exist a large file.

Anyway, you'd better believe that Rufus __always__ tries to pick the option that is best for you. If Rufus picks a file system, then you really ought to stick with that one, unless you know what you're doing.

Hope that answers your questions.

closed 2016-11-12 by pbatard

nunojpg, 2016-11-13:
Thank you so much for the clarification. And thank you again for Rufus.