pbatard/rufus #856: Microsoft Nextbook (next1012bcp) Unable to Start Driver: [1] Load Error

necrolingus, 2016-11-13:
Checklist
---------
- [x ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x ] I performed a search in the issue tracker for similar issues, using keywords relevant to my problem.
- [x ] I clicked the `Log` button in Rufus and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [x ] I ran a bad blocks check, by clicking the "bad blocks" check box in Rufus, and confirmed that my USB is not defective
- [x ] I also tried one or more of the following:
  - [x ] Using a different USB drive
  - [ x] Plugging the USB into a different port
  - [x ] Running Rufus on a different computer
- [x ] If using an ISO image, I clicked on the `#` button (at the bottom of the Rufus interface), to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Good day

I hope you can help me out.

I have a Microsoft Nextbook on which I am trying to install Windows 7 64bit. I know that my UEFI bootloader is a 32bit bootloader as I managed to load Ubuntu a long time ago (had to get the 32bit bootia32 files and all that).

I created my bootable flash drive using Rufus 2.11 and chose the following options:
GPT Partioning Scheme for UEFI
NTFS File System
Used a genuine Windows 7 DVD from which I create an ISO image

I notice the following:
A folder gets created by rufus in the same directory as my source Windows 7 ISO. This folder has 2 subfolders (boot and Rufus). Each contains 3 files:
Boot contains bootarm, bootia32, bootx64
Rufus contains ntfs_arm, ntfs_ia32, ntfs_x64

After my bootable flash drive gets created by Rufus 2.11, I plugged it into my nextbook and it booted into Grub, saying that it cannot find "efi/boot/Rufus/ntfs_ia32". I proceeded to copy the file from the location above onto my flashdrive into "/efi/Rufus/ntfs_ia32". I plugged my flashdrive back into my Nextbook and it gave me the attached photo of my error.

I tried replacing the bootia32 file that rufus automatically creates with the one above, but same error. I also extracted it out of the Windows 7 ISO in the sources folder, but then I get another error (bit I dont think the latter error is of importance as the file size is so much different from the one Rufus creates).
[image]

If you need more information please let me know, or if you need me to test some things also.

Thank you in advance

Log
---
```
<FULL LOG>
```

pbatard, 2016-11-13:
> A folder gets created by rufus in the same directory as my source Windows 7 ISO

This is very weird. Unfortunately, you decided not to post the log of how you created your USB with Rufus, so I have no idea why that might be the case.

Please post your full log. If needed, please recreate your USB to get the log (and make sure you confirm that the new USB has the same issue). I __really__ can't help you if you decide to ignore the guidelines and don't post a log.

pbatard, 2016-11-13:
Also:

>  it booted into Grub, saying that it cannot find "efi/boot/Rufus/ntfs_ia32". I proceeded to copy the file from the location above onto my flashdrive into "/efi/Rufus/ntfs_ia32".

Don't do that. You're just creating a Frankenstein monster of boot. That will __NEVER__ work.

necrolingus, 2016-11-14:
Hi @pbatard 

Thank you for the quick reply and apologies about this logs. Not sure how I fogot about it.

What I did was to do a clean create of a bootable flash. In other words, I put the ISO in its own folder, removed that "uefi files" folder (I believe it is from uefi-ntfs.iso file that i Think I got from one of your older posts). I can confirm rufus does not create this folder.

Please see attached the logs.
[logs.txt](https://github.com/pbatard/rufus/files/588487/logs.txt)

When I plug my flash drive (the one created with Rufus) into my linux machine, I see it has 2 partitions:
[image]

From what I could gather, through my manual attempts to get the nextbook to boot, this is correct right? One partition has the EFI files, and the other one the Windows files.

Let me know if you need me to check anything else.

Thank you

necrolingus, 2016-11-14:
Hi @pbatard 

I figured seeing  I have a clean flash drive now with no messing about with the EFI files and creating a Frankenstein I will try again. 

I plugged the flash drive into my nextbook and now got, which should be, the real error (last night I mucked about and was quite late so I might have messed things up a bit).

Here is the error with a clean flash drive created using Rufus (no mucking about):
[image]

But then I checked on the fat16 partition (the one in the gparted screenshot) and I did see the /efi/boot/bootia32.efi there. Then I checked in the NTFS partition and saw /efi/boot on that partition has the "bootx64.efi" file. Then out of curiosity, I copied the bootia32 from the fat16 partition into /efi/boot on the NFTS partition, just to see where my nextbook is looking for the file. I booted from it again and then it found the file in that folder (but then in was looking for /efi/Rufus/ntfs-ia32.efi file so I think it is creating like a loop or something....). This was just for my own curiosity though. I recreated the flash drive again using Rufus, just so I have a clean drive again.

Regards

pbatard, 2016-11-14:
Thanks for the log.

But now I realize (and I have to apologize for it, coz I should have noticed that way sooner) that you are trying to install Windows 7 32-bit in UEFI mode.

__THAT WILL NEVER WORK!__

The reason is that (despite what many would like to convinced themselves otherwise) Windows 7 is an old OS and was __NEVER__ designed by Microsoft to support 32-bit UEFI boot and installation. As such is is missing __criticial__ pieces, such as a `bootia32.efi` bootloader (which is not something you can replace with the first `bootia32.efi` you see, or even the `bootia32.efi` from later versions of Windows). The __only__ versions of Windows that support 32-bit installation are Windows 8 or later.

So, I'm sorry, but what you are trying to accomplish is not something that the official Windows 7 installation media support, and, as such, not something that Rufus can support.

As a result, I would encourage you to install Windows 8 or later, and will be closing this issue.

closed 2016-11-14 by pbatard

necrolingus, 2016-11-14:
Hi @pbatard 

Thank you for the quick replies.

The Windows I am trying to install is a 64bit Windows 7, but I see what you are saying about this 32bit UEFI and the 64bit CPU. I will try and source a Windows 8 to see if I can get it to work, or if I can maybe get hold of a recovery for my nextbook.

Keep well and thanks for the great work!

pbatard, 2016-11-14:
Well, if your UEFI is 32-bit (and your screenshots shot it is), then even if the underlying CPU is 64-bit (which can happen), you will not be able to install Windows 64-bit in UEFI mode. You must go with the 32-bit version. So that leaves only Windows 8 or later...