pbatard/rufus #857: Custom Syslinux Builds?

flakefree, 2016-11-15:
Checklist
---------
- [x ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues, using keywords relevant to my problem.
- [x ] I clicked the `Log` button in Rufus and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking the "bad blocks" check box in Rufus, and confirmed that my USB is not defective
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive
  - [ ] Plugging the USB into a different port
  - [ ] Running Rufus on a different computer
- [ ] If using an ISO image, I clicked on the `#` button (at the bottom of the Rufus interface), to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
More of an additional feature than fix really. Just wondering when Rufus will use custom syslinux versions such as the ones used in recent builds of Ubuntu (Syslinux 6.03/20160618) to resolve the error below!

Many Thanks

Log
---
```
Rufus version: 2.11.995
Windows version: Windows 10 64-bit (Build 14393)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta3
System locale ID: 0x0809
Will use default UI locale 0x0809
Found USB 3.0 device 'Generic STORAGE DEVICE USB Device' (05E3:0749)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 7764, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x02B75995
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: exFAT (0x07)
  Size: 59.5 GB (63863521280 bytes)
  Start Sector: 2048, Boot: Yes
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/isolinux/isolinux.cfg' for Syslinux
  Detected Syslinux version: 6.03/20160618 (from '/isolinux/isolinux.bin')
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'Ubuntu 16.10 amd64'
  Size: 1593835520 bytes
  Has a >64 chars filename
  Uses: Syslinux/Isolinux v6.03
  Uses: EFI
  Note: This ISO uses symbolic links, which will not be replicated due to file system limitations.
  Because of this, some features from this image may not work...
Using image: ubuntu-16.10-desktop-amd64.iso
Downloading 'ldlinux.sys' from http://rufus.akeo.ie/files/syslinux-6.03/20160618/ldlinux.sys
Unable to access file: 404
Extended version was not found, trying main version
Downloading 'ldlinux.sys' from http://rufus.akeo.ie/files/syslinux-6.03/ldlinux.sys
File length: 68599 bytes
Successfully downloaded 'ldlinux.sys'
Downloading 'ldlinux.bss' from http://rufus.akeo.ie/files/syslinux-6.03/20160618/ldlinux.bss
Unable to access file: 404
Extended version was not found, trying main version
Downloading 'ldlinux.bss' from http://rufus.akeo.ie/files/syslinux-6.03/ldlinux.bss
File length: 512 bytes
Successfully downloaded 'ldlinux.bss'

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE3 for write access
Will use 'F:' as volume mountpoint
I/O boundary checks disabled
Analyzing existing boot records...
Drive has a Syslinux Master Boot Record
Volume has an unknown Partition Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 2048 sectors
Deleting partitions...
Partitioning (MBR)...
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (Large FAT32)...
Opened drive \\?\Volume{ab701598-653f-11e6-ba1a-d43d7edf2bc7} for write access
I/O boundary checks disabled
Size : 59.5 GB 124733440 sectors
Cluster size 32768 bytes, 512 Bytes Per Sector
Volume ID is 1b17:1b65
32 Reserved Sectors, 15223 Sectors per FAT, 2 FATs
1948483 Total clusters
1948482 Free Clusters
Clearing out 30542 sectors for reserved sectors, FATs and root cluster...
Initializing reserved sectors and FATs...
FAT #0 sector at address: 32
FAT #1 sector at address: 15255
Writing partition boot record...
Using Standard FAT32 partition boot record
Confirmed new volume has a primary FAT32 boot sector
Setting primary FAT32 boot sector for boot...
Confirmed new volume has a secondary FAT32 boot sector
Setting secondary FAT32 boot sector for boot...
Setting Label (This may take while)...
Format completed.
Writing master boot record...
Drive has a Zeroed Master Boot Record
Partition is already FAT32 LBA...
Set bootable USB partition as 0x80
Using Syslinux MBR
Found volume GUID \\?\Volume{ab701598-653f-11e6-ba1a-d43d7edf2bc7}\
Installing Syslinux 6.03...
Opened drive \\?\Volume{ab701598-653f-11e6-ba1a-d43d7edf2bc7} for write access
Using existing './rufus_files/syslinux-6.03/20160618/ldlinux.sys' ✓
Using existing './rufus_files/syslinux-6.03/20160618/ldlinux.bss' ✓
Successfully wrote 'ldlinux.sys'
Could not patch Syslinux files.
WARNING: This could be caused by your firewall having modifed downloaded content, such as 'ldlinux.sys'...
\\?\Volume{ab701598-653f-11e6-ba1a-d43d7edf2bc7}\ was already mounted as F:\
Re-mounted volume as 'F:' after error

Found USB 3.0 device 'Generic STORAGE DEVICE USB Device' (05E3:0749)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 7764, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x01FB025F
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 59.5 GB (63863521280 bytes)
  Start Sector: 2048, Boot: Yes
```

pbatard, 2016-11-15:
This is not an error. If you look at the next message you see that it says:
```
Extended version was not found, trying main version
```

So what happens is Rufus tries to first see if it can find the custom version that was used by the ISO, and if it doesn't, it falls back to the regular/main. Custom versions of Syslinux are only ever uploaded to the server if the main doesn't work. So the question becomes: Do you __actually__ see an issue with Ubuntu 16.10 during the early boot phase. If not, then it means that the main Syslinux version works just fine, and there is nothing to do.

As you can imagine, with tons of ISOs out there, using tons of versions of Syslinux, if I were to try to rebuild (because they would need to be rebuilt) and upload every single version out there to avoid the message about the fallback, I'd have to literally spend my live building and uploading Syslinux binaries on the server. That's not realistic.

So I'm just going to have to ask Rufus users not to take every single __normal__ warning as an error that needs to be addressed. Warnings can and do happen as part of the normal execution of an application, and there's no reason to freak out about them or consider that the application should be improved by silencing them.

pbatard, 2016-11-15:
On the other hand, I see that you get:
```
Could not patch Syslinux files.
WARNING: This could be caused by your firewall having modifed downloaded content, such as 'ldlinux.sys'...
```

Now, this is a genuine issue (sorry, I missed that), but as the log says, this could be caused by the firewall.

Could you please zip your `rufus_files` folder and mail it to me (pete@akeo.ie). I'm pretty sure that your downloads are being corrupted by an actor that is external to Rufus, but that will allow me to confirm it.

pbatard, 2016-11-15:
This may also be caused by USB flash driver you are using. I see that you have chosen to ignore the parts of the checklist that could really help identify what your issue really, so could you __PLEASE__ perform ALL of the following:
- [ ] I ran a bad blocks check, by clicking the "bad blocks" check box in Rufus, and confirmed that my USB is not defective
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive
  - [ ] Plugging the USB into a different port
  - [ ] Running Rufus on a different computer

These steps are not meant to annoy you but to __HELP__ troubleshoot the issue. If you chose not to perform them, it's a lot harder to help you out.

flakefree, 2016-11-16:
New log attached

- [x] I ran a bad blocks check, by clicking the "bad blocks" check box in Rufus, and confirmed that my USB is not defective
- [x] I also tried one or more of the following:
  - [x] Using a different USB drive
  - [x] Plugging the USB into a different port
  - [ ] Running Rufus on a different computer
- [x] If using an ISO image, I clicked on the `#` button (at the bottom of the Rufus interface), to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Log
---
```
Rufus version: 2.11.995
Windows version: Windows 10 64-bit (Build 14393)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta3
System locale ID: 0x0809
Will use default UI locale 0x0809
Found USB 2.0 device 'JetFlash Transcend 8GB USB Device' (8564:1000)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 951, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x02BABD22
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 7.3 GB (7821328384 bytes)
  Start Sector: 2048, Boot: Yes
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/isolinux/isolinux.cfg' for Syslinux
  Detected Syslinux version: 6.03/20160618 (from '/isolinux/isolinux.bin')
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'Ubuntu 16.10 amd64'
  Size: 1593835520 bytes
  Has a >64 chars filename
  Uses: Syslinux/Isolinux v6.03
  Uses: EFI
  Note: This ISO uses symbolic links, which will not be replicated due to file system limitations.
  Because of this, some features from this image may not work...
Using image: ubuntu-16.10-desktop-amd64.iso
Downloading 'ldlinux.sys' from http://rufus.akeo.ie/files/syslinux-6.03/20160618/ldlinux.sys
Unable to access file: 404
Extended version was not found, trying main version
Downloading 'ldlinux.sys' from http://rufus.akeo.ie/files/syslinux-6.03/ldlinux.sys
File length: 68599 bytes
Successfully downloaded 'ldlinux.sys'
Downloading 'ldlinux.bss' from http://rufus.akeo.ie/files/syslinux-6.03/20160618/ldlinux.bss
Unable to access file: 404
Extended version was not found, trying main version
Downloading 'ldlinux.bss' from http://rufus.akeo.ie/files/syslinux-6.03/ldlinux.bss
File length: 512 bytes
Successfully downloaded 'ldlinux.bss'

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE3 for write access
Will use 'G:' as volume mountpoint
I/O boundary checks disabled
Analyzing existing boot records...
Drive has a Syslinux Master Boot Record
Volume has an unknown Partition Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Deleting partitions...
Partitioning (MBR)...
[... 79 lines cut ...]
Using existing './rufus_files/syslinux-6.03/20160618/ldlinux.sys' ✓
Using existing './rufus_files/syslinux-6.03/20160618/ldlinux.bss' ✓
Unable to create 'ldlinux.sys'
\\?\Volume{4ffa7553-aad5-11e6-ba54-d43d7edf2bc7}\ was already mounted as G:\
Re-mounted volume as 'G:' after error

Found USB 2.0 device 'JetFlash Transcend 8GB USB Device' (8564:1000)
Found USB 2.0 device 'RIM BlackBerry SD USB Device' (0FCA:8004)
Device eliminated because it appears to contain no media
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 951, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x04C96441
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 7.3 GB (7821328384 bytes)
  Start Sector: 2048, Boot: Yes

Computing checksum for 'C:\Users\Edward\Downloads\ubuntu-16.10-desktop-amd64.iso'...
  MD5:	 3f50877c05121f7fd8544bef2d722824
  SHA1:	 aee0709d07d43494d7aa13457b4e255b0a12d1cf
  SHA256: 4405c37d61b5cac6c89eaf379c035058ed7db8594abd209337276c7c4556787e
```

pbatard, 2016-11-16:
Thanks.

What I'm seeing are inconsistent errors during syslinux installation. Sometimes you get:
```
Successfully wrote 'ldlinux.sys'
Could not patch Syslinux files.
```
and other times you get:
```
Unable to create 'ldlinux.sys'
```

This leads me to think that the problem is not linked to Rufus, else I would expect to consistently get the same error. 
The 2 possibilities I see are:
* You are using security or other software on your computer, that is interfering with Rufus operations. There's a list of such software [at the end of the FAQ](https://github.com/pbatard/rufus/wiki/FAQ#Software), but please bear in mind that it isn't exhaustive.
* There is something special about your computer hardware that makes the installation of Syslinux fail.

Considering this, I would really like to hear what happens when you try from a another computer, preferably a relatively clean one, with a fresh installation of Windows and few applications installed.

But even then, I'm afraid this is not an issue I or anybody else seems to be able to replicate, so I don't think there's much I can do to fix it. I have tested the same ISO as the one you are using on a computer with the same version of Windows 10 as you use, and found no issues. The only thing I can advise you to try is write Ubuntu in DD mode. Remember how Rufus prompts you whether you want to write in ISO or DD mode, and how you should retry in the other mode if you get an issue in the first mode? Well, that's what you should try here, especially as DD mode doesn't require Syslinux installation.

flakefree, 2016-11-16:
Ok Thanks, I will blame it on InterGuard, the current protection software on my machine, as I don’t think that ESET antivirus would cause an issue, it has never in the past.

 

Many thanks

pbatard, 2016-11-18:
I'm gonna close this issue now (after making the errors more explicit in the title so that if someone encounters the same issue, they'll find this one and may provide some help as to what might be the underlying problem). Hopefully, DD mode should be usable as a workaround.

closed 2016-11-18 by pbatard

Mindriel, 2017-12-02:
Hi there.
I was having the same issue today. Thank You for the note about how some security software may interfere.
In my case it was K9.
Below I attach zipped rufus_files folder that were created when K9 was enabled on my machine and then when it wasn't.
[rufus_files-K9enabled.zip](https://github.com/pbatard/rufus/files/1524156/rufus_files-K9enabled.zip)
[rufus_files-K9disabled.zip](https://github.com/pbatard/rufus/files/1524157/rufus_files-K9disabled.zip)

pbatard, 2017-12-02:
When in doubt, just look at the files with a text editor. In your case, you would have seen that __this__ is the content of the files:
```
K9 Web Protection Alert
Filtering Alert
Category Blocked
The site you tried to visit belongs to a category that your computer is set to block.
rufus.akeo.ie/files/syslinux-6.03/20170128/ldlinux.sys is blocked because it is currently categorized as:

Phishing
    Sites that are designed to appear as a legitimate bank or retailer with the intent to fraudulently capture sensitive data (i.e. credit card numbers, pin numbers). 

Computers / Internet
    Sites that sponsor or provide information on computers, technology, the Internet and technology-related organizations and companies. 

Go Back Return to the previous page.
```

rrifaldiu, 2020-04-12:
I was getting the same error and resolve it by removing rufus files folder and then download the syslinux again. I guess it's caused by the corrupted syslinux file