pbatard/rufus #859: The file system currently selected can not be used with this type of iso.

Norpy, 2016-11-25:
Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues, using keywords relevant to my problem.
- [x] I clicked the `Log` button in Rufus and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [x] I ran a bad blocks check, by clicking the "bad blocks" check box in Rufus, and confirmed that my USB is not defective
- [x] I also tried one or more of the following:
  - [x] Using a different USB drive
  - [x] Plugging the USB into a different port
  - [x] Running Rufus on a different computer
- [x] If using an ISO image, I clicked on the `#` button (at the bottom of the Rufus interface), to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

**Issue description**
-----------------

Error message

The file system currently selected can not be used with this type of iso. Please select a different file system or use a different ISO.

Screenshot of errormessage:

https://s22.postimg.org/ubx0vxcup/Rufus_Error.jpg

Works fine in version  2.10.973, presented with the errormessage above for version 2.11.995.

The iso-image is created with System Center Build 1606.

**FULL Log**
-----------------

**Log from non-working version** 

Rufus version: 2.11.995
Windows version: Windows 10 64-bit (Build 14393)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta3
System locale ID: 0x0409
Will use default UI locale 0x0409
Found USB 3.0 device 'Kingston DataTraveler 3.0 USB Device' (0951:1666)
Using autorun.inf label for drive D: 'Configuration Manager'
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 3774, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x5D604307
Drive has a Windows 7 Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 28.9 GB (31042568192 bytes)
  Start Sector: 2048, Boot: Yes
Scanning image...
ISO analysis:
  Image is an UDF image
Disk image analysis:
  Image does not have an x86 Master Boot Record
ISO label: 'Configuration Manager'
  Size: 505169920 bytes
  Uses: EFI
  Uses: Bootmgr
Using image: I2X_Boot_161030.iso

Computing checksum for 'C:\Users\aa217\Desktop\I2X_Boot_161030.iso'...
  MD5:	 d223e003a9086f6d582fc33d42feae94
  SHA1:	 474bc8d87cb65e4cfcff2fafec824e8798a91371
  SHA256: 9b3174ec57e2b746ff5f9092eec18d2c40a36d5c78926f9b236c2e81dd0ccc23

**Log from working version**

Rufus version: 2.10.973
Windows version: Windows 10 64-bit (Build 14393)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta3
System locale ID: 0x0409
Will use default UI locale 0x0409
Found USB 3.0 device 'Kingston DataTraveler 3.0 USB Device' (0951:1666)
Using autorun.inf label for drive D: 'Configuration Manager'
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 3774, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x5D604307
Drive has a Windows 7 Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 28.9 GB (31042568192 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes
Scanning image...
ISO analysis:
  Image is an UDF image
Disk image analysis:
  Image does not have an x86 Master Boot Record
ISO label: 'Configuration Manager'
  Size: 505169920 bytes
  Uses: EFI
  Uses: Bootmgr
Using image: I2X_Boot_161030.iso

Computing checksum for 'C:\Users\aa217\Desktop\I2X_Boot_161030.iso'...
  MD5:	 d223e003a9086f6d582fc33d42feae94
  SHA1:	 474bc8d87cb65e4cfcff2fafec824e8798a91371
  SHA256: 9b3174ec57e2b746ff5f9092eec18d2c40a36d5c78926f9b236c2e81dd0ccc23

pbatard, 2016-11-25:
Yes, this was a __bug__ in 2.10 and earlier that allowed the selection of FAT32 for Windows based images, as opposed to what was intended. As per the Changelog on the Homepage, this was fixed in 2.11:
```
Fix an issue that enabled FAT32 file system selection on some Windows images
```
I don't suspect you will encounter an issue by selecting NTFS, which is what you should do (and what Rufus should select by default - at least, that's what Rufus advises you to do in the message), so I will close this issue.

closed 2016-11-25 by pbatard

Norpy, 2016-11-25:
The problem is that some devices refuses to boot when NTFS is selected. When selecting boot-media in bios, the screen just blinks and returns to the menu.

pbatard, 2016-11-25:
Can you elaborate on these devices (manufacturer and exact model name)? Are they BIOS or UEFI based?

Right now, I am not aware of any BIOS based device that should not be able to boot NTFS. If they do, then that's a BIOS issue that needs to be fixed by the manufacturer, and considering the advantages of enforcing NTFS for BIOS boot and Windows images, I don't think I want to reinstate FAT32 support just to work around a handful of BIOSes that were designed poorly.

As to UEFI devices, they should work too, as long as you disable Secure Boot.

Norpy, 2016-11-25:
Disabling Secure Boot is not an option for us.

So we will stick with version 2.10

Thanks for clarifying.

pbatard, 2016-11-25:
As you wish. I should point out (because this is a common misconception) that it's not because you installed Windows with Secure Boot disabled that:
1. You won't be able to run Windows with Secure Boot once the installation is completed (you can re-enable it afterwards, and it will be exactly as if Secure Boot had been enabled the whole time)
2. Your system will be any less trustworthy from running the installation process with Secure Boot disabled (as long as you used an official Microsoft ISO or created your image with official Microsoft tools and data).

A lot of people seem to be under the impression that, if you install Windows with Secure Boot disabled, then it's game over for ensuring the system's security, which is wrong. All Secure Boot does is delegate the task of ensuring "trustworthiness" to an automated process. But that doesn't mean an installation running with Secure Boot disabled, and where the person ensured that the source they obtained for installation were trustworthy (e.g. by checking the SHA-1 of their Windows ISO, and checking the digital signature of the application that converts their ISO to USB, if they use one), is any less trustworthy.

Finally, if your interest is only with booting UEFI systems with Secure Boot enabled, then you do realize that you don't need Rufus. You can just format your USB to FAT32 and then copy the ISO content wholesale (which is pretty much what Rufus would do if using FAT32).

Norpy, 2016-11-25:
Thanks again for the elaborate explanation.

Since Rufus is already know by most of our technicians, it’s easy for them to create the odd boot-media now and then.

When mas-creating sticks, we have a PowerShell-script, that has a fully automated process with Robocop and diskpart to do the necessary configuration. 

All the major manufactures, Dell, HP and Lenovo has Secure Boot turned on by default, so it’s easier to keep it on. 

If there is a manual step required to enable Secure Boot after the computer is finished, there is a chance it will be forgotten. 

Ofc, it all boils down to laziness.