pbatard/rufus #860: add Support for Hebrew

pinim, 2016-11-30:
Checklist
---------
- [V] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [V] I performed a search in the issue tracker for similar issues, using keywords relevant to my problem.
- [V] I clicked the `Log` button in Rufus and copy/pasted the log into the line that says `<FULL LOG>` below.
- [V] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [V] I ran a bad blocks check, by clicking the "bad blocks" check box in Rufus, and confirmed that my USB is not defective
- [V] I also tried one or more of the following:
  - [V] Using a different USB drive
  - [V] Plugging the USB into a different port
  - [V] Running Rufus on a different computer
- [V] If using an ISO image, I clicked on the `#` button (at the bottom of the Rufus interface), to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<I want to help to translate into Hebrew, How Can I help it?>

Log
---
```
<FULL LOG>
```

pbatard, 2016-11-30:
Please use existing #204 to register your interest and go for it. The way to proceed is detailed [here](https://github.com/pbatard/rufus/wiki/Localization). Also you might want to pay attention to how the Arabic or Persian translation are done, as right-to-left language require a few quirks when mixing English and non English. 

Now, since this is a duplicate of #204, I'm going to close this issue.

By the way, thanks for volunteering!
But please be mindful that translating Rufus is __a lot__ more work than people seem to imagine, and, surprisingly, all of the volunteers I've had for a Hebrew translation (and there have been quite a few now) appear to have given up after a while. Let's just hope you'll find motivated enough to complete it, as Hebrew is the very last language Rufus still needs to be translated into...

closed 2016-11-30 by pbatard

NSbuilder, 2016-12-01:
Here is my loc file - in case someone needs it: http://www77.zippyshare.com/v/tNc7e61O/file.html

pbatard, 2016-12-01:
Much appreciated!! I had a look at it and can see that you certainly spent a lot of time on this, so thank you very much.

I have now uploaded a slightly modified version of your translation [here](https://rufus.akeo.ie/testing/rufus_loc.7z), for any good soul who would like to complete your work.

pinim, 2016-12-03:
,I finished the translation

pbatard, 2016-12-03:
Can you please e-mail it to me? pete@akeo.ie

pinim, 2016-12-03:
I sent it

pbatard, 2016-12-03:
Got it. I'll take a look at it and let you know if there's anything that needs to be modified (this might take a couple of days).

Much appreciate the work you and NSbuilder did on that. At last, all the languages I wanted to be supported will be complete, and I'm sure many Hebrew speaking people will be very thankful for your efforts!

pbatard, 2016-12-05:
I have reviewed the Hebrew translation and found no issue 😄. It will be made available in the next version of Rufus.

Thanks again to the both of you for working on this!

NSbuilder, 2017-01-17:
I sent you the updated translation.