pbatard/rufus #861: Cannot format USB Key after Rufus fail

MacGeorges, 2016-12-02:
Checklist
---------
- [X] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [X] I performed a search in the issue tracker for similar issues, using keywords relevant to my problem.
- [X] I clicked the `Log` button in Rufus and copy/pasted the log into the line that says `<FULL LOG>` below.
- [X] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [X] I ran a bad blocks check, by clicking the "bad blocks" check box in Rufus, and confirmed that my USB is not defective
- [X] I also tried one or more of the following:
  - [X] Using a different USB drive
  - [X] Plugging the USB into a different port
  - [X] Running Rufus on a different computer
- [X] If using an ISO image, I clicked on the `#` button (at the bottom of the Rufus interface), to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Hello !

I used Rufus to creat a Windows 10 installation device.
I first tried with a Kingston 16Gb USB key. This key have been buyed new for this. never used for anything before. I just opened the box, plugged the key on the laptop and lunched Rufus.

The process failed (I sadly don't have this log) and then I can't format the key anymore. I cannot add, modify or remove the data inside. (data seems to be parts of the windows 10 iso)
I tried to format using windows explorer, Computer Management, Diskpart with same result: cannot format, unknow error.
I tried to format from a Ubuntu live CD, with no success.

Then I used Rufus with an 32Gb SD Card, succefully.

As a final tentative to save my beloved (XD) usb drive, I used Rufus to format the key.
The process failed again, and I join the log to this message. (I ran a fast format, then a bad block check)

The worst is that I know it's just software issue. My key is not dead, but I can't use it. Makes it a zombie out of it ?

Best Regards,

MacGeorges

Log
---
```
Rufus version: 2.11.995
Windows version: Windows 10 64-bit (Build 14393)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta3
System locale ID: 0x040C
Will use default UI locale 0x040C
Found USB 2.0 device 'Kingston DataTraveler 2.0 USB Device' (0951:1665)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 1881, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00510034
Drive has a Rufus Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 14.4 GB (15478548480 bytes)
  Start Sector: 2048, Boot: Yes

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE1 for write access
Will use 'D:' as volume mountpoint
I/O boundary checks disabled
Analyzing existing boot records...
Drive has a Rufus Master Boot Record
Volume has an unknown Partition Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Deleting partitions...
Partitioning (MBR)...
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (NTFS)...
Using cluster size: 4096 bytes
Quick format was selected
Creating file system...
Error while formatting
Format error: Undetermined error while formatting.
Re-mounted volume as 'D:' after error

Found USB 2.0 device 'Kingston DataTraveler 2.0 USB Device' (0951:1665)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 1881, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00510034
Drive has a Rufus Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 14.4 GB (15478548480 bytes)
  Start Sector: 2048, Boot: Yes

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE1 for write access
Will use 'D:' as volume mountpoint
I/O boundary checks disabled
Analyzing existing boot records...
Drive has a Rufus Master Boot Record
Volume has an unknown Partition Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Deleting partitions...
Bad Blocks: Checking from block 0 to 30233587
Bad Blocks: Using offset 0 for fake device check
Bad Blocks: Writing test pattern 0xAA
Bad Blocks: Reading and comparing
Bad Blocks: 0
Bad Blocks: 1
Bad Blocks: 2
Bad Blocks: 3
Bad Blocks: 4
Bad Blocks: 5
Bad Blocks: 6
Bad Blocks: 7
Bad Blocks: 8
Bad Blocks: 9
Bad Blocks: 10
Bad Blocks: 11
Bad Blocks: 12
Bad Blocks: 13
Bad Blocks: 14
Bad Blocks: 15
Bad Blocks: 16
Bad Blocks: 17
Bad Blocks: 18
Bad Blocks: 19
Bad Blocks: 20
Bad Blocks: 21
Bad Blocks: 22
Bad Blocks: 23
Bad Blocks: 24
Bad Blocks: 25
Bad Blocks: 26
Bad Blocks: 27
Bad Blocks: 28
Bad Blocks: 29
Bad Blocks: 30
Bad Blocks: 31
Bad Blocks: 32
Bad Blocks: 33
Bad Blocks: 34
Bad Blocks: 35
Bad Blocks: 36
Bad Blocks: 37
Bad Blocks: 38
Bad Blocks: 39
Bad Blocks: 40
Bad Blocks: 41
Bad Blocks: 42
Bad Blocks: 43
Bad Blocks: 44
Bad Blocks: 45
Bad Blocks: 46
Bad Blocks: 47
Bad Blocks: 48
Bad Blocks: 49
Bad Blocks: 50
Bad Blocks: 51
Bad Blocks: 52
Bad Blocks: 53
Bad Blocks: 54
Bad Blocks: 55
Bad Blocks: 56
Bad Blocks: 57
Bad Blocks: 58
Bad Blocks: 59
Bad Blocks: 60
Bad Blocks: 61
Bad Blocks: 62
Bad Blocks: 63
Bad Blocks: 64
Bad Blocks: 65
Bad Blocks: 66
Bad Blocks: 67
Bad Blocks: 68
Bad Blocks: 69
Bad Blocks: 70
Bad Blocks: 71
Bad Blocks: 72
Bad Blocks: 73
Bad Blocks: 74
Bad Blocks: 75
Bad Blocks: 76
Bad Blocks: 77
Bad Blocks: 78
Bad Blocks: 79
Bad Blocks: 80
Bad Blocks: 81
Bad Blocks: 82
Bad Blocks: 83
Bad Blocks: 84
Bad Blocks: 85
Bad Blocks: 86
Bad Blocks: 87
Bad Blocks: 88
Bad Blocks: 89
Bad Blocks: 90
Bad Blocks: 91
Bad Blocks: 92
Bad Blocks: 93
Bad Blocks: 94
Bad Blocks: 95
Bad Blocks: 96
Bad Blocks: 97
Bad Blocks: 98
Bad Blocks: 99
Bad Blocks: 100
Bad Blocks: 101
Bad Blocks: 102
Bad Blocks: 103
Bad Blocks: 104
Bad Blocks: 105
Bad Blocks: 106
Bad Blocks: 107
Bad Blocks: 108
Bad Blocks: 109
Bad Blocks: 110
Bad Blocks: 111
Bad Blocks: 112
Bad Blocks: 113
Bad Blocks: 114
Bad Blocks: 115
Bad Blocks: 116
Bad Blocks: 117
Bad Blocks: 118
Bad Blocks: 119
Bad Blocks: 120
Bad Blocks: 121
Bad Blocks: 122
Bad Blocks: 123
Bad Blocks: 124
Bad Blocks: 125
Bad Blocks: 126
Bad Blocks: 127
Bad Blocks: 128
Bad Blocks: 129
Bad Blocks: 130
Bad Blocks: 131
Bad Blocks: 132
Bad Blocks: 133
Bad Blocks: 134
Bad Blocks: 135
Bad Blocks: 136
Bad Blocks: 137
Bad Blocks: 138
Bad Blocks: 139
Bad Blocks: 140
Bad Blocks: 141
Bad Blocks: 142
Bad Blocks: 143
Bad Blocks: 144
Bad Blocks: 145
Bad Blocks: 146
Bad Blocks: 147
Bad Blocks: 148
Bad Blocks: 149
Bad Blocks: 150
Bad Blocks: 151
Bad Blocks: 152
Bad Blocks: 153
Bad Blocks: 154
Bad Blocks: 155
Bad Blocks: 156
Bad Blocks: 157
Bad Blocks: 158
Bad Blocks: 159
Bad Blocks: 160
Bad Blocks: 161
Bad Blocks: 162
Bad Blocks: 163
Bad Blocks: 164
Bad Blocks: 165
Bad Blocks: 166
Bad Blocks: 167
Bad Blocks: 168
Bad Blocks: 169
Bad Blocks: 170
Bad Blocks: 171
Bad Blocks: 172
Bad Blocks: 173
Bad Blocks: 174
Bad Blocks: 175
Bad Blocks: 176
Bad Blocks: 177
Bad Blocks: 178
Bad Blocks: 179
Bad Blocks: 180
Bad Blocks: 181
Bad Blocks: 182
Bad Blocks: 183
Bad Blocks: 184
Bad Blocks: 185
Bad Blocks: 186
Bad Blocks: 187
Bad Blocks: 188
Bad Blocks: 189
Bad Blocks: 190
Bad Blocks: 191
Bad Blocks: 192
Bad Blocks: 193
Bad Blocks: 194
Bad Blocks: 195
Bad Blocks: 196
Bad Blocks: 197
Bad Blocks: 198
Bad Blocks: 199
Bad Blocks: 200
Bad Blocks: 201
Bad Blocks: 202
Bad Blocks: 203
Bad Blocks: 204
Bad Blocks: 205
Bad Blocks: 206
Bad Blocks: 207
Bad Blocks: 208
Bad Blocks: 209
Bad Blocks: 210
Bad Blocks: 211
Bad Blocks: 212
Bad Blocks: 213
Bad Blocks: 214
Bad Blocks: 215
Bad Blocks: 216
Bad Blocks: 217
Bad Blocks: 218
Bad Blocks: 219
Bad Blocks: 220
Bad Blocks: 221
Bad Blocks: 222
Bad Blocks: 223
Bad Blocks: 224
Bad Blocks: 225
Bad Blocks: 226
Bad Blocks: 227
Bad Blocks: 228
Bad Blocks: 229
Bad Blocks: 230
Bad Blocks: 231
Bad Blocks: 232
Bad Blocks: 233
Bad Blocks: 234
Bad Blocks: 235
Bad Blocks: 236
Bad Blocks: 237
Bad Blocks: 238
Bad Blocks: 239
Bad Blocks: 240
Bad Blocks: 241
Bad Blocks: 242
Bad Blocks: 243
Bad Blocks: 244
Bad Blocks: 245
Bad Blocks: 246
Bad Blocks: 247
Bad Blocks: 248
Bad Blocks: 249
Bad Blocks: 250
Bad Blocks: 251
Bad Blocks: 252
Bad Blocks: 253
Bad Blocks: 254
Bad Blocks: 255
Too many bad blocks, aborting test
Bad Blocks: Check completed, 256 bad blocks found. (0/0/256 errors)
Re-mounted volume as 'D:' after error

Found USB 2.0 device 'Kingston DataTraveler 2.0 USB Device' (0951:1665)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 1881, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00510034
Drive has a Rufus Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 14.4 GB (15478548480 bytes)
  Start Sector: 2048, Boot: Yes

```

pbatard, 2016-12-02:
> - [x] I ran a bad blocks check, by clicking the "bad blocks" check box in Rufus, and confirmed that my USB is not defective

Your log __disproves__ that. Your USB is defective if it reports bad blocks. It doesn't matter if it's brand new or anything. As a matter of fact, it is possible that you are simply the victim of a "fake" USB key, where you were sold a key that promised X capacity, but has a lot less capacity and was instead modified by a malicious seller to report much more capacity than it has. Unfortunately this is a relatively common practice (especially with cheap sellers), to fool people into believing that the key works... until it loops over its __real__ capacity, in which case important data such as its firmware gets erased, which usually renders the key inoperable.

> The worst is that I know it's just software issue

I'm sorry but your log __clearly demonstrates__ otherwise. It is a pure hardware issue with your key.

Unless the bad blocks check reports no error, there's nothing that can be done for your key, and you will need to contact the seller to get it exchanged.

As such, I will close this issue.

closed 2016-12-02 by pbatard

MacGeorges, 2016-12-02:
Damn....

Okay, to be honest I buyed this key in Serbia....

So it's shurely this. I just lost 6€.

thanks a lot for your fast response !