pbatard/rufus #862: Unknown error chkdsk cannot continue in read-only mode

hichristiangoodbye, 2016-12-02:
Checklist
---------
- [ ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ ] I performed a search in the issue tracker for similar issues, using keywords relevant to my problem.
- [ ] I clicked the `Log` button in Rufus and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking the "bad blocks" check box in Rufus, and confirmed that my USB is not defective
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive
  - [ ] Plugging the USB into a different port
  - [ ] Running Rufus on a different computer
- [ ] If using an ISO image, I clicked on the `#` button (at the bottom of the Rufus interface), to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<Please describe your issue here>

Log
---
```
<FULL LOG>
```

hichristiangoodbye, 2016-12-02:
I tried to boot it from another laptop and it worked but I'm scared to proceed as to I don't know if it was complete and was ready for Windows 10 formatting of other devices or was it just able to boot it.

pbatard, 2016-12-02:
duplicate of #379. I don't think you should worry about it. Next time, __please__ follow the checklist.

closed 2016-12-02 by pbatard