pbatard/rufus #864: Unable to create bootable disks from TrueCrypt/VeraCrypt Rescue Disk Images

vinzBad, 2016-12-06:
Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues, using keywords relevant to my problem.
- [x] I clicked the `Log` button in Rufus and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- none

Issue description
-----------------
I'm unable to bring a TrueCrypt RescueDisk Image on an USB Stick. When I burn the ISO-File on a CD its bootable and I can use it restore key headers / decrypt drives. We have many TrueCrypt ISOs, so it would be nice to stop wasting CDs :) 

These entries are printed to the log when I select a TrueCrypt Rescue Disk. 
```
Scanning image...
ISO analysis:
  Image is an ISO9660 image
libcdio: from_733: broken byte order
Could not access directory
Disk image analysis:
  Image does not have an x86 Master Boot Record
Failed to scan image
```
I get the same error with Veracrypt Rescue Disks. (Fork of TrueCrypt as TrueCrypt is no longer maintained)

If you want to take a look at this issue, you can download VeraCrypt from here: https://veracrypt.codeplex.com/  I can create you an exemplary ISO if you dont want to install it ;)

Full Log
---
```
Rufus version: 2.11.995
Windows version: Windows 7 SP1 64-bit
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta3
System locale ID: 0x0407
Will use default UI locale 0x0407
Found card reader device ' SD SCSI Disk Device'
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 3946, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00000000
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 30.2 GB (32458670080 bytes)
  Start Sector: 8192, Boot: No
Checking for Rufus updates...
Checking release channel...
No new release version found.
Found card reader device ' SD SCSI Disk Device'
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 3946, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00000000
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 30.2 GB (32458670080 bytes)
  Start Sector: 8192, Boot: No
Found card reader device ' SD SCSI Disk Device'
Found USB 3.0 device 'SanDisk Extreme USB Device' (0781:5580)
2 devices found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 3946, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00000000
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 30.2 GB (32458670080 bytes)
  Start Sector: 8192, Boot: No
2 devices found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 1907, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00000000
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 14.6 GB (15693647872 bytes)
  Start Sector: 32, Boot: No
Scanning image...
ISO analysis:
  Image is an ISO9660 image
libcdio: from_733: broken byte order
Could not access directory
Disk image analysis:
  Image does not have an x86 Master Boot Record
Failed to scan image
Found card reader device ' SD SCSI Disk Device'
Found USB 3.0 device 'SanDisk Extreme USB Device' (0781:5580)
2 devices found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 1907, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00000000
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 14.6 GB (15693647872 bytes)
  Start Sector: 32, Boot: No

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE3 for write access
Will use 'E:' as volume mountpoint
I/O boundary checks disabled
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Volume has an unknown Partition Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 32 sectors
Deleting partitions...
Partitioning (MBR)...
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (FAT32)...
Using cluster size: 8192 bytes
Quick format was selected
Creating file system...
Format completed.
Writing master boot record...
Drive has a Zeroed Master Boot Record
Partition is already FAT32 LBA...
Set bootable USB partition as 0x80
Using Windows 7 MBR
Found volume GUID \\?\Volume{6455ec72-bb7e-11e6-9718-acfdce4c432a}\
Opened drive \\?\Volume{6455ec72-bb7e-11e6-9718-acfdce4c432a} for write access
Writing partition boot record...
Using FreeDOS FAT32 partition boot record
Confirmed new volume has a primary FAT32 boot sector
Setting primary FAT32 boot sector for boot...
Confirmed new volume has a secondary FAT32 boot sector
Setting secondary FAT32 boot sector for boot...
Successfully remounted Volume{6455ec72-bb7e-11e6-9718-acfdce4c432a}\ on E:\
Copying DOS files...
Successfully wrote 'E:\COMMAND.COM' (65814 bytes)
Successfully wrote 'E:\KERNEL.SYS' (45494 bytes)
Successfully wrote 'E:\LOCALE\DISPLAY.EXE' (3648 bytes)
Successfully wrote 'E:\LOCALE\KEYB.EXE' (11446 bytes)
Successfully wrote 'E:\LOCALE\MODE.COM' (16254 bytes)
Successfully wrote 'E:\LOCALE\KEYBOARD.SYS' (40360 bytes)
Successfully wrote 'E:\LOCALE\KEYBRD2.SYS' (29750 bytes)
Successfully wrote 'E:\LOCALE\KEYBRD3.SYS' (32178 bytes)
Successfully wrote 'E:\LOCALE\KEYBRD4.SYS' (13105 bytes)
Successfully wrote 'E:\LOCALE\ega.cpx' (6464 bytes)
Successfully wrote 'E:\LOCALE\ega2.cpx' (7059 bytes)
Successfully wrote 'E:\LOCALE\ega3.cpx' (5469 bytes)
Successfully wrote 'E:\LOCALE\ega4.cpx' (4431 bytes)
Successfully wrote 'E:\LOCALE\ega5.cpx' (7217 bytes)
Successfully wrote 'E:\LOCALE\ega6.cpx' (7409 bytes)
Successfully wrote 'E:\LOCALE\ega7.cpx' (5387 bytes)
Successfully wrote 'E:\LOCALE\ega8.cpx' (6973 bytes)
Successfully wrote 'E:\LOCALE\ega9.cpx' (5785 bytes)
Successfully wrote 'E:\LOCALE\ega10.cpx' (5543 bytes)
Successfully wrote 'E:\LOCALE\ega11.cpx' (7228 bytes)
Successfully wrote 'E:\LOCALE\ega12.cpx' (8119 bytes)
Successfully wrote 'E:\LOCALE\ega13.cpx' (6281 bytes)
Successfully wrote 'E:\LOCALE\ega14.cpx' (7758 bytes)
Successfully wrote 'E:\LOCALE\ega15.cpx' (6458 bytes)
Successfully wrote 'E:\LOCALE\ega16.cpx' (7793 bytes)
Successfully wrote 'E:\LOCALE\ega17.cpx' (8929 bytes)
Successfully wrote 'E:\LOCALE\ega18.cpx' (5158 bytes)
Windows KBID 0x00000407
Will use DOS keyboard 'gr' [German]
Will use codepage 858 [Western-European (Euro)]
Successfully wrote 'CONFIG.SYS'
Successfully wrote 'AUTOEXEC.BAT'
Finalizing, please wait...
Created: E:\autorun.inf
Created: E:\autorun.ico
Successfully remounted Volume{6455ec72-bb7e-11e6-9718-acfdce4c432a}\ on E:\

Found card reader device ' SD SCSI Disk Device'
Found USB 3.0 device 'SanDisk Extreme USB Device' (0781:5580)
Using autorun.inf label for drive E: '16GB'
2 devices found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 1907, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x019925A1
Drive has a Windows 7 Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 14.6 GB (15692615680 bytes)
  Start Sector: 2048, Boot: Yes
Scanning image...
ISO analysis:
  Image is an ISO9660 image
libcdio: from_733: broken byte order
Could not access directory
Disk image analysis:
  Image does not have an x86 Master Boot Record
Failed to scan image
Found card reader device ' SD SCSI Disk Device'
Found USB 3.0 device 'SanDisk Extreme USB Device' (0781:5580)
Using autorun.inf label for drive E: '16GB'
2 devices found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 1907, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x019925A1
Drive has a Windows 7 Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 14.6 GB (15692615680 bytes)
  Start Sector: 2048, Boot: Yes

```

pbatard, 2016-12-06:
Well, the first thing you need to realize is that optical boot and USB boot rely on completely different methods, and anybody can come along and created their own custom bootable ISO. So, for every new _El Torito_ ISO bootloader that exists out there, a way to properly convert it to USB (which usually is a lot more tricky than people imagine) has to be programmed into Rufus.

As you may expect, trying to handle every new custom method out there (provided that they can be converted, which isn't always a given!) is not something that I'm willing to do, as I'd literally have to spend my time on it, so Rufus tends to only support ISOs that use common boot methods... and it looks to me like the VeraCryp developers chose not to go with one of those.

I can try to have a look, but:
1. Unless it's _super straightforward_, it's very likely that I'm just going to tell you that I don't think adding support for VeraCrypt rescue ISOs in Rufus is worth it, as I have to weight the time I'd have to spend adding a new method (if feasible) vs the proportion of Rufus users who will have use for it.
2. I'll need you to upload a rescue ISO somewhere that I can test (as I tried to play with VC, but I don't think I want to go as far as encrypting a whole system partition, just to get the software to create a rescue ISO, which seems to be the only way).

0nesAndZer0s, 2016-12-10:
I believe this may be of some use to the OP.  [https://sourceforge.net/projects/veracrypt/files/Contributions/](url) Within this Contributions folder you will find a zip file titled "VeraCryptUsbRescueDisk.zip", that contains all that is needed to create a bootable recovery USB rescue disk.

vinzBad, 2016-12-13:
Thanks @0nesAndZer0s 
@pbatard I'll try to create a RescueDisk ISO in the coming days, maybe you're motivated to take a look at it over the holidays ;)

pbatard, 2016-12-13:
Thanks for that! As you may guess, this isn't very high priority on my list of things to do right now, but if you produce an ISO, I'll definitely take a look.

pbatard, 2016-12-28:
@vinzBad, 
Are you still planning to create and upload an ISO somewhere?
If not, I will just close this issue until someone can provide one.

pbatard, 2017-01-02:
I'm going to close this issue, and will reopen it when/if someone provides a RescueDisk bootable ISO.

closed 2017-01-02 by pbatard

vinzBad, 2017-02-16:
Hi @pbatard I was finally able to create a dummy [TrueCrypt Rescue Disk](https://github.com/pbatard/rufus/files/779269/TrueCrypt.Rescue.Disk.zip), Password is `test`.

pbatard, 2017-02-16:
I had a look at it, and I'm afraid that neither Rufus not 7-zip are able to handle the content from the ISO (have you tried extracting the data from your ISO in 7-zip) &rarr; this is __NOT__ a valid ISO9660/UDF image. As such it will not be supported, as this isn't an real ISO.

vinzBad, 2017-02-16:
Ok :(  Thanks for checking it out! I have no clue how this works. 
There is this [guide ](http://stdout-dev-null.blogspot.de/2010/02/truecrypt-rescue-disk-on-usb.html) on creating a bootable truecrypt usb drive. Maybe something similar can be done in rufus?

pbatard, 2017-02-16:
> Maybe something similar can be done in rufus?

The guide just sets up virtual CD emulation using Grub4DOS. I'm afraid I have no plans to add such a mode, as it will create more issues (by encouraging people to create all kind of uncommon ISO images, which will result in all sorts of weird boot issues that I'll then be asked to troubleshoot, instead of trying to stick with tried and tested boot methods and ISO file systems, __which are really not that had to implement__).

I would strongly encourage you to alert the TrueCrypt people to the fact that their "iso"'s are __NOT__ ISO compliant. They chose to deviate from the ISO9660 specs when it comes to the layout of their file systems, so they shouldn't expect __others__ to spend valuable time trying to work with data created by people who think specs should not apply to them.

vinzBad, 2017-02-16:
Okay, I'll check what TrueCrypts successor VeraCrypt is doing. AFAIK they have two rescue disk types one for UEFI/GPT and one for legacy BIOS/MBR. The legacy one probably has the same problems as the TrueCrypt Rescue Disk I submitted. The UEFI/GPT is just a .ZIP one has to unpack on a FAT-formatted USB-Stick. 

Thanks for helping me out! :) Luckily we don't create new BIOS/MBR-Installations, but Rufus-Support for our large collection of legacy system, would've been nice.