pbatard/rufus #867: Undetermined error while formatting please help

jafarxd, 2016-12-13:
Checklist
---------
- [ ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ ] I performed a search in the issue tracker for similar issues, using keywords relevant to my problem.
- [ ] I clicked the `Log` button in Rufus and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking the "bad blocks" check box in Rufus, and confirmed that my USB is not defective
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive
  - [ ] Plugging the USB into a different port
  - [ ] Running Rufus on a different computer
- [ ] If using an ISO image, I clicked on the `#` button (at the bottom of the Rufus interface), to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<Been trying to make this usb boot win10 using media creation tool that didnt work so im using rufus and now it says undetermined error please help>

Log
Rufus version: 2.11.995
Windows version: Windows 10 64-bit (Build 10586)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta3
System locale ID: 0x0409
Will use default UI locale 0x0409
Found USB 2.0 device 'SanDisk Cruzer USB Device' (0781:5530)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 973, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: SFD (Super Floppy Disk) or Unpartitioned
Partition 0:
  Type: FAT32 (0x00)
  Size: 7.5 GB (8004304896 bytes)
  Start Sector: 0, Boot: No
Scanning image...
ISO analysis:
  Image is an UDF image
Disk image analysis:
  Image does not have an x86 Master Boot Record
ISO label: 'ESD-ISO'
  Size: 3630563328 bytes
  Uses: EFI
  Uses: Bootmgr
Using image: Windows.iso

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE1 for write access
Will use 'H:' as volume mountpoint
I/O boundary checks disabled
Analyzing existing boot records...
Drive has an unknown Master Boot Record
Volume has an unknown Partition Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Deleting partitions...
Partitioning (MBR)...
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (NTFS)...
Using cluster size: 4096 bytes
Quick format was selected
Creating file system...
Format completed.
Writing master boot record...
Drive has a Zeroed Master Boot Record
Set bootable USB partition as 0x80
Using Rufus MBR
Found volume GUID \\?\Volume{cb8d4ce7-2042-11e6-bff5-a0d3c16698a3}\
Opened drive \\?\Volume{cb8d4ce7-2042-11e6-bff5-a0d3c16698a3} for write access
Writing partition boot record...
Using Standard NTFS partition boot record
Confirmed new volume has an NTFS boot sector
Successfully remounted Volume{cb8d4ce7-2042-11e6-bff5-a0d3c16698a3}\ on H:\
Copying ISO files...
Extracting files...
Image is an UDF image
Extracting: H:\boot\en-us\bootsect.exe.mui (16 KB)
Extracting: H:\boot\fonts\chs_boot.ttf (3.5 MB)
Extracting: H:\boot\fonts\cht_boot.ttf (3.7 MB)
Extracting: H:\boot\fonts\jpn_boot.ttf (1.9 MB)
Extracting: H:\boot\fonts\kor_boot.ttf (2.3 MB)
Extracting: H:\boot\fonts\malgunn_boot.ttf (170.8 KB)
  Unable to create file: [0x00000057] The parameter is incorrect.

Found USB 2.0 device 'SanDisk Cruzer USB Device' (0781:5530)
Device eliminated because it appears to contain no media
0 devices found
```
<FULL LOG>
```

jafarxd, 2016-12-13:
This is the log of when it checked for bad blocks 

LOG:   
Rufus version: 2.11.995
Windows version: Windows 10 64-bit (Build 10586)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta3
System locale ID: 0x0409
Will use default UI locale 0x0409
Found USB 2.0 device 'SanDisk Cruzer USB Device' (0781:5530)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 973, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00542310
Drive has a Rufus Master Boot Record
Partition 1:
  Type: FAT32 (0x0b)
  Size: 7.5 GB (8003256320 bytes)
  Start Sector: 2048, Boot: Yes
Scanning image...
ISO analysis:
  Image is an UDF image
Disk image analysis:
  Image does not have an x86 Master Boot Record
ISO label: 'ESD-ISO'
  Size: 3630563328 bytes
  Uses: EFI
  Uses: Bootmgr
Using image: Windows.iso

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE1 for write access
Will use 'H:' as volume mountpoint
I/O boundary checks disabled
Analyzing existing boot records...
Drive has a Rufus Master Boot Record
Volume has an unknown Partition Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Deleting partitions...
Bad Blocks: Checking from block 0 to 15633407
Bad Blocks: Using offset 0 for fake device check
Bad Blocks: Writing test pattern 0xAA
Bad Blocks: Reading and comparing
Bad Blocks: Check completed, 0 bad blocks found. (0/0/0 errors)
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Partitioning (MBR)...
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (NTFS)...
Using cluster size: 4096 bytes
Quick format was selected
Creating file system...
Format completed.
Writing master boot record...
read_sectors: Read error [0x00000002] The system cannot find the file specified.
  Read: 0, Expected: 512
  StartSector: 0x00000000, nSectors: 0x1, SectorSize: 0x200
read_sectors: Read error [0x00000002] The system cannot find the file specified.
  Read: 0, Expected: 512
  StartSector: 0x00000000, nSectors: 0x1, SectorSize: 0x200
read_sectors: Read error [0x00000037] The specified network resource or device is no longer available.
  Read: 0, Expected: 512
  StartSector: 0x00000000, nSectors: 0x1, SectorSize: 0x200
read_sectors: Read error [0x00000037] The specified network resource or device is no longer available.
  Read: 0, Expected: 512
  StartSector: 0x00000000, nSectors: 0x1, SectorSize: 0x200
read_sectors: Read error [0x00000037] The specified network resource or device is no longer available.
  Read: 0, Expected: 512
  StartSector: 0x00000000, nSectors: 0x1, SectorSize: 0x200
read_sectors: Read error [0x00000037] The specified network resource or device is no longer available.
  Read: 0, Expected: 512
  StartSector: 0x00000000, nSectors: 0x1, SectorSize: 0x200
Drive has an unknown Master Boot Record
read_sectors: Read error [0x00000037] The specified network resource or device is no longer available.
  Read: 0, Expected: 512
  StartSector: 0x00000000, nSectors: 0x1, SectorSize: 0x200
Could not read MBR

0 devices found
0 devices found

pbatard, 2016-12-13:
> Been trying to make this usb boot win10 using media creation tool that didnt work

If two different tools don't work, then it's most likely an error with your USB or your environment, and there's nothing specific to Rufus about it.
Also, you didn't follow the guidelines &rarr; Closed.

closed 2016-12-13 by pbatard