pbatard/rufus #868: [INTERNAL] List of 3.0 small(ish) enhancements and fixes

pbatard, 2016-12-15:
- [ ] Drop XP support
- [ ] Windows 10 type notifications for Rufus updates (see WinSCP)
- [ ] Switch to GPT for UEFI by default on dual BIOS + EFI images, instead of MBR for BIOS

NSbuilder, 2016-12-15:
The next version is going to be 3.0 or 2.12?

pbatard, 2016-12-15:
No idea. It'll be what it will be. Also, please avoid commenting on issues that are marked as __[INTERNAL]__. As their name indicate, they're just for internal notes, which can be ALTERED IN ANY DIRECTION AT ANY MOMENT and do __NOT__ mean that what you see written will actually ever be done.

pbatard, 2016-12-15:
Also please see the [milestones](https://github.com/pbatard/rufus/milestones) if you are interested in the VERY VAGUE PLAN that I have for upcoming releases. This too can be altered drastically at any time, so please don't try to read too much into it.

positr0nic, 2017-03-31:
I cant seem to find a way to make a bootable usb with persistence via rufus,  if thats is not supported it would be a fine addition.

pbatard, 2017-03-31:
Please don't use issues flagged as [INTERNAL] to ask for features, especially as, if you had created a new issue, and followed the checklist, you would have found #691.

positr0nic, 2017-04-01:
Apologies, its not clear anywhere that this is not a thing todo. I simply saw a checklist and thought it was relevant.

closed 2018-05-29 by pbatard