pbatard/rufus #869: The device is in use by another process. Please close any other process that may be accessing the device.

DirtyBCG, 2016-12-20:
Checklist
---------
- [x ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x ] I performed a search in the issue tracker for similar issues, using keywords relevant to my problem.
- [x ] I clicked the `Log` button in Rufus and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [x ] I ran a bad blocks check, by clicking the "bad blocks" check box in Rufus, and confirmed that my USB is not defective
- [ x] I also tried one or more of the following:
  - [ x] Using a different USB drive
  - [ x] Plugging the USB into a different port
  - [x ] Running Rufus on a different computer
- [x ] If using an ISO image, I clicked on the `#` button (at the bottom of the Rufus interface), to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<Attempting to write ISO image to a 2tb USB hard drive.>

Log
---
```
<FULL LOG>
``Rufus version: 2.11.995
Windows version: Windows 10 64-bit (Build 10240)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta3
System locale ID: 0x0409
Will use default UI locale 0x0409
Found USB 2.0 device 'WD My Passport 0837 USB Device' (1058:0837)
NOTE: This device is an USB 3.0 device operating at lower speed...
Device eliminated because it was detected as a Hard Drive (score 23 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives in 'Advanced Options' (after clicking the white triangle)
Found USB 3.0 device 'WD My Passport 25E2 USB Device' (1058:25E2)
Device eliminated because it was detected as a Hard Drive (score 20 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives in 'Advanced Options' (after clicking the white triangle)
0 devices found
Found USB 2.0 device 'WD My Passport 0837 USB Device' (1058:0837)
NOTE: This device is an USB 3.0 device operating at lower speed...
Found USB 3.0 device 'WD My Passport 25E2 USB Device' (1058:25E2)
2 devices found
Disk type: Fixed, Sector Size: 512 bytes
Cylinders: 121597, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00865E7B
Drive has a Windows 7 Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 931.5 GB (1000169537024 bytes)
  Start Sector: 2048, Boot: Yes
2 devices found
No volume information for drive 0x82
Disk type: Fixed, Sector Size: 512 bytes
Cylinders: 243197, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 0
Disk ID: 0x052835C1
Drive has a Zeroed Master Boot Record
Scanning image...
ISO analysis:
  Image is an UDF image
Disk image analysis:
  Image does not have an x86 Master Boot Record
ISO label: 'Win10PESE'
  Size: 4071849984 bytes
  Uses: EFI
  Uses: Bootmgr
Using image: Gandalf's_Win10PE_x64_TH2-05-13-2016.ISO

Format operation started
Notice: Large drive detected (may produce short writes)
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE2 for write access
No drive letter was assigned...
Will use 'D:' as volume mountpoint
No logical drive found (unpartitioned?)
Drive does not appear to be partitioned
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Deleting partitions...
Partitioning (MBR)...
Waiting for logical drive to reappear...
Formatting (NTFS)...
Using cluster size: 4096 bytes
Quick format was selected
Volume is in use
Format error: The device is in use by another process. Please close any other process that may be accessing the device.
\\?\Volume{05294d8a-0000-0000-0000-100000000000}\ was already mounted as D:\
Re-mounted volume as 'D:' after error

Found USB 2.0 device 'WD My Passport 0837 USB Device' (1058:0837)
NOTE: This device is an USB 3.0 device operating at lower speed...
Found USB 3.0 device 'WD My Passport 25E2 USB Device' (1058:25E2)
2 devices found
Disk type: Fixed, Sector Size: 512 bytes
Cylinders: 243197, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x05294D8A
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 1.8 TB (2000364240896 bytes)
  Start Sector: 2048, Boot: Yes

Format operation started
Notice: Large drive detected (may produce short writes)
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE2 for write access
Will use 'D:' as volume mountpoint
I/O boundary checks disabled
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Volume has an unknown Partition Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Deleting partitions...
Partitioning (MBR)...
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (NTFS)...
Using cluster size: 4096 bytes
Quick format was selected
Volume is in use
Format error: The device is in use by another process. Please close any other process that may be accessing the device.
\\?\Volume{05296d56-0000-0000-0000-100000000000}\ was already mounted as D:\
Re-mounted volume as 'D:' after error

Found USB 2.0 device 'WD My Passport 0837 USB Device' (1058:0837)
NOTE: This device is an USB 3.0 device operating at lower speed...
Found USB 3.0 device 'WD My Passport 25E2 USB Device' (1058:25E2)
2 devices found
Disk type: Fixed, Sector Size: 512 bytes
Cylinders: 243197, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x05296D56
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 1.8 TB (2000364240896 bytes)
  Start Sector: 2048, Boot: Yes

Format operation started
Notice: Large drive detected (may produce short writes)
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE2 for write access
Will use 'D:' as volume mountpoint
I/O boundary checks disabled
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Volume has an unknown Partition Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Deleting partitions...
Partitioning (MBR)...
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (NTFS)...
Using cluster size: 4096 bytes
Quick format was selected
Volume is in use
Format error: The device is in use by another process. Please close any other process that may be accessing the device.
\\?\Volume{0529906f-0000-0000-0000-100000000000}\ was already mounted as D:\
Re-mounted volume as 'D:' after error

Found USB 2.0 device 'WD My Passport 0837 USB Device' (1058:0837)
NOTE: This device is an USB 3.0 device operating at lower speed...
Found USB 3.0 device 'WD My Passport 25E2 USB Device' (1058:25E2)
2 devices found
Disk type: Fixed, Sector Size: 512 bytes
Cylinders: 243197, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x0529906F
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 1.8 TB (2000364240896 bytes)
  Start Sector: 2048, Boot: Yes

Computing checksum for 'C:\Users\HBVU\Documents\Gandalf's_Win10PE_x64_TH2-05-13-2016.ISO'...
  MD5:	 7dd1c0f603738c208e137d077dddec70
  SHA1:	 fda69a630b4ac76d7e15cb24ee203472afa4e4c9
  SHA256: 8edda6518c6970db32c41ce259ac64af09823c2f3e7b25b81f19e37c3ec2d78c

pbatard, 2016-12-20:
Not much that can be done when Rufus especially indicates that "The device is in use by another process". __You__ will need to find what process is accessing the drive and prevent it to do so. It may be a security solution or one of the software listed at the [end of the FAQ](https://github.com/pbatard/rufus/wiki/FAQ#Software).

Unfortunately, there's really not much I can do to help, as it's impossible to take a guess at what is causing the access issue. You may have some luck identifying what it is by going through every single running process in [Process Explorer](https://technet.microsoft.com/en-us/sysinternals/processexplorer.aspx) and seeing which one seems to keep a lock on your USB drive.

closed 2016-12-20 by pbatard

awyork, 2017-08-07:
Try turning off Auto Play.  In Windows 10 just search for it.  Then reboot.

n3vmx2, 2017-10-28:
Its an old Post but .. i had the same problem with "the device is used by another process" ((I am using win 8.1 ))

So i easily managed to fix that problem ( by using Process Hacker)
The process that was keeping my device busy was SkyDrive, so i just susspended it!
And that problem was solved , didn't even needed a pc reboot.

Hope it was helpfull !
Regards

ch123123, 2017-12-22:
_"The device is in use by another process. Please close any other process that may be accessing the device._ "
I had the same problem. Before using Rufus, I renamed the ISO file to "w7+sp1_x32.iso," resulting in the error message above. After I renamed the iso back to the manufacturer's name, "GSP1RMCHPFRER_EN_DVD.iso," Rufus worked normally. 
(PS: I also renamed the flash drive from a previous ISO name to 16Gig.)

sidantaa, 2018-04-18:
I too tried many techniques but finally i followed hit and trial....i just copied the same iso file to other location i.e directory within the drive and launched rufus and after then picked up the iso image copied in the other directory,,Doing this everything worked like a charm!!!!

adeelmughal4u, 2018-05-30:
You have to just stop your windows defender because windows defender is accessing your device, after stopping windows defender it will work.

fotzepoliticz, 2018-08-16:
I was having the same problem in my new Win10 install and this solved it:

Windows Defender Security System ---> Virus & Threat Protection Settings ---> Controlled Folder Access (Manage Controlled Folder Access) ---> Allow an App through Controlled Folder Access ---> Add an Allowed App

In this case I added Rufus to my list. Alternatively you can disable Controlled Folder Access altogether. I think Microsoft increasing the security access level in general is good news for most customers but having settings buried in obscure menus is counterproductive to others, IMHO.

I hope this helps.

pbatard, 2018-08-16:
FYI, the latest version of Rufus will tell you if _Controlled Folder Access_ is preventing access in the log. You will see a message like:
```
WARNING: 'Controlled Folder Access' appears to be enabled on this system
You may need to disable this feature, or add an exception, for Rufus to to work...
```
If you encounter an issue, __please__ try to check the log, as it will give you more details about what the issue can be.

JOFTAA, 2018-09-03:
I was able to correct the problem by formatting my USB drive with ExFAT.  Rufus displayed an message or two during the process of writing the ISO, but the USB was bootable with my software when finished.

buttererror, 2018-10-03:
I had the same problem but i didn't read the log, however, you can try clearing your USB label and try again, or format your USB as normal with windows, then use the rufus with out "quick format" checked.

masterking32, 2018-10-05:
Open Rufus Select ISO file then, Open Task Manager -> Details, find explorer Click on End task, Then on rufus click write !
after write done on task manager -> file -> run & type explorer & click ok :)

Inferillum, 2018-11-14:
Just noticed that using Kaspersky (probably some kind of Controlled Folder Access option) causes this issue, too. Just disable that and you'll be fine to use Rufus.

SteveB1313, 2019-01-16:
Sophos Endpoint Protection causes the same issue, disable that and Rufus works fine again.

sunk818, 2019-02-23:
Windows 10 - go to settings, search for auto play

[image]

dawoodjee, 2019-02-25:
I had to properly copy my iso file. Didn't notice it was incomplete

asalom, 2019-03-15:
Deactivating AutoPlay + machine restart worked for me

pbatard, 2019-03-15:
Note that Rufus does include code to disable AutoPlay while it is running (`NoDriveTypeAutorun` which is [set through Local Group Policies](https://github.com/pbatard/rufus/blob/efd6d9f3455f383f9fcf0747a1e38548d3149858/src/rufus.c#L3108-L3110)) so I'm a bit puzzled about these reports about AutoPlay also having be disabled manually. I'm more inclined to think that the __reboot__ is what killed whatever process was accessing the drive...

Signy13, 2019-04-29:
I am facing the issue "The device is in use by another process" after format the device. Of course I can kill the process via Task manager but it would be much better if Rufus could just ignore the fact that the device is used by another process.
If Rufus can format the device, why not to ignore another processes? Or why not to ask user "The device is in use, cancel or continue?"
There are another pieces of SW which can detect using of another process, but they just ask weather force continue... In such case is user much less bothered by weird behavior of some processes... It is much better than being forced to change some settings on computer, restart computer etc.

pbatard, 2019-04-29:
> If Rufus can format the device, why not to ignore another processes?

It can.

> Or why not to ask user "The device is in use, cancel or continue?"

It already does.

Do this: Open a DOS prompt on your device and press START. __This__ is what you'll get:

[image]

As you can see, Rufus already has code [dedicated to check for conflicting processes](https://github.com/pbatard/rufus/blob/master/src/process.c) (which we stole from [Process Hacker](https://processhacker.sourceforge.io/), so we're pretty confident it is very comprehensive in terms of being able to enumerate the processes we seek) and you __do__ get a prompt about other processes accessing the drive and asking you if you want to ignore them.

So, Rufus __already__ does what you suggest.

Now, the problem is that, even as we are doing this, there are some processes and services that are accessing the drive that Windows doesn't always seem to report, and, since you mention that your issue occurs __after__ you format the device (though you may want to clarify when exactly), we can of course not do anything about processes that are designed to monitor new volumes and access them.

For instance, as soon as you format a drive to FAT32 or NTFS, Windows will create a `System Volume Information` directory and add some files there. And security solutions may decide to scan the newly arrived device, therefore preventing you to access it. And of course there is all the poorly designed software listed at the [end of the FAQ](https://github.com/pbatard/rufus/wiki/FAQ#Incompatible_Software) that seems to open drives for write access.

In that case then you may get a device in use message before Rufus can copy its file, and there's not much we can do about this.

The one thing you may try is the <kbd>Alt</kbd>-</bkd>,</kbd> cheat mode, which disables the exclusive locking that Rufus tries to use, but otherwise, I will respectfully ask that you don't assume that Rufus is not already doing something you think it should do, and instead provide more details about your actual issue.

For instance, you don't mention at all the process that you can kill in Task Manager. You do realize that knowing what process or application appears to be causing your issue is the very first thing you should be disclosing here...

Signy13, 2019-05-07:
I am sorry, it is my fault that I had not noticed that Explorer++ i in the list of incompatible SW.
Now my solution is to use LockHunter to unlock Explorer++ and get rid of the issue. Killing all instances of Explorer++ is not solution for me - I use a lot of them and kill all of them because some lock is not an acceptable solution.

But still - Truecrypt (for example) has the same problem, when it faces the same issue (folder/directory locked by Explorer++ and some system process) - and yes, Truecrypt just asks "Force dismount" and it is able to dismount volume - without necessity of any other software.

If you say that Rufus can ask (before format) whether it can continue anyway - why Rufus is not successful in other cases like after format? From user's point of view it looks like that Rufus tries to do something and it is not successful in all cases. If other software can do it, than it looks as if it was Rufus' fault.
I agree that unnecessary locking of Explorer++ bad thing and Explorer++ should be fixed but stil - should not be Rufus fixed as well?

pbatard, 2019-05-07:
"Force dismount" is a different thing than what we need.

Dismount applies to __volumes__ (which more or less equates a partition), and Rufus does request a force dismount, probably in the same way as like TrueCrypt (since there aren't that many APIs you can use to "force dismount" a volume) [through the `FSCTL_DISMOUNT_VOLUME` IOCTL](https://github.com/pbatard/rufus/blob/2ff6da49f0d9141f318e01c82d4c27f3adba849d/src/drive.c#L1208-L1220).

Now, if you look at the [doc for `FSCTL_DISMOUNT_VOLUME`](https://msdn.microsoft.com/en-us/library/windows/desktop/aa364562.aspx) you'll see that is says:

> The FSCTL_DISMOUNT_VOLUME control code will attempt to dismount a volume regardless of whether or not any other processes are using the volume, which can have unpredictable results for those processes if they do not hold a lock on the volume.

So, yeah, we're no even asking politely as TrueCrypt does, we just force things.

However, that's fine for volumes, but the main issue is you can have applications that hold a lock to the disk itself, and those are a lot trickier to control, even more so as, with its latest instance of disk virtualization (VDS) Microsoft has found nothing better than require that an application that wants to issue low level calls such as deleting or refreshing a partition layout first make sure that they don't have a handle open to the disk... which of course prevents us from keeping a lock on it, which in turn gives ample opportunity for poorly designed applications or even the OS to get their foot in the door before we get a chance to try to reacquire the lock.

Anyway, I'm always happy to accept fixes if you think Rufus can be improved. GitHub has a nice pull request feature, so, if you think Rufus is not doing enough with regards to keeping applications that are a bit too intrusive at bay, I sure wouldn't mind see you (or someone else) take a stab at improving that.

At least considering that I'm currently elbow deep into trying to get Windows to acknowledge a partition refresh without having to issue a USB port cycle (which is the only way I know that conistently works at telling everyone, including the OS, to keep their hands of the disk/volumes), so that we can have a standalone `ext2`/`ext3` formatting option that doesn't only work half the time, it would provide me with some company with regards to commiserating on how poorly Windows seems to be designed when it comes to letting an application completely overhaul disks and volumes, without being faced with constant hurdles.