pbatard/rufus #871: Cannot mount boot-repair-disk.

warxed, 2016-12-24:
Checklist
---------
- [X] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [X] I performed a search in the issue tracker for similar issues, using keywords relevant to my problem.
- [X] I clicked the `Log` button in Rufus and copy/pasted the log into the line that says `<FULL LOG>` below.
- [X] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [X] I ran a bad blocks check, by clicking the "bad blocks" check box in Rufus, and confirmed that my USB is not defective
- [X] I also tried one or more of the following:
  - [ ] Using a different USB drive
  - [X] Plugging the USB into a different port
  - [ ] Running Rufus on a different computer
- [X] If using an ISO image, I clicked on the `#` button (at the bottom of the Rufus interface), to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
I am trying to get a boot-repair-disk iso mounted onto my usb but haven't been able to.

Log
---
```
<FULL LOG>
Rufus version: 2.11.995
Windows version: Windows 7 SP1 64-bit
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta3
System locale ID: 0x0409
Will use default UI locale 0x0409
Found USB 2.0 device 'PNY USB 2.0 FD USB Device' (154B:007A)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 992, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x1737BC46
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 7.6 GB (8165654528 bytes)
  Start Sector: 2048, Boot: Yes
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/isolinux/isolinux.cfg' for Syslinux
  Detected Syslinux version: 4.05/20140113 (from '/isolinux/isolinux.bin')
Disk image analysis:
  Image does not have an x86 Master Boot Record
ISO label: 'Boot-Repair-Disk 64bit'
  Size: 328708212 bytes
  Uses: Syslinux/Isolinux v4.05
  Uses: EFI
  Note: This ISO uses symbolic links, which will not be replicated due to file system limitations.
  Because of this, some features from this image may not work...
Using image: boot-repair-disk-64bit.iso

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE1 for write access
Will use 'G:' as volume mountpoint
I/O boundary checks disabled
Analyzing existing boot records...
Drive has a Syslinux Master Boot Record
Volume has an unknown Partition Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Deleting partitions...
Partitioning (MBR)...
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (FAT32)...
Using cluster size: 4096 bytes
Quick format was selected
Creating file system...
Format completed.
Writing master boot record...
Drive has a Zeroed Master Boot Record
Set bootable USB partition as 0x80
Using Syslinux MBR
Found volume GUID \\?\Volume{652d165e-46e1-11e5-a2b2-e0cb4e432990}\
Installing Syslinux 4.05...
Opened drive \\?\Volume{652d165e-46e1-11e5-a2b2-e0cb4e432990} for write access
Successfully wrote 'ldlinux.sys'
Successfully wrote Syslinux boot record
Successfully remounted Volume{652d165e-46e1-11e5-a2b2-e0cb4e432990}\ on G:\
Copying ISO files...
Extracting files...
Image is an ISO9660 image
This image will be extracted using Rock Ridge extensions (if present)
Extracting: G:\autorun.inf (67 bytes)
Extracting: G:\boot\grub\efi.img (2.3 MB)
  Error reading ISO9660 file /boot/grub/efi.img at LSN 298248
Created: G:\syslinux.cfg

Found USB 2.0 device 'PNY USB 2.0 FD USB Device' (154B:007A)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 992, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x173A3A89
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 7.6 GB (8165654528 bytes)
  Start Sector: 2048, Boot: Yes

Computing checksum for 'C:\Users\The Amazing Carlos\Desktop\boot-repair-disk-64bit.iso'...
  MD5:	 01db2c92639df18e047379c74c0504b5
  SHA1:	 8bb4ffbf27be68eb573030bf21b32fe61ef8934d
  SHA256: 5cc8b06cf2124fa7cf57fe0f476841aed8be4a3822bb42cef0cd0978676dd8c7

pbatard, 2016-12-24:
`Error reading ISO9660 file /boot/grub/efi.img at LSN 298248` seems to indicate that your ISO might be corrupted, or that it may not have been mastered properly by the people who created it.

And just in case you happen to have burned the same ISO as an optical media, and was able to boot from it, I'll just point out that CD/DVD burners are a lot more forgiving with writing corrupted data and that, as long as that data is not used during boot, then you might think the ISO was fine, even though it really wasn't.

Unfortunately, Rufus isn't that forgiving with these kind of ISOs, and will produce the kind of errors you see as soon as it encounters a corrupted files.

Now, I'm assuming that your `boot-repair-disk-64bit.iso` was downloaded from https://sourceforge.net/p/boot-repair-cd/home/Home/, right?
If so, then your ISO was definitely corrupted. Here are the checksums I get on my machine:
```
Computing checksum for 'C:\Downloads\boot-repair-disk-64bit.iso'...
  MD5:	 2aa0cab5db7ae0ae00fcf32d4ffdd14c
  SHA1:	 f2565b382d592c5f3ff416856f3f15bfecce9bb4
  SHA256: a8d7b32576a383109743d13f58567caa599f4046a3d4181a19e7e564b9c608ed
```

Either you need to download that ISO again, until you get the same checksums as the ones above, or you need to indicate where you downloaded your ISO from, and whether the people who created that ISO do provide a checksum that you can verify to validate that your download was OK.

warxed, 2016-12-24:
You're absolutely right! I did get the ISO from sourceforge. Is there
somewhere you would recommend that has non-corrupted ISO?

pbatard, 2016-12-24:
It looks to me like it's your network connection or disk that is the issue. I was able to download the ISO from SourceForge without issues (and with what I assume is the correct checksum).

You should try to download the ISO from SourceForge a few times until you get the same checksums as I do.
If you can't do that, then you have a bigger problem on your hand, which might be that either the HDD is failing on your computer or you have a network connection that is so unstable that you'll need to fix it.

warxed, 2016-12-24:
You hit the nail on the head with the unstable network issues. I'll try
downloading it somewhere more stable and get back to you!

warxed, 2016-12-24:
I downloaded the ISO on a more stable network connection and everything
worked perfectly! Thanks for the help!

pbatard, 2016-12-24:
Okay - glad to hear you were able to sort it out. I'll close the issue now.

closed 2016-12-24 by pbatard