pbatard/rufus #876: USB created with Rufus won't work properly with Windows 10 Pro build (July 2016)

alopezf1987, 2016-12-28:
Hello,

I've recently experienced some problems when trying to install Windows 10 Pro 64 bits (Redstone 1 - build 10.0.14393). Everything seemed to work fine until Windows installation started: it didn't detect my hard disk drives.

I finally had to use the official [tool](https://www.microsoft.com/es-es/download/windows-usb-dvd-download-tool).

How can I help to get this solved?

closed 2016-12-28 by pbatard

pbatard, 2016-12-28:
Closed because you failed to post the log of how you created your USB with Rufus, as per the checklist, which could have provided some clues as to why that might be happening. As explained in the checklist, failure to follow the guidelines will result in issues being summarily closed.

Since you are the only person to report this, it is likely something to do with your settings and not Rufus. Are you sure you selected the proper options for a BIOS or UEFI computer? Was your USB drive partitioned for GPT or MBR? And how was the drive that worked partitioned?

alopezf1987, 2016-12-28:
I didn't post any kind of log because this was on the computer BEFORE its hard disk was erased. Default options were chosen and USB drive was partitioned for MBR using diskpart util.

I just wanted to warn you about this fact, really don't care if the issue is or not open. I've tried with some USB drives on different machines.

Merry christmas.