pbatard/rufus #877: Video drivers disabled after application started

mike45nik, 2016-12-29:
It sounds strange, but video card drivers are disabled after the start of the program! Both the video adapter (nvidia 770 and intel 4000) appear as unknown devices. After reinstalling drivers, everything becomes OK, but running the application again disables them.

Rufus version: 2.11.995
Windows version: Windows 10 64-bit (Build 14393)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta3
System locale ID: 0x0419
Will use default UI locale 0x0419
Found non-USB removable device 'Crucial_CT250MX200SSD1' => Eliminated
0 devices found
Found non-USB removable device 'Crucial_CT250MX200SSD1' => Eliminated
0 devices found
Found non-USB removable device 'Crucial_CT250MX200SSD1' => Eliminated
0 devices found
Found non-USB removable device 'Crucial_CT250MX200SSD1' => Eliminated
0 devices found
Found non-USB removable device 'Crucial_CT250MX200SSD1' => Eliminated
0 devices found
Found non-USB removable device 'Crucial_CT250MX200SSD1' => Eliminated
0 devices found
Found non-USB removable device 'Crucial_CT250MX200SSD1' => Eliminated
0 devices found

mike45nik, 2016-12-29:
In device manager these drivers marked as "Installation prevented by policy".

pbatard, 2016-12-29:
Considering that, in millions and millions of downloads of Rufus, you are the __only__ person to ever report this issue, I find it hard to believe that this has anything to do with Rufus. You're certainly not the only person to run Rufus with both an nVidia and intel adapters (and, considering the odds, even the exact same combination), so I think you may want to look elsewhere.

There is nothing in Rufus that even remotely deals with graphic adapters, and I'm pretty sure that none of your graphic adapters are USB based (since regular USB is usually way to slow for anything graphical), so, even if you currently believe that Rufus is the issue, there's virtually nothing I can do to even start trying to help with your issue. There's certainly no "Oh yeah, Rufus pokes graphical adapters on start, so I'll try to fix that", that I can give you here.

For one thing, Rufus does not deal with devices or devices drivers directly, and its device enumeration process for USB devices is completely passive (we don't poke devices or send "special commands" that may interfere with a device operations) and based on Windows standard practices, so, while you may think that maybe the application has a bug that somehow makes it send commands to unwanted devices such as graphical adapters, this is just not logically possible in the Windows device framework (that is unless Windows itself has a major bug, but in that case it's Microsoft you need to contact).

Also, I'm very puzzled at to how your graphic drivers can appear disabled, but you can still see the screen. Are you accessing your Windows platform remotely when that happens, and are actually losing your display?
If you are not losing your display, then it means that, contrary to what device manager reports, your graphic drivers are not disabled, and are certainly not actually handled by Windows as unknown devices.

So, my guess is that, on your specific machine (since, in what I estimate are more than 60 million downloads of Rufus, you really are the only person to ever report an issue such as this one), you have a benign issue with Windows regarding the listing of drivers in device manager (since it's just the display report that is wrong, and your drivers are still installed and working), that somehow can get triggered when running some applications in elevated mode. It may be a hardware issue. It may have to do with your system configuration. Or it may very well be a Windows bug.

The first thing I'd do then, is try to run other applications in elevated mode, to see if you encounter the same issue. Then you may want to contact Microsoft to hear what they have to say (since your issue is really with the Device Manager, and __clearly__, not matter what another application does, and especially one that __does not deal with drivers or graphical display at all__, the Device Manager should not flag installing and working graphic drivers as unknow device).

Now, since you mention that your drivers are flagged as "Installation prevented by policy" (which is even more puzzling &mdash; How comes you do manage to reinstall the drivers then??), I have to ask, are you running on a corporate machine or network, that is enforcing a set of restrictions? For one, I could see something like some advanced (but flawed) corporate software policies detecting when someone is running an application elevated, and applying policies to try to lock down sensitive operations, such as the ability to install drivers. This way, it would allow a user to run elevated apps, which they might need for their job, but without giving them so much freedom as to be able to reinstall drivers and so on.

So, once again, I'm afraid that you will have to look elsewhere to solve this issue, as, unless other people start reporting the same issue, I am positive that, while this may very well be triggered by launching Rufus (due to your config, or other software or security policies), there's nothing I can do in Rufus to help you out.

mike45nik, 2016-12-29:
I found policy "Administrative templates > System > Device Installation > Device Installation Restrictions" which disables auto-installation of graphic driver. After removing it - problem solved. It is not clear what relationship it had to rufus.

pbatard, 2016-12-29:
Can you tell which one of the policies below you changed (I see multiple ones in Device Installation Restrictions)?
[image]

Again, I don't see how Rufus could have anything to do with this, but I'd like to see if there's any way to try to replicate your issue, just in case.

mike45nik, 2016-12-29:
third one

pbatard, 2016-12-29:
Thanks. 

 may need a bit more information then, as the 3rd option should also have a list of GUIDs associated with the policy, as per the screenshot below (once you click the _Show_ button). Obviously, my list is empty, since I have yet to restrict stuff. Can I ask you if you see anything on your side, and if so, what GUIDs are listed there?

[image]

mike45nik, 2016-12-29:
Allow administrators... : enabled
Prevent installation...: enabled. Unfortunately I deleted the code and don't remember it. But it was a class for my video adapter.

pbatard, 2016-12-29:
Okay. I think that's enough information for me to work with - thanks.

pbatard, 2016-12-29:
Wow, enabling that policy sure messes up stuff. As soon as I enabled that policy with the class for my nVidia GPU, Windows switched back to the default VGA display driver, and reset DPI settings.

[image]

Note that I wasn't running Rufus at all when this happened, so I can definitely see how this policy can seemingly screw up one's display drivers, regardless of any application being run...

mike45nik, 2016-12-29:
In your case this happened because the driver was installed automatically. In my case, it first disable automatic driver installation, and then I manually installed the nvidia driver.

mike45nik, 2016-12-29:
So, for reproduce this issue:
1. Activate policy restriction
2. Install driver from nvidia.com
3. Run rufus
4. Profit...

pbatard, 2016-12-29:
FYI, I did install the drivers manually, so I don't think this has anything to do with manual vs 

But I was able to replicate the issue, and I think I have identified how this is triggered. I was also able to confirm that this is a pure Microsoft issue.

When Rufus starts, it uses the Microsoft Group Policy APIs to disable autorun (so that users aren't prompted with annoying autorun notification when they plug a new device that they are planning to format), and from what I can see (and __regardless of the application doing so__) as soon as you alter the group policies and have the policy we mentioned above enabled, Windows will reset the drivers for the devices that were restricted.

So, you will see the drivers being reverted when launching Rufus... but you will also see the exact same thing happen if, say, you enable the _Display a custom message_ policy in GPEdit and save your changes (it takes a few seconds before the change is registered and the drivers are reset).

Whether this is intentional or not, I can't say, but clearly, this is a pure Windows issue. When any local policy is altered, Windows seems to reset all the the drivers of the devices that are being restricted by the driver installation policy.

Now, you may ask if Rufus really needs to disable Autorun, or if Rufus should really do it through the Local Group Policies, but I think the benefits far outweigh this specific issue and I actually went great length to modify the autorun settings in the manner that is advocated by Microsoft, which is through LGPs (so that, if any issue happens during this change, it can be detected and handled safely by Windows, and lot result in one's system in an unknown state). This means that I'm not planning to change the current behaviour.

However, if it really bothers you, and you want to use Rufus, you should be able to recompile your own application that disable the LGP invocation. Just comment out the `SetLGP()` in `rufus.c` and you should be good.

Still, a better approach might be to ask Microsoft if they are aware that LPG change always seems to reset the drivers concerned by the _Prevent installation of drivers..." policy, and whether they think this is something that needs to be fixed.

mike45nik, 2016-12-29:
My goal is to eliminate the automatic update of nvidia drivers. This can be solved differently: http://winsupersite.com/windows-10/stop-automatic-driver-updates-windows-10
Thanks for the research :)

NSbuilder, 2016-12-29:
@pbatard Why is my device Installation Settings screen is different?

My Windows: http://i.imgur.com/Zu7yEiZ.png
Internet: http://winsupersite.com/site-files/winsupersite.com/files/imagecache/large_img/uploads/2015/08/deviceinstallhero.png

pbatard, 2016-12-29:
@NSbuilder, I'm going to have to delete your post. This is neither relevant to the issue at hand or Rufus in general. Please post these kind of questions on a Windows user forum, not the Rufus issue tracker.

NSbuilder, 2016-12-29:
I know but can you please answer?

pbatard, 2016-12-29:
Sorry but I have no idea. This is specific to what system you are using and how it was set.

pbatard, 2017-01-02:
Since the issue has been solved, I will close it.

closed 2017-01-02 by pbatard