pbatard/rufus #878: NRG file is supported?

kavin-90, 2016-12-30:
How to use NRG file to make Bootable USB?

pbatard, 2016-12-30:
NRG is a proprietary format, and too few people use it to warrant spending time adding support for it in Rufus. You will have to convert it to ISO.

closed 2016-12-30 by pbatard

kavin-90, 2016-12-31:
@pbatard i tried to convert it to iso but refus not pick it said invalid ISO but iso is working file i try to use it on Windows 7 DVD Download same error can you please tell me good program for it?

My Generated ISO was ISO096 like something

pbatard, 2016-12-31:
Rufus does not support all bootable ISOs either for the same reason why it doesn't support NRG: there are too many ways of making an ISO bootable and each one needs a custom process to convert to to bootable USB, so only the most popular ISO booting mechanisms are supported.

> can you please tell me good program for it

I'd ask the developer of the NRG to tell you what application they recommend.