pbatard/rufus #880: Rufus error could not get disk extents/drive has an unknown master boot record

Kerzenflamme, 2017-01-04:
Checklist
---------
- [ ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ ] I performed a search in the issue tracker for similar issues, using keywords relevant to my problem.
- [ ] I clicked the `Log` button in Rufus and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking the "bad blocks" check box in Rufus, and confirmed that my USB is not defective
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive
  - [ ] Plugging the USB into a different port
  - [ ] Running Rufus on a different computer
- [ ] If using an ISO image, I clicked on the `#` button (at the bottom of the Rufus interface), to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<Please describe your issue here>

Log
---
```
<FULL LOG>
```

pbatard, 2017-01-04:
Thanks for the report. Can you try [this version](https://rufus.akeo.ie/testing/rufus-2.12_TEST1.exe) ant let me know if you still see the issue?

Kerzenflamme, 2017-01-05:
I went to a shop today and bought a new 16gb drive and that worked without a problem. 

thats the error message with your test build:

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE4 for write access
Ignoring drive '\\.\E:' as it spans multiple disks (RAID?)
Will use 'J:' as volume mountpoint
I/O boundary checks disabled
Analyzing existing boot records...
Drive has a Windows 7 Master Boot Record
Drive has a FAT32 FreeDOS Partition Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 2048 sectors
Deleting partitions...
Writing Image...
Timeout while waiting for logical drive
Drive has an unknown Master Boot Record

Found USB 1.0 device 'SanDisk Cruzer Blade USB Device' (????:????)
Ignoring drive '\\.\E:' as it spans multiple disks (RAID?)
Ignoring drive '\\.\E:' as it spans multiple disks (RAID?)
Found USB 2.0 device 'ZTE MMC Storage USB Device' (19D2:1403)
Device eliminated because it appears to contain no media
1 device found
No volume information for drive 0x84
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 7782, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x467E850F
Drive has an unknown Master Boot Record
Partition 2:
  Type: EFI FAT (0xef)
  Size: 30 MB (31457280 bytes)
  Start Sector: 212, Boot: No

Kerzenflamme, 2017-01-05:
the device which worked btw was also a SD cruzer disk, but a 16gb one.
anyways, thats the message

```
Processing Hub 1:
  Hub[1] = '\\?\usb#root_hub#4&209dff44&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[086]: USB\VID_1B1C&PID_0C04\5&15D9C317&0&4
Processing Hub 2:
  Hub[2] = '\\?\usb#root_hub#4&21f60ac4&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[027]: USB\VID_413C&PID_2107\5&1CC56EFC&0&3
Processing Hub 3:
Processing Hub 4:
Processing Hub 5:
Processing Hub 6:
  Hub[3] = '\\?\usb#root_hub20#4&37566865&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[245]: USB\VID_19D2&PID_1403\MF8230ZTED010000CP261718SKYDWQ5II0PESHF3BW707_8C64&&&&&&&&&&&&&0
  Found ID[060]: USB\VID_0781&PID_5567\4C531001410422106201
Processing Hub 7:
  Hub[4] = '\\?\usb#root_hub20#4&3892a9e2&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[184]: USB\VID_17CC&PID_0815\SN-Q7EORNE4MK___
Processing Hub 8:
Processing Hub 9:
  Hub[5] = '\\?\usb#vid_040e&pid_0100&asmediausbd_hub#5&3708c5e2&0&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[185]: USB\VID_1532&PID_001C\6&1918F7EE&0&3
Processing IDs belonging to 'USBSTOR':
  USB\VID_19D2&PID_1403&MI_02\7&74329b6&0&0002
  USB\VID_0781&PID_5567\4C531001410422106201
Processing 'SCSI' device:
  Hardware ID: 'SCSI\DiskCrucial_CT120M500SSD1___MU03'
Found non-USB non-removable device 'Crucial_ CT120M500SSD1 SATA Disk Device' => Eliminated
Processing 'SCSI' device:
  Hardware ID: 'SCSI\DiskTOSHIBA_DT01ACA200______MX4O'
Found non-USB non-removable device 'TOSHIBA DT01ACA200 SATA Disk Device' => Eliminated
Processing 'SCSI' device:
  Hardware ID: 'SCSI\DiskTOSHIBA_DT01ACA300______MX6O'
Found non-USB non-removable device 'TOSHIBA DT01ACA300 SATA Disk Device' => Eliminated
Processing 'USBSTOR' device:
  Hardware ID: 'USBSTOR\DiskSanDisk_Cruzer_Blade____1.27'
  Matched with ID[060]: USB\VID_0781&PID_5567\4C531001410422106201
  Matched with Hub[3]: '\\?\usb#root_hub20#4&37566865&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
Found USB 2.0 device 'SanDisk Cruzer Blade USB Device' (0781:5567)
Ignoring drive '\\.\E:' as it spans multiple disks (RAID?)
Using autorun.inf label for drive J: '64GB'
Ignoring drive '\\.\E:' as it spans multiple disks (RAID?)
Processing 'USBSTOR' device:
  Hardware ID: 'USBSTOR\DiskZTE_____MMC_Storage_____2.31'
NOTE: Matched instance from sibling for 'USB\VID_19D2&PID_1403&MI_02\7&74329b6&0&0002'
  Matched with ID[090]: USB\VID_19D2&PID_1403&MI_02\7&74329b6&0&0002
  Matched with (GP) ID[245]: USB\VID_19D2&PID_1403\MF8230ZTED010000CP261718SKYDWQ5II0PESHF3BW707_8C64&&&&&&&&&&&&&0
  Matched with Hub[3]: '\\?\usb#root_hub20#4&37566865&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
Found USB 2.0 device 'ZTE MMC Storage USB Device' (19D2:1403)
Device eliminated because it appears to contain no media
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 7782, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
```

pbatard, 2017-01-05:
Hmmm, I don't quite get how your new log says:
```
Found USB 2.0 device 'SanDisk Cruzer Blade USB Device' (0781:5567)
```
whereas previously you had:
```
Found USB 1.0 device 'SanDisk Cruzer Blade USB Device' (????:????)
```

Can you please confirm that, with the same device plugged in, you really see `USB 1.0 device 'SanDisk Cruzer Blade USB Device' (????:????)` when degug enumeration is disabled, and `USB 2.0 device 'SanDisk Cruzer Blade USB Device' (0781:5567)` with enumeration enabled, and that the device you are testing then is the one that gives the `Timeout while waiting for logical drive` message (which is the real issue at hand).

If possible, I'd like you to copy paste the __complete__ log from Rufus, with the problematic device plugged, where you first press <kbd>Alt</kbd>-<kbd>.</kbd> (to toggle debug enumeration - it doesn't really matter if it was enabled or disabled before, as Rufus will re-enumerate devices then, so I'll get both versions) and then try to write your image.

I think that with a complete log like this, I might be able to see a bit clearer...

Also, since you say that between two SanDisk Cruzer Blade, there's one working and one that doesn't, I'd appreciate if you could tell me if the following part is the same for both devices:
```
Processing 'USBSTOR' device:
  Hardware ID: 'USBSTOR\DiskSanDisk_Cruzer_Blade____1.27'
```

Especially, I suspect the 1.27 might be a firmware version for the device, and if you don't get the same version for both devices, then that may mean that there's a bug with some Cruzer Blade firmwares.

Kerzenflamme, 2017-01-05:
yes they are the same devices. The other unbranded usb stick (it's actually an SD card in an adapter) i have gives the same error as the Cruzer Blade dongle when burning the DD image, as above.

interstingly enough this time round it's being labeled as an USB 1.0 device again in debug mode:
(ZTE MMC Storage USB Device is my phone i think btw)

```
Processing Hub 1:
  Hub[1] = '\\?\usb#root_hub#4&209dff44&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[086]: USB\VID_1B1C&PID_0C04\5&15D9C317&0&4
Processing Hub 2:
  Hub[2] = '\\?\usb#root_hub#4&21f60ac4&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[027]: USB\VID_413C&PID_2107\5&1CC56EFC&0&3
Processing Hub 3:
Processing Hub 4:
  Hub[3] = '\\?\usb#root_hub#4&ca1a60d&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[010]: USB\VID_1532&PID_001C\5&3AFD18BD&0&3
Processing Hub 5:
Processing Hub 6:
  Hub[4] = '\\?\usb#root_hub20#4&37566865&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[245]: USB\VID_19D2&PID_1403\MF8230ZTED010000CP261718SKYDWQ5II0PESHF3BW707_8C64&&&&&&&&&&&&&0
Processing Hub 7:
  Hub[5] = '\\?\usb#root_hub20#4&3892a9e2&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[184]: USB\VID_17CC&PID_0815\SN-Q7EORNE4MK___
Processing Hub 8:
Processing Hub 9:
  Hub[6] = '\\?\usb#vid_040e&pid_0100&asmediausbd_hub#5&3708c5e2&0&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[060]: USB\VID_0781&PID_5567\4C531001410422106201
Processing IDs belonging to 'USBSTOR':
  USB\VID_0781&PID_5567\4C531001410422106201
  USB\VID_19D2&PID_1403&MI_02\7&74329b6&0&0002
Processing 'SCSI' device:
  Hardware ID: 'SCSI\DiskCrucial_CT120M500SSD1___MU03'
Found non-USB non-removable device 'Crucial_ CT120M500SSD1 SATA Disk Device' => Eliminated
Processing 'SCSI' device:
  Hardware ID: 'SCSI\DiskTOSHIBA_DT01ACA200______MX4O'
Found non-USB non-removable device 'TOSHIBA DT01ACA200 SATA Disk Device' => Eliminated
Processing 'SCSI' device:
  Hardware ID: 'SCSI\DiskTOSHIBA_DT01ACA300______MX6O'
Found non-USB non-removable device 'TOSHIBA DT01ACA300 SATA Disk Device' => Eliminated
Processing 'USBSTOR' device:
  Hardware ID: 'USBSTOR\DiskSanDisk_Cruzer_Blade____1.27'
  Matched with ID[060]: USB\VID_0781&PID_5567\4C531001410422106201
  Matched with Hub[6]: '\\?\usb#vid_040e&pid_0100&asmediausbd_hub#5&3708c5e2&0&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
Found USB 1.0 device 'SanDisk Cruzer Blade USB Device' (????:????)
Ignoring drive '\\.\E:' as it spans multiple disks (RAID?)
Using autorun.inf label for drive J: '64GB'
Ignoring drive '\\.\E:' as it spans multiple disks (RAID?)
Processing 'USBSTOR' device:
  Hardware ID: 'USBSTOR\DiskZTE_____MMC_Storage_____2.31'
NOTE: Matched instance from sibling for 'USB\VID_19D2&PID_1403&MI_02\7&74329b6&0&0002'
  Matched with ID[090]: USB\VID_19D2&PID_1403&MI_02\7&74329b6&0&0002
  Matched with (GP) ID[245]: USB\VID_19D2&PID_1403\MF8230ZTED010000CP261718SKYDWQ5II0PESHF3BW707_8C64&&&&&&&&&&&&&0
  Matched with Hub[4]: '\\?\usb#root_hub20#4&37566865&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
Found USB 2.0 device 'ZTE MMC Storage USB Device' (19D2:1403)
Device eliminated because it appears to contain no media
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 7782, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x000701E3
Drive has a Windows 7 Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 59.6 GB (64015171584 bytes)
```

the 16gb usb stick (it looks completly different btw) i actually cannot find atm, had a look everywhere in my room. X_x

Kerzenflamme, 2017-01-05:
Complete log for debug and burning

`Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/isolinux/isolinux.cfg' for Syslinux
  Detected Syslinux version: 6.03 (from '/isolinux/isolinux.bin')
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'SolusLive'
  Size: 946 MB
  Uses: Syslinux/Isolinux v6.03
Using image: Solus-1.2.0.5.iso
Found USB 1.0 device 'SanDisk Cruzer Blade USB Device' (????:????)
Ignoring drive '\\.\E:' as it spans multiple disks (RAID?)
Using autorun.inf label for drive J: '64GB'
Ignoring drive '\\.\E:' as it spans multiple disks (RAID?)
Found USB 2.0 device 'ZTE MMC Storage USB Device' (19D2:1403)
Device eliminated because it appears to contain no media
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 7782, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x000701E3
Drive has a Windows 7 Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 59.6 GB (64015171584 bytes)
  Start Sector: 2048, Boot: Yes
Processing Hub 1:
  Hub[1] = '\\?\usb#root_hub#4&209dff44&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[086]: USB\VID_1B1C&PID_0C04\5&15D9C317&0&4
Processing Hub 2:
  Hub[2] = '\\?\usb#root_hub#4&21f60ac4&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[027]: USB\VID_413C&PID_2107\5&1CC56EFC&0&3
Processing Hub 3:
Processing Hub 4:
  Hub[3] = '\\?\usb#root_hub#4&ca1a60d&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[010]: USB\VID_1532&PID_001C\5&3AFD18BD&0&3
Processing Hub 5:
Processing Hub 6:
  Hub[4] = '\\?\usb#root_hub20#4&37566865&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[245]: USB\VID_19D2&PID_1403\MF8230ZTED010000CP261718SKYDWQ5II0PESHF3BW707_8C64&&&&&&&&&&&&&0
Processing Hub 7:
  Hub[5] = '\\?\usb#root_hub20#4&3892a9e2&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[184]: USB\VID_17CC&PID_0815\SN-Q7EORNE4MK___
Processing Hub 8:
Processing Hub 9:
  Hub[6] = '\\?\usb#vid_040e&pid_0100&asmediausbd_hub#5&3708c5e2&0&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[060]: USB\VID_0781&PID_5567\4C531001410422106201
Processing IDs belonging to 'USBSTOR':
  USB\VID_0781&PID_5567\4C531001410422106201
  USB\VID_19D2&PID_1403&MI_02\7&74329b6&0&0002
Processing 'SCSI' device:
  Hardware ID: 'SCSI\DiskCrucial_CT120M500SSD1___MU03'
Found non-USB non-removable device 'Crucial_ CT120M500SSD1 SATA Disk Device' => Eliminated
Processing 'SCSI' device:
  Hardware ID: 'SCSI\DiskTOSHIBA_DT01ACA200______MX4O'
Found non-USB non-removable device 'TOSHIBA DT01ACA200 SATA Disk Device' => Eliminated
Processing 'SCSI' device:
  Hardware ID: 'SCSI\DiskTOSHIBA_DT01ACA300______MX6O'
Found non-USB non-removable device 'TOSHIBA DT01ACA300 SATA Disk Device' => Eliminated
Processing 'USBSTOR' device:
  Hardware ID: 'USBSTOR\DiskSanDisk_Cruzer_Blade____1.27'
  Matched with ID[060]: USB\VID_0781&PID_5567\4C531001410422106201
  Matched with Hub[6]: '\\?\usb#vid_040e&pid_0100&asmediausbd_hub#5&3708c5e2&0&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
Found USB 1.0 device 'SanDisk Cruzer Blade USB Device' (????:????)
Ignoring drive '\\.\E:' as it spans multiple disks (RAID?)
Using autorun.inf label for drive J: '64GB'
[... 44 lines cut ...]
Drive has a FAT32 FreeDOS Partition Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 2048 sectors
Deleting partitions...
Writing Image...
Timeout while waiting for logical drive
Drive has an unknown Master Boot Record

Found USB 1.0 device 'SanDisk Cruzer Blade USB Device' (????:????)
Ignoring drive '\\.\E:' as it spans multiple disks (RAID?)
Ignoring drive '\\.\E:' as it spans multiple disks (RAID?)
Found USB 2.0 device 'ZTE MMC Storage USB Device' (19D2:1403)
Device eliminated because it appears to contain no media
1 device found
No volume information for drive 0x83
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 7782, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x467E850F
Drive has an unknown Master Boot Record
Partition 2:
  Type: EFI FAT (0xef)
  Size: 30 MB (31457280 bytes)
  Start Sector: 212, Boot: No
`

pbatard, 2017-01-05:
Thanks for the logs!

I see a few things of interest here:
1. When we get `USB 1.0 device 'SanDisk Cruzer Blade USB Device' (????:????)` your device is being enumerated on an ASMedia USB 3.0 Hub (`Matched with Hub[6]: '\\?\usb#vid_040e&pid_0100&asmediausbd_hub...`)
2. When we get `USB 2.0 device 'SanDisk Cruzer Blade USB Device' (0781:5567)` your device is being enumerated on a standard USB 2.0 Hub (`Matched with Hub[3]: '\\?\usb#root_hub20...`)
3. You are using Windows 7, which does not have native USB 3.0 support and relies on __proprietary__ drivers for anything USB 3.0, some of which are __known to be buggy__. Especially, I see that the ASMedia driver does not support __standard__ API requests to retrieve the device speed and other features (which I've seen happen with other drivers as well, as they tend to be a lot less comprehensive and and lot less tested than Microsoft's native ones).

So the first thing I would advise you to try is plug your flash drive on an USB 2.0 port (so that you get `USB 2.0 device 'SanDisk Cruzer Blade USB Device' (0781:5567)` in Rufus) and report what you get then.

Also, you may want to check if ASMedia has newer drivers and/or firmwares for the device you have in your PC, and install those.

Or, you could also try to upgrade to Windows 10, which provides full feature native USB 3.0 support. 😉

Kerzenflamme, 2017-01-06:
windows 10? never! why do you think i burned a linux usb for my laptop lol.

Well actually my Laptop was running windows 10 when it came and had exactly the same error as i described above in my first post. So maybe its nothing to do with the drivers? I can't show you the log as that laptop is now running solus linux....

pbatard, 2017-01-06:
> So maybe its nothing to do with the drivers?

That's why I asked you to test on an USB 2.0 port. Can you please do that?

And just for the record, not everything in Windows 10 (or earlier versions) is because Microsoft wants people to upgrade for the sake of upgrading. There's a lot to say about Windows 10, with very questionable choices for users, but as a software developer, I can tell you that there are many things that more modern versions of Windows do that users do benefit from, and the the idea that "What more could I need from Windows 7?" is quite shortsighted. For instance, besides the atrocious state of proprietary USB 3.0 drivers, Rufus cannot support Windows To Go on Windows 7 (even though it doesn't rely on anything from the OS that has anything to do with Windows To Go), due to long missing functionality that was introduced in Windows 8. In other words, software developers tend to upgrade their software for __very good reasons__, most of which being for the benefit of their users.

Kerzenflamme, 2017-01-07:
This is on an **USB 3.0** port: (It lists it as an usb 1.0 device, although when i plugged it in rufus listed it as an Usb 2.0 device)

```
Processing Hub 1:
  Hub[1] = '\\?\usb#root_hub#4&209dff44&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[086]: USB\VID_1B1C&PID_0C04\5&15D9C317&0&4
Processing Hub 2:
  Hub[2] = '\\?\usb#root_hub#4&21f60ac4&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[027]: USB\VID_413C&PID_2107\5&1CC56EFC&0&3
Processing Hub 3:
Processing Hub 4:
  Hub[3] = '\\?\usb#root_hub#4&ca1a60d&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[010]: USB\VID_1532&PID_001C\5&3AFD18BD&0&3
Processing Hub 5:
Processing Hub 6:
  Hub[4] = '\\?\usb#root_hub20#4&37566865&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[245]: USB\VID_19D2&PID_1403\MF8230ZTED010000CP261718SKYDWQ5II0PESHF3BW707_8C64&&&&&&&&&&&&&0
Processing Hub 7:
  Hub[5] = '\\?\usb#root_hub20#4&3892a9e2&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[184]: USB\VID_17CC&PID_0815\SN-Q7EORNE4MK___
Processing Hub 8:
  Hub[6] = '\\?\usb#vid_040e&pid_0100&asmediausbd_hub#5&2ae38653&0&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[060]: USB\VID_0781&PID_5567\4C531001410422106201
Processing Hub 9:
Processing IDs belonging to 'USBSTOR':
  USB\VID_19D2&PID_1403&MI_02\7&74329b6&0&0002
  USB\VID_0781&PID_5567\4C531001410422106201
Processing 'SCSI' device:
  Hardware ID: 'SCSI\DiskCrucial_CT120M500SSD1___MU03'
Found non-USB non-removable device 'Crucial_ CT120M500SSD1 SATA Disk Device' => Eliminated
Processing 'SCSI' device:
  Hardware ID: 'SCSI\DiskTOSHIBA_DT01ACA200______MX4O'
Found non-USB non-removable device 'TOSHIBA DT01ACA200 SATA Disk Device' => Eliminated
Processing 'SCSI' device:
  Hardware ID: 'SCSI\DiskTOSHIBA_DT01ACA300______MX6O'
Found non-USB non-removable device 'TOSHIBA DT01ACA300 SATA Disk Device' => Eliminated
Processing 'USBSTOR' device:
  Hardware ID: 'USBSTOR\DiskSanDisk_Cruzer_Blade____1.27'
  Matched with ID[060]: USB\VID_0781&PID_5567\4C531001410422106201
  Matched with Hub[6]: '\\?\usb#vid_040e&pid_0100&asmediausbd_hub#5&2ae38653&0&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
Found USB 1.0 device 'SanDisk Cruzer Blade USB Device' (????:????)
Ignoring drive '\\.\E:' as it spans multiple disks (RAID?)
Ignoring drive '\\.\E:' as it spans multiple disks (RAID?)
Processing 'USBSTOR' device:
  Hardware ID: 'USBSTOR\DiskZTE_____MMC_Storage_____2.31'
NOTE: Matched instance from sibling for 'USB\VID_19D2&PID_1403&MI_02\7&74329b6&0&0002'
  Matched with ID[090]: USB\VID_19D2&PID_1403&MI_02\7&74329b6&0&0002
  Matched with (GP) ID[245]: USB\VID_19D2&PID_1403\MF8230ZTED010000CP261718SKYDWQ5II0PESHF3BW707_8C64&&&&&&&&&&&&&0
  Matched with Hub[4]: '\\?\usb#root_hub20#4&37566865&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
Found USB 2.0 device 'ZTE MMC Storage USB Device' (19D2:1403)
Device eliminated because it appears to contain no media
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 7782, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x467E850F
Drive has an unknown Master Boot Record
Partition 2:
  Type: EFI FAT (0xef)
  Size: 30 MB (31457280 bytes)
  Start Sector: 212, Boot: No
Found USB 1.0 device 'SanDisk Cruzer Blade USB Device' (????:????)
Ignoring drive '\\.\E:' as it spans multiple disks (RAID?)
Ignoring drive '\\.\E:' as it spans multiple disks (RAID?)
Found USB 2.0 device 'ZTE MMC Storage USB Device' (19D2:1403)
Device eliminated because it appears to contain no media
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 7782, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
[... 59 lines cut ...]
Cylinders: 7782, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x467E850F
Drive has an unknown Master Boot Record
Partition 2:
  Type: EFI FAT (0xef)
  Size: 30 MB (31457280 bytes)
  Start Sector: 212, Boot: No
Found USB 2.0 device 'SanDisk Cruzer Blade USB Device' (0781:5567)
Ignoring drive '\\.\E:' as it spans multiple disks (RAID?)
Ignoring drive '\\.\E:' as it spans multiple disks (RAID?)
Found USB 2.0 device 'ZTE MMC Storage USB Device' (19D2:1403)
Device eliminated because it appears to contain no media
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 7782, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x467E850F
Drive has an unknown Master Boot Record
Partition 2:
  Type: EFI FAT (0xef)
  Size: 30 MB (31457280 bytes)
  Start Sector: 212, Boot: No
```
I doublechecked with another usb 3.0 and usb 2.0 port. It still seems to say usb 1.0 on the 3.0 port and 2.0 on the 2.0 port. strange. I swear I never actually used the usb 3.0 ports...

pbatard, 2017-01-07:
So, what happens if you try to create the USB when the drive is reported as USB 2.0?
Can you please post that part of the log?

Kerzenflamme, 2017-01-07:
sorry, i was gonna edit this in but forgot to click comment.

Still the same as issue as before:

`Found USB 2.0 device 'SanDisk Cruzer Blade USB Device' (0781:5567)
Ignoring drive '\\.\E:' as it spans multiple disks (RAID?)
Using autorun.inf label for drive J: '64GB'
Ignoring drive '\\.\E:' as it spans multiple disks (RAID?)
Found USB 2.0 device 'ZTE MMC Storage USB Device' (19D2:1403)
Device eliminated because it appears to contain no media
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 7782, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x0002AF8F
Drive has a Windows 7 Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 59.6 GB (64015171584 bytes)
  Start Sector: 2048, Boot: Yes
Processing Hub 1:
  Hub[1] = '\\?\usb#root_hub#4&209dff44&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[086]: USB\VID_1B1C&PID_0C04\5&15D9C317&0&4
Processing Hub 2:
  Hub[2] = '\\?\usb#root_hub#4&21f60ac4&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[027]: USB\VID_413C&PID_2107\5&1CC56EFC&0&3
Processing Hub 3:
Processing Hub 4:
  Hub[3] = '\\?\usb#root_hub#4&ca1a60d&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[011]: USB\VID_1532&PID_001C\5&3AFD18BD&0&4
Processing Hub 5:
Processing Hub 6:
  Hub[4] = '\\?\usb#root_hub20#4&37566865&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[245]: USB\VID_19D2&PID_1403\MF8230ZTED010000CP261718SKYDWQ5II0PESHF3BW707_8C64&&&&&&&&&&&&&0
  Found ID[060]: USB\VID_0781&PID_5567\4C531001410422106201
Processing Hub 7:
  Hub[5] = '\\?\usb#root_hub20#4&3892a9e2&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[184]: USB\VID_17CC&PID_0815\SN-Q7EORNE4MK___
Processing Hub 8:
Processing Hub 9:
Processing IDs belonging to 'USBSTOR':
  USB\VID_19D2&PID_1403&MI_02\7&74329b6&0&0002
  USB\VID_0781&PID_5567\4C531001410422106201
Processing 'SCSI' device:
  Hardware ID: 'SCSI\DiskCrucial_CT120M500SSD1___MU03'
Found non-USB non-removable device 'Crucial_ CT120M500SSD1 SATA Disk Device' => Eliminated
Processing 'SCSI' device:
  Hardware ID: 'SCSI\DiskTOSHIBA_DT01ACA200______MX4O'
Found non-USB non-removable device 'TOSHIBA DT01ACA200 SATA Disk Device' => Eliminated
Processing 'SCSI' device:
  Hardware ID: 'SCSI\DiskTOSHIBA_DT01ACA300______MX6O'
Found non-USB non-removable device 'TOSHIBA DT01ACA300 SATA Disk Device' => Eliminated
Processing 'USBSTOR' device:
  Hardware ID: 'USBSTOR\DiskSanDisk_Cruzer_Blade____1.27'
  Matched with ID[060]: USB\VID_0781&PID_5567\4C531001410422106201
  Matched with Hub[4]: '\\?\usb#root_hub20#4&37566865&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
Found USB 2.0 device 'SanDisk Cruzer Blade USB Device' (0781:5567)
Ignoring drive '\\.\E:' as it spans multiple disks (RAID?)
Using autorun.inf label for drive J: '64GB'
Ignoring drive '\\.\E:' as it spans multiple disks (RAID?)
Processing 'USBSTOR' device:
  Hardware ID: 'USBSTOR\DiskZTE_____MMC_Storage_____2.31'
NOTE: Matched instance from sibling for 'USB\VID_19D2&PID_1403&MI_02\7&74329b6&0&0002'
  Matched with ID[090]: USB\VID_19D2&PID_1403&MI_02\7&74329b6&0&0002
  Matched with (GP) ID[245]: USB\VID_19D2&PID_1403\MF8230ZTED010000CP261718SKYDWQ5II0PESHF3BW707_8C64&&&&&&&&&&&&&0
  Matched with Hub[4]: '\\?\usb#root_hub20#4&37566865&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
Found USB 2.0 device 'ZTE MMC Storage USB Device' (19D2:1403)
Device eliminated because it appears to contain no media
1 device found
Disk type: Removable, Sector Size: 512 bytes
[... 58 lines cut ...]
  Hardware ID: 'USBSTOR\DiskSanDisk_Cruzer_Blade____1.27'
  Matched with ID[060]: USB\VID_0781&PID_5567\4C531001410422106201
  Matched with Hub[4]: '\\?\usb#root_hub20#4&37566865&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
Found USB 2.0 device 'SanDisk Cruzer Blade USB Device' (0781:5567)
Ignoring drive '\\.\E:' as it spans multiple disks (RAID?)
Ignoring drive '\\.\E:' as it spans multiple disks (RAID?)
Processing 'USBSTOR' device:
  Hardware ID: 'USBSTOR\DiskZTE_____MMC_Storage_____2.31'
NOTE: Matched instance from sibling for 'USB\VID_19D2&PID_1403&MI_02\7&74329b6&0&0002'
  Matched with ID[090]: USB\VID_19D2&PID_1403&MI_02\7&74329b6&0&0002
  Matched with (GP) ID[245]: USB\VID_19D2&PID_1403\MF8230ZTED010000CP261718SKYDWQ5II0PESHF3BW707_8C64&&&&&&&&&&&&&0
  Matched with Hub[4]: '\\?\usb#root_hub20#4&37566865&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
Found USB 2.0 device 'ZTE MMC Storage USB Device' (19D2:1403)
Device eliminated because it appears to contain no media
1 device found
No volume information for drive 0x84
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 7782, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x467E850F
Drive has an unknown Master Boot Record
Partition 2:
  Type: EFI FAT (0xef)
  Size: 30 MB (31457280 bytes)
  Start Sector: 212, Boot: No`

pbatard, 2017-01-07:
It's starting to dawn on me that I'm not seeing any kind of actual errors in all these logs.
Have you actually tried the USB drive?

Note that it's not because you can't see your drive's content in Windows that it isn't working, __especially__ when writing in DD-mode (as the image may be created in a manner that is completely alien to Windows), even more so as the `Timeout while waiting for logical drive` message intervenes __after__ the image has been written to USB, and is linked to Rufus attempting to see if there's any partition that might be mountable in Windows, which is never a given.

Unless you see Rufus stop its operations before the progress bar goes to 100% and actually create a popup to report an error (remember, what you may see in the log might not have anything to do with an error, and be a perfectly normal report that some non-critical or optional aspect of the process didn't complete), then Rufus __successfully__ completed the USB creation process, and any issue you encounter when trying to boot the drive may have to do with your BIOS settings, etc.

Of course you may wonder why one drive might report a Timeout when trying to access the EFI System Partition (ESP) on one drive (the `EFI FAT` partition from your log), and not another, but this can be explained by the presence of absence of the FIXED attribute on your USB (which is a hardware property of the drive). I know that Microsoft went some length to prevent access to to the ESP, and that wehther a drive is FIXED or REMOVABLE also has some incidence to it.

Thus, at this stage, I'd really like to know: Do you actually see an error when trying to boot the Solus USB created by Rufus. And if so, what is that error?

Kerzenflamme, 2017-01-07:
I have obviously tried the finished usb stick on my laptop. The problem is still the same as before.

When you put the usb into the laptop it does go to the boot menu and you can pick "solus 1.0.2.5" but it will then not actually boot the system but just give lots of error messages and cannot load the system.  
I assume partititon 1 which made the stick bootable has been written by Rufus but not partition 2 which actually has got the iso on it.

I have just written the image to the stick again and taken a picture of what happens after you try to boot solus from the usb stick:

[img]http://www2.pic-upload.de/img/32444589/IMG_20170107_193327.jpg[/imh]

What I can tell you is that the 16gb USB stick (which i still cant find lol) said in the log that it had a zeroed master boot record rather than unknown master boot record.

pbatard, 2017-01-07:
> I guess partititon 1 which made the stick bootable has been written by Rufus but not partition 2 which actually has got the iso on it.

No, that's not how it works. An ISOHybrid contains ALL the partitions, and Rufus writes everything at once. In other words, if the ISOHybrid has 10 partitions, and Rufus went to 100%, then the 10 partitions were written. In DD mode, Rufus has actually __no awareness__ of partitions, or MBR, or anything. In fact, the DD writing process is so straightforward that the full content of the image is written exactly __as is__ onto the USB, with the expectation that whoever created the image properly set all the partitions, boot records, EFI loaders and so on. If you were to open your ISO with an hex editor, and then copy every byte you see onto the USB, then you will achieve exactly what Rufus is doing in DD Image mode.

Thus, it looks to me like, whatever error you are getting on boot is something that the Solus people should help you with, and that you will get regardless of the tool you use to write a DD Image. For instance, can you tell me what happens if you use [Win32DiskImager](https://sourceforge.net/projects/win32diskimager/)? Clearly, if you get the same error, then it demonstrates that this has nothing to do with Rufus in the first place.

Also, is there any specific reason you used DD Image mode rather than ISO Image mode in Rufus?

Kerzenflamme, 2017-01-07:
eh as far as they are concerned the problem is solved and the usb sticks need testing as it worked with another one. Its really not a big deal as it works fine with my other usb stick. i just keep posting out of interest. Might be worth trying another DD image but i dont have one at hand.

I'm not sure why it has to be in DD mode. If the image is written in iso mode its not bootable. That was the only reason I found in their forums "Those options are all incorrect and will butcher the USB. You need to use a recent Rufus version, and write it in DD mode, as
ours is an isohybrid ISO"

Win32 disk manager wont open the iso. Only seems to want .img files. If you try to open the solus iso anyways it will say "the file is set to read only. Try again with a different file name".

pbatard, 2017-01-08:
> Win32 disk manager wont open the iso. If you try to open the solus iso anyways it will say "the file is set to read only. Try again with a different file name".

That's not what I see on my side. I was able to open the `Solus-1.2.0.5.iso` after applying the `*.*` filter, and Win32 Disk Imager wrote it without issue. Thus, if you also encounter issues against some of your USB drives when using Win32 Disk Imager, it seems pretty clear that the problem resides with your USB flash drives, and nothing to do with Rufus. As such, I will close this issue.

Oh, and for the record, the Solus people seem to have deliberately chosen not to include a `efi/boot/bootx64.efi` bootloader on their EFI partition, so I can't help to find it strange that they're accusing Rufus of "butchering the USB" in ISO mode, when they are the ones who deliberately chose to ignore the UEFI specs in the first place... When followed properly, UEFI actually makes it very easy to boot stuff by simply copying files over onto a FAT32 partition, which makes the requirement of using tools like Rufus or Win32 Disk Imager obsolete. So it's a bit gratuitous of them of pointing the finger at external tools, when it's the limitation of their boot process that actually limits users to DD image mode only.

closed 2017-01-08 by pbatard