pbatard/rufus #881: 0 devices detected

NuunuMoeller, 2017-01-04:
Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues, using keywords relevant to my problem.
- [x] I clicked the `Log` button in Rufus and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking the "bad blocks" check box in Rufus, and confirmed that my USB is not defective
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive
  - [ ] Plugging the USB into a different port
  - [ ] Running Rufus on a different computer
- [ ] If using an ISO image, I clicked on the `#` button (at the bottom of the Rufus interface), to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<0 devices are detected when I try to run RUFUS.> 

Log
---
```
<Rufus version: 2.11.995
Windows version: Windows XP SP3 64-bit
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta3
System locale ID: 0x0406
Will use default UI locale 0x0406
NOTE: The Danish (Dansk) translation requires an update, but the current translator hasn't submitted one. Because of this, some messages will only be displayed in English.
If you think you can help update this translation, please e-mail the author of this application
0 devices found>
```

pbatard, 2017-01-04:
What is the nature of __USB__ device you are trying to use with Rufus.
Please be mindful that unless it's an __USB__ mass storage device, it will not be detected.

What's the exact manufacturer and model of your device?

pbatard, 2017-01-04:
Also, is there a reason why you didn't attempt to perform these steps:
> - [ ] I also tried one or more of the following:
>   - [ ] Using a different USB drive
>   - [ ] Plugging the USB into a different port
They are requested as part of the checklist precisely to help people who can't see their device being detected, so please don't skip them!

NuunuMoeller, 2017-01-04:
Thank you for your reply. I just tried with a different USB drive with 4,1 GB, and I also tried a hard disk. I always get 0 devices detected.
Maybe I should try running it with a different computer.

pbatard, 2017-01-04:
This is strange. Do you see these devices in Windows _Device Manager_? And are you using a specific driver on these USB ports (I know that ASUS has some USB 3.0 special driver that users may install in Windows, and that can prevent regular applications from accessing devices plugged on such ports).

If you still can't see any device, please try the following:
- Press <kbd>Alt<kbd>-<kbd>.<kbd> in Rufus and post the new log you get.

This enables enumeration debug, and you should see debug information similar to the one below if you do it properly:
```
Processing Hub 1:
  Hub[1] = '\\?\usb#vid_04b3&pid_301a#5&23ce17bd&0&12#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[063]: USB\VID_04B3&PID_301B\6&1D18823C&0&1
  Found ID[068]: USB\VID_046D&PID_C03E\6&1D18823C&0&3
Processing Hub 2:
Processing Hub 3:
  Hub[2] = '\\?\usb#vid_045b&pid_0209#5&23ce17bd&0&9#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[158]: USB\VID_1A86&PID_7584\6&2489E876&0&2
Processing Hub 4:
  Hub[3] = '\\?\usb#root_hub20#4&e427330&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[018]: USB\VID_8087&PID_8002\5&9D9154E&0&1
Processing Hub 5:
Processing Hub 6:
Processing Hub 7:
  Hub[4] = '\\?\usb#root_hub20#4&935f5ea&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[074]: USB\VID_8087&PID_800A\5&27276061&0&1
Processing Hub 8:
  Hub[5] = '\\?\usb#vid_0424&pid_2134#5&23ce17bd&0&7#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[081]: USB\VID_045E&PID_028E\1F3251C
  Found ID[162]: USB\VID_047F&PID_C009\6&2FB23F66&0&3
Processing Hub 9:
  Hub[6] = '\\?\usb#root_hub30#4&2b7d45a3&0&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[202]: USB\VID_045B&PID_0210\5&23CE17BD&0&20
  Found ID[161]: USB\VID_0424&PID_2134\5&23CE17BD&0&7
  Found ID[179]: USB\VID_045B&PID_0209\5&23CE17BD&0&9
  Found ID[177]: USB\VID_04B3&PID_301A\5&23CE17BD&0&12
Processing 'SCSI' device:
  Hardware ID: 'SCSI\DiskSAMSUNG_MZHPV256HDGL-000BXW2'
Found non-USB non-removable device 'SAMSUNG MZHPV256HDGL-00000' => Eliminated
Processing 'SCSI' device:
  Hardware ID: 'SCSI\Disk________ST4000DM000-1F21CC54'
Found non-USB non-removable device 'ST4000DM000-1F2168' => Eliminated
Processing 'STORAGE' device:
  Disabled by policy
```

pbatard, 2017-01-07:
Any update on what I asked? If not, I will close the issue.

closed 2017-01-11 by pbatard