pbatard/rufus #884: Please add Cheat mode to allow FAT32 on "MBR partition scheme for BIOS or UEFI"

Hlsgs, 2017-01-10:
Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues, using keywords relevant to my problem.
- [x] I clicked the `Log` button in Rufus and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking the "bad blocks" check box in Rufus, and confirmed that my USB is not defective
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive
  - [ ] Plugging the USB into a different port
  - [ ] Running Rufus on a different computer
- [ ] If using an ISO image, I clicked on the `#` button (at the bottom of the Rufus interface), to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<Please describe your issue here>

Please add Cheat mode to allow FAT32 on "MBR partition scheme for BIOS or UEFI" atleast while using alt+E to enable hybrid UEFI/BIOS meadia creation.
This would be highly useful for us power users that know what we are doing and would not interfere with the way most people use Rufus being a cheat mode(i totally agree with your vision of enforcing BIOS or UEFI for standard users, as these can be difficult to grasp for a normal user).
Thanks for the this excellent tool, btw!

Log
---
```
<FULL LOG>
```

pbatard, 2017-01-10:
Already done in #816

closed 2017-01-10 by pbatard

Hlsgs, 2017-01-10:
Excellent! This will save me some headaches.
Can I get a beta to test?

pbatard, 2017-01-10:
The BETA will be available when it's available...

Hlsgs, 2017-01-10:
Oh, I thought i just couldn't find it.
Thank you and keep up the excellent work!