pbatard/rufus #886: USB Redirection

Clockorama, 2017-01-11:
Checklist
---------
- [ X] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [X ] I performed a search in the issue tracker for similar issues, using keywords relevant to my problem.
- [X ] I clicked the `Log` button in Rufus and copy/pasted the log into the line that says `<FULL LOG>` below.
- [X ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking the "bad blocks" check box in Rufus, and confirmed that my USB is not defective
- [ X] I also tried one or more of the following:
  - [ X] Using a different USB drive
  - [ X] Plugging the USB into a different port
  - [X ] Running Rufus on a different computer
- [ ] If using an ISO image, I clicked on the `#` button (at the bottom of the Rufus interface), to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Hi guys,
I'm using USB Redirection in a remote session. Basically my local USB drive is redirected to machine, where I have Rufus. 
Most probably this isn't meant to be working, but Rufus doesn't detect the USB. I tried alt+F, but again it doesn't detect anything (Local drives, USB redirected or other redirected drives). 

Could you advise?

Log
---
```
Rufus version: 2.11.995
Windows version: Windows Server 10 64-bit (Build 14393)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta3
System locale ID: 0x0409
Will use default UI locale 0x0409
0 devices found
0 devices found
```

pbatard, 2017-01-11:
I'm afraid that Rufus is only designed for use with local devices, and, as much as I'd like to be able to try to help with every request, there's only so much that my time allows, therefore I have to pass on improvements that, even if possible, are only likely to help a handful of people (especially as I don't really understand why, if you are remotely connecting to a computer, you cannot run Rufus on the machine where the USB device is plugged in the first place). Adding support for virtual USB remote devices is such a feature.

As such, I will close this request.

closed 2017-01-11 by pbatard

Clockorama, 2017-01-12:
Sounds fair. The reason behind running Rufus on the remote machine are several. 
Restrictions on the local machine. 
I have ISO library, that I cannot carry around with myself or download the image every time.
Flexibility.