pbatard/rufus #887: Rufus detecting 0

DeJayDevelopment, 2017-01-12:
Checklist
---------
- [X] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [X] I performed a search in the issue tracker for similar issues, using keywords relevant to my problem.
- [X] I clicked the `Log` button in Rufus and copy/pasted the log into the line that says `<FULL LOG>` below.
- [X] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [X] I ran a bad blocks check, by clicking the "bad blocks" check box in Rufus, and confirmed that my USB is not defective
- [X] I also tried one or more of the following:
  - [X] Using a different USB drive
  - [ ] Plugging the USB into a different port
  - [ ] Running Rufus on a different computer
- [] If using an ISO image, I clicked on the `#` button (at the bottom of the Rufus interface), to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Hi there! I'm running Windows 10 64 bit as shown in below log and Rufus 2.11.995 isn't picking up on ANY drives!
A gif of me doing Alt F and Alt (dot) as well as checking all the UI options.
http://i.dejaydev.com/nbEh3c.gif
I am looking for the newly created L drive courtesy of the Windows Disk Management which allowed me to turn storage from the D: (SSD) drive to a L: drive.

Yes, I have already seen https://github.com/pbatard/rufus/issues/134

Log
---
```
Rufus version: 2.11.995
Windows version: Windows 10 64-bit (Build 14393)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta3
System locale ID: 0x0409
Will use default UI locale 0x0409
0 devices found
0 devices found
0 devices found
0 devices found
Processing Hub 1:
  Hub[1] = '\\?\usb#root_hub30#4&390f7fea&0&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[123]: USB\VID_1532&PID_0046\5&350590A7&0&2
  Found ID[201]: USB\VID_04F2&PID_B414\0X0001
  Found ID[032]: USB\VID_0B05&PID_1819\5&350590A7&0&8
  Found ID[117]: USB\VID_1532&PID_0203\5&350590A7&0&6
  Found ID[025]: USB\VID_8087&PID_0A2A\5&350590A7&0&9
  Found ID[001]: USB\VID_0B05&PID_1822\5&350590A7&0&10
  Found ID[100]: USB\VID_12BA&PID_00FF\5&350590A7&0&1
Processing 'SCSI' device:
  Hardware ID: 'SCSI\DiskHGST____HTS721010A9E630_JB0O'
Found non-USB non-removable device 'HGST HTS721010A9E630' => Eliminated
0 devices found
```

pbatard, 2017-01-12:
Rufus is designed to only list USB and VHD drives. You're talking about an SSD drive which, by design, is not meant to be detected.

What is the type, manufacturer and model of the drive you are trying to detect?

pbatard, 2017-01-13:
Can you answer my questions above? If not, I will have to close this issue.

closed 2017-01-14 by pbatard

ZaveThe, 2017-06-22:
In my case it's a samsung

pbatard, 2017-06-22:
@ZaveThe, try <kbd>Alt</kbd>-<kbd>F</kbd> (or check [this entry from the FAQ](https://github.com/pbatard/rufus/wiki/FAQ#Help_my_USB_drive_is_not_detected)).

ZaveThe, 2017-06-22:
I tried. It, unfortunately, doesn't work

pbatard, 2017-06-22:
Please provide a full log after using <kbd>Alt</kbd>-<kbd>.</kbd> to enable enumeration debug. I cannot help you further without seeing your debug log.

ZaveThe, 2017-06-23:
> Please provide a full log after using Alt-. to enable enumeration debug.

It's okay, I got some help from a friend. Apparently the version on the
machine was out of date and after we downloaded the latest version, hit
Alt-F and it worked.

Thanks you for your time.