pbatard/rufus #888: 64GB Intenso USB stick is detected as Hard Drive

andreas-profitlich, 2017-01-17:
Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues, using keywords relevant to my problem.
- [x] I clicked the `Log` button in Rufus and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking the "bad blocks" check box in Rufus, and confirmed that my USB is not defective
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive
  - [ ] Plugging the USB into a different port
  - [ ] Running Rufus on a different computer
- [ ] If using an ISO image, I clicked on the `#` button (at the bottom of the Rufus interface), to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
64GB Intenso USB stick is detected as Hard Drive

Log
---
```
Rufus version: 2.11.995
Windows version: Windows 10 64-bit (Build 14393)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta3
System locale ID: 0x0407
Will use default UI locale 0x0407
Found USB 2.0 device 'Intenso Rainbow Line USB Device' (058F:6387)
Device eliminated because it was detected as a Hard Drive (score 1 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives in 'Advanced Options' (after clicking the white triangle)
0 devices found
0 devices found
Found USB 2.0 device 'Intenso Rainbow Line USB Device' (058F:6387)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 3926, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x8F3681C7
Drive has an unknown Master Boot Record
Partition 1:
  Type: exFAT (0x07)
  Size: 30.1 GB (32295178240 bytes)
  Start Sector: 1880, Boot: No
0 devices found
0 devices found
Found USB 2.0 device 'Intenso Rainbow Line USB Device' (058F:6387)
Device eliminated because it was detected as a Hard Drive (score 1 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives in 'Advanced Options' (after clicking the white triangle)
0 devices found
```

pbatard, 2017-01-17:
Thanks for the report. I'll make sure that device is detected as a flash drive in the next version of Rufus.

pbatard, 2017-01-17:
This will be fixed in the 2.12 release, which should be out by the end of the month.

commit 9f1f164a (2017-01-17): [dev] adjust score for Alcor based UFDs

* Closes #888
closed 2017-01-17 by pbatard