pbatard/rufus #891: Windows To Go Problems | Rufus 2.12 BETA

NSbuilder, 2017-01-17:
-------------------------------------------------------
Issue 1:
When you have to choose an edition for windows to go, the OK and CANCEL buttons are flickering when you hover on them with your mouse: [http://bit.ly/2jkZjAO](url)
You can also see the messages read MESSAGE_291_UNTRANSLATED and  MESSAGE_292_UNTRANSLATED ( I think you said it was on purpose but I don't know ).
It is no a big deal but I just want to report this.
-------------------------------------------------------
Issue 2 - I compare Rufus to WinToUSB:
I can choose the edition of windows to go with my windows 7 iso ( I think it is a customized iso ):
http://i.imgur.com/cof6zKN.png ( The Russian labels do not matter - it is the iso )

But when using the iso from windows 10 creation tool that have 2 install.esd ( bios + uefi, x86 + x64 ) it won't even allow me to create a Windows to go drive: http://i.imgur.com/TqgSrVD.png while WinToUSB have no problem doing so.
Would it be possible to add esd support for rufus or something like that?
-------------------------------------------------------

pbatard, 2017-01-17:
> When you have to choose an edition for windows to go, the OK and CANCEL buttons are flickering when you hover on them with your mouse

I'm not seeing that behaviour when testing. __PLEASE, PLEASE, PLEASE, ALWAYS provide a full log__ (even if you don't actually go through the USB creation). The checklist is there for a reason, and I could find out many things that you didn't tell from the log (such as the version of Windows you are using, the ISO, and other important stuff).

> You can also see the messages read MESSAGE_291_UNTRANSLATED and MESSAGE_292_UNTRANSLATED

That shouldn't be the case, and right now, I can't see how that's even possible for the English version. Again, I'm really curious about the log data, because I can't guess anything about the cause of the issue from the youtube video... Also, are you sure you don't have an old `rufus.loc` in the same directory as the one you are testing Rufus? If that is the case, then that would explain your issue, and you should delete that `rufus.loc` file.

> I think you said it was on purpose but I don't know

I never said that.

Be careful about __assuming__ things (such as that you don't need to provide a log, or that I said something I didn't say), because this really gets in the way of trying to identify and fix potential issues.

> Issue 2

__Please__, if you want to report multiple issues (or an issue and and enhancement request), I will ask you to create 2 separate entries in the issue tracker. This will help everyone.

> I compare Rufus to WinToUSB

Don't. You'll find that I generally have little interest in trying to copy what others do, just for the sake of it.

> But when using the ISO from windows 10 creation tool that have 2 install.esd ( bios + uefi, x86 + x64 ) it won't even allow me to create a WTG drive. Would it be possible to add esd support for Rufus or something like that?

I __ALREADY__ explained that Rufus does not support ESD, when you asked [earlier today](https://github.com/pbatard/rufus/commit/0d45bae5e56883d2af04712c97464133701e199f#commitcomment-20506028), and that I have no plans to do so, as I don't see it as a priority. There are just more important features that are planned (some of which are not mentioned in the issue tracker, and, no, my priority list is not up to discussion).

> but, I think Rufus should be able to create a vhd and vhdx based windows to go drives because it can be better than normal windows to go drive sometimes.

I disagree, because it deviates from the official Microsoft process, and anything that deviates from Microsoft recommendations can be broken at any moment. Also it adds one level of virtualization (the disk is not being accessed directly), which therefore reduces performance. I also think that people who really want to use vhd/vhdx Win2Go support can simply use WinToUSB (which I'm sure does a very decent job with it), so, once again, I fail to see how this is a priority for Rufus, especially when I have plenty of more important things to keep me busy.

>  but Rufus does allow WTG even on removable drives and also on drives smaller than 32gb, and that's not official.

I don't think the 32GB limit matters that much (that's a recommendation as far as I know), and there is a BIG WARNING when you don't use a FIXED drive that tells you it probably won't work. I did leave the ability to create REMOVABLE just in case someone manages to find a way to make it work without using a VHD.

Please understand one thing: There's a world of difference between following a procedure that is explicitly incompatible with VHD usage, and allowing people to do things, that deviate from the procedure, but that are NOT going to create a major incompatibility with the procedure from trying them, in case they work.

I'm afraid it looks to me like, as many people before you, you are pushing hard to get Rufus to do something that only a few people will actually want or even benefit from (allowing Win2Go on REMOVABLE will create all kind of performance issues as people try with slow drives). Of course, when you are the person requesting something that you really, really want, it's hard to see the rationale of the developer's choices, but I hope you will give me the benefit of having a reasonably good idea of what the majority of users of Rufus will want, from dealing with them day in and day out.

Of course, with Rufus being Open Source, anyone who disagrees with my choices is free to create their own fork to add the features they want.

NSbuilder, 2017-01-18:
> I'm not seeing that behavior when testing. But then again, I have no idea what ISO you are using. PLEASE, PLEASE, PLEASE, ALWAYS provide a full log (even if you don't actually go through the USB creation). The checklist is there for a reason, and I could find out many things that you didn't tell from the log (such as the version of Windows you are using, the ISO, and other important stuff).

Wow, I really had this old loc file - so the messages are fine but the OK and CANCEL buttons are still flickering.

Log
---
```
<Scanning image...
ISO analysis:
  Image is an UDF image
Disk image analysis:
  Image does not have an x86 Master Boot Record
ISO label: 'GRMCULF(X)REO_EN-RU_DVD'
  Size: 3.9 GB
  Uses: EFI (win7_x64)
  Uses: Bootmgr
  Uses: Install.wim (version 0.13.1)
Using image: Windows7_EN_x86_x64_18in1_Pre-Activated-by-roeibajayo.blogspot.com.iso
WIM extraction method(s) supported: 7-Zip, wimgapi.dll
WIM apply method supported: wimgapi.dll
Opening: \\.\CDROM0\sources\install.wim:[0] (API)
Extracting: C:\Users\Noam\AppData\Local\Temp\Ruf6A91.tmp (From [1].xml)
Closing: \\.\CDROM0\sources\install.wim
Found USB 2.0 device 'Kingston DataTraveler G3 USB Device' (0951:1643)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 1910, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 3
Disk ID: 0xC3072DD0
Drive has an unknown Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 1.6 GB (1756364800 bytes)
  Start Sector: 2048, Boot: No
Partition 2:
  Type: FAT32 (0x0b)
  Size: 5 GB (5368709120 bytes)
  Start Sector: 3432448, Boot: Yes
Partition 3:
  Type: GNU/Linux (0x83)
  Size: 8 GB (8589934592 bytes)
  Start Sector: 13918208, Boot: No>
```

> I never said that.
I mean, you said that maybe we won't need those 2 new messages but you want it translated ( If I'm not mistaking ).

> I ALREADY explained that Rufus does not support ESD, when you asked earlier today, and that I have no plans to do so, as I don't see it as a priority. There are just more important features that are [...]

Sorry, I haven't got your new message...
I just thought that maybe it is something "easy" to add ( I know developing is hard )

> I'm afraid it looks to me like, as many people before you, you are pushing hard to get Rufus to do something that only a few people will actually want or even benefit from (allowing Win2Go on REMOVABLE will create all kind of performance issues as people try with slow drives).

I'm not trying to push hard or something, I just mean that I think vhd/vhdx will make Rufus WTG more useful because if it just follow the official rules - you can just use the official tool from windows 8/10.
I mean, it is just doing what the official tool does...
I also requested the ability to choose an edition - which you have added to Rufus - thank you for that!
I'm not asking for a replacement for WinToUSB, I just offered an idea I think is good - Of course it is up to you to choose if you want it.

> Also it adds one level of virtualization (the disk is not being accessed directly), which therefore reduces performance

I read it just reduce a little performance, and also It worked for me on my old usb flash drive ( it was slow a bit but it worked ) while the official from Rufus got me stuck in the boot screen.

> I hope you will give me the benefit of having a reasonably good idea of what the majority of users of Rufus will want, from dealing with them day in and day out.

No problem.
And thank you for your great utility!

pbatard, 2017-01-18:
> but the OK and CANCEL buttons are still flickering.

Thanks to the log, I can tell that this is because you are using an __unofficial__ ISO, that attempts to cram every possible version of Windows onto the same image, and which is __UNSUPPORTED__.

Especially, Rufus doesn't support `install.wim` that contain something like 18 versions of Windows (which is what your ISO appears to do), so, yeah, the selection box, when you try to use Rufus with such an image, is unlikely to show properly.

As an aside, and notwithstanding the fact that your "pre-Activated" ISO is most likely illegal (therefore something I will never support anyway), the reason why Rufus does not support "All-In-One" (AIO) ISOs is that it is exceedingly foolish to install or run an OS from an image that was created by a third party, which is unaffiliated with the OS developer.

For instance, if I was a government backed security agency (especially one from a country where the head of state has close ties to security agencies), you can bet that the first thing I'd do would be to produce my own version of Windows, with attractive features (such as pre-activation, comprehensive choice of versions, or all the latest Microsoft patches streamlined), with dormant spyware that I'd break in a myriad of small pieces (since I have control of the whole OS, in order to make it virtually undetectable to regular security solutions), so that can try to entice people into installing it. Heck, with the very low cost of such an operation vs its potential effectiveness, and the "reasonable" excuse of hiding my real identify so as "not to be caught by Microsoft", as a spy agency, it's be very foolish not to release my own modified version of Windows for people to use.

Which is why, from a security standpoint, I can't say I have much of an issue if people trying to use an AOI Windows ISO get a subpar experience (or even can't use said ISO at all), as they will always be much better off installing Windows from an official ISO.

Besides, it's not like accessing the official ISOs is that hard. For instance, if I wanted to install a specific version of Russian Windows 7, I'd:
1. Go to MSDN subscriber downloads, and pick [Windows 7 and Russian](https://msdn.microsoft.com/en-us/subscriptions/downloads/#searchTerm=&ProductFamilyId=350&Languages=en&FileExtensions=.iso&PageSize=10&PageIndex=0&FileId=0).
2. Click on "Details" for the version of Windows I am interested in (e.g. _Windows 7 Ultimate with Service Pack 1 (x86) - DVD (Russian)_) and copy its SHA1 (`D224888CFA62847FAA3505630A206BDCD10A9930`)
3. Google that SHA1 + torrents and locate a torrent. Then, once the ISO is downloaded, I would use one of the many SHA1 applications (including Rufus) to confirm that the ISO I got is the same as the official one from Microsoft, and, only then, use it to install my OS.

So really, there's no reason from wanting to use a third party AIO that has even a slim chance to be infected with malware...

pbatard, 2017-01-18:
Since I don't have access to an ISO with more than 6 options, I have tested the selection dialog in a different manner, for up to 16 options (which is the max I set it to support), and found that there were a few issues that could use fixing, such as a typo in the `.rc` and some improvement with the resizing.

I have now done that, and you should find that the release should work better (though it will __never__ allow you to choose all 18 versions from your AIO).

commit db234442 (2017-01-18): [ui] improve selection dialog when there exist more than 6 options

* Closes #891
closed 2017-01-18 by pbatard