pbatard/rufus #892: USB drive not detected

shogunreaper, 2017-01-19:
Rufus doesn't seem to want to detect my usb drive. Its a Transcend 32gb usb 2 drive.

tried doing alt+f and checking list drives but nothing.

Log:
Found non-USB removable device 'JetFlash Transcend 32GB SCSI Disk Device' => Eliminated
0 devices found

pbatard, 2017-01-19:
Please try <kbd>Alt</kbd>-<kbd>.</kbd> with your device plugged, and post the new log. It should contain extensive device enumeration debug information, which will help troubleshoot the issue.

shogunreaper, 2017-01-19:
Rufus version: 2.11.995
Windows version: Windows 7 SP1 64-bit
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta3
System locale ID: 0x0409
Will use default UI locale 0x0409
Found non-USB removable device 'JetFlash Transcend 32GB SCSI Disk Device' => Eliminated
0 devices found
Processing Hub 1:
  Hub[1] = '\\?\iusb3#root_hub30#4&1148bc98&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[253]: USB\VID_054C&PID_0BA0\5&20C67EFD&0&4
  Found ID[049]: USB\VID_0C45&PID_7603\5&20C67EFD&0&6
  Found ID[027]: USB\VID_1D57&PID_FA60\5&20C67EFD&0&13
  Found ID[013]: USB\VID_0764&PID_0501\000000000000
  Found ID[101]: VID_8564&PID_1000&ASUS\{98F2FC55-265C-4405-9AA3-57C78C242DAF}\U3NVYWCF
Processing Hub 2:
Processing IDs belonging to 'USBSTOR':
Processing IDs belonging to 'ASUSSTPT':
  ASUSCR\{d6ef3e17-93db-4bd9-9cdd-e23bd67eaa55}\U3NVYWCF
Processing 'SCSI' device:
  Hardware ID: 'SCSI\DiskJetFlashTranscend_32GB__8.07'
Found non-USB removable device 'JetFlash Transcend 32GB SCSI Disk Device' => Eliminated
If you *REALLY* need, you can enable listing of this device with <Ctrl><Alt><F>
Processing 'SCSI' device:
  Hardware ID: 'SCSI\DiskOCZ-ARC1______________001.01'
Found non-USB non-removable device 'OCZ-ARC1 00 SCSI Disk Device' => Eliminated
Processing 'SCSI' device:
  Hardware ID: 'SCSI\DiskST3000DM______001-1CH166CC29'
Found non-USB non-removable device 'ST3000DM 001-1CH166 SCSI Disk Device' => Eliminated
Processing 'SCSI' device:
  Hardware ID: 'SCSI\Disk_TOSHIB_______DT01ACA100MS2O'
Found non-USB non-removable device 'TOSHIB  DT01ACA100 SCSI Disk Device' => Eliminated
0 devices found

pbatard, 2017-01-19:
Well, first off, your device is not seen by Windows as an USB device. It might be plugged on an USB port, but I suspect you installed a proprietary ASUS driver (I know ASUS tend to like to do that, and that `ASUSSTPT` is the _ASUS USB 3.0 Boost Storage Driver_) which, on the pretence of increasing performance, turns your USB flash drive into a Frankenstein device that is neither an USB not an SCSI drive, but something in between. So of course, for an application that is designed to work with USB devices, this is going to create all kind of issues.

If I were you, I would therefore uninstall the ASUS USB 3.0 Boost Storage Driver_ and just use the regular USB 3.0 driver for device access.

Otherwise, as the log indicates, you can use <kbd>Ctrl</kbd>-<kbd>Alt</kbd>-<kbd>F</kbd> (the extra <kbd>Ctrl</kbd> is important) to get your device listed...

shogunreaper, 2017-01-19:
well it only seems to happen to this usb drive

my 16gb and 8gb drives work fine

pbatard, 2017-01-19:
Are they USB 3.0?

shogunreaper, 2017-01-19:
no they aren't usb 3, they are usb 2 drives.

i bought them ages ago

idk what uas is

pbatard, 2017-01-19:
Well, as the name tells, the _ASUS USB 3.0 Boost Storage Driver_ is an __USB 3.0__ driver, so, of course, it will only affect USB 3.0 devices, and not USB 2.0 ones. I also think that it only affects USB 3.0 that also support the UAS protocol (which not all USB 3.0 devices do).

So, yeah, that will explain why some devices, that don't try to rely on the driver for anything, will work just fine, and the ones that do, won't.

By the way, did <kbd>Ctrl</kbd>-<kbd>Alt</kbd>-<kbd>F</kbd> work?

pbatard, 2017-01-19:
Oh, and with regards to UAS, you can check the [Wikipedia entry](https://en.wikipedia.org/wiki/USB_Attached_SCSI). Or you can look at [ASUS' page for _Boost Storage_](http://event.asus.com/mb/2010/The_Best_USB3_Experience/The_UASP_For_USB3.0.htm) (being mindful that this is a promotional page by the manufacturer, so not entirely objective about advantages and drawbacks).

shogunreaper, 2017-01-19:
but like i said this is not a usb 3 device.

this is the drive i have
https://www.transcend-info.com/Products/No-267

and pressing ctrl+alt+f worked, this is the log

1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 3904, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x31B4CB59
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 29.9 GB (32108445696 bytes)
  Start Sector: 2048, Boot: Yes

pbatard, 2017-01-19:
UAS can apply to USB 2.0 devices as well, and I'm pretty sure that your device is UAS, which means it will be accessed through the ASUS driver and not the native USB 2.0 stack from Windows.

At any rate, from what I see in the debug log, I can __guarantee__ that your device is not being accessed by the native Windows USB stack, and, again, I would encourage you to uninstall the ASUS driver if you don't want that to happen.

pbatard, 2017-01-20:
Since there haven't been further updates, I'm going to close this issue.

closed 2017-01-20 by pbatard